me-no-dev/ESPAsyncWebServer #966: How to serve huge generated content

copych, 2021-04-13:
I have a functions in my project that generate graphic content. Several MBs, but in case of ESP we can call it huge )
What is the right way to feed such content. Intuitively I'd use chunked response, but I don't quite get how to wrap this. When I tried response->print(), it only feeds a certain amount of kB and then reboots :-/
Please, advise.

lorol, 2021-04-13:
Try websockets

zekageri, 2021-04-16:
I would suggest you to write it to a file and download that file client side with ajax.

copych, 2021-05-02:
> I would suggest you to write it to a file and download that file client side with ajax.

This is what I do, but don't like it, cause I spend precious FS bytes while I could do that just sending content generated on the fly.

closed 2021-07-16 by stale[bot]