me-no-dev/ESPAsyncWebServer #968: Max character in array json to send to web server

Cutiino, 2021-04-15:
Hello, I am triying to send a json array to local web server with esp32. Until now, I can send a array json with a max of 10.000 characters. Once over this character, the esp32 is reset.

This is my code in the part that I send the array json

  server.on("/cosecha", HTTP_GET, [](AsyncWebServerRequest * request) {
    char response[10000]; 
    JsonCosecha.printTo(response, JsonCosecha.measureLength() + 1);
    request->send(200, "application/json", response);
    //request->send(200, "text/plain", JsonCosecha);
  });

I aprecciate any help

zekageri, 2021-04-16:
Your client side language is Javascript?

Cutiino, 2021-04-16:
No, I have been checking with a notebook connected to ESP32 network. Then I put the URL 192.168.4.1 in the browser and in web page appear the array json.

zekageri, 2021-04-18:
But are you in controll of the client's script?

Cutiino, 2021-04-20:
Yes, the client's script is in Javascript.  Finally, I used WebSockets for fixing this issue.

closed 2021-07-08 by stale[bot]