me-no-dev/ESPAsyncWebServer #969: CHunked response callback called again after returning 0

MichielfromNL, 2021-04-17:
I  use chunked response for sending log data .
The process requires heap, which I need for other tasks.  So the logreader is dynamic, deleted when done., Since the handler terminates after the response is send, the reader class can only be deleted when the last chunk is done. Code below illustrates the process.
The problem:  the lambda function is called 1 more time, after it returns "0" ,  Hence a crash, because the reader has been deleted. I have tried everything, nothings helps
See code below, and pls advise, why does the webserver call the chunk after it has returned zero ?

//
void handleLog (AsyncWebServerRequest * request) {

  const char * cts;
  logformat_t logformat;
  bool archive = false;
  char s[20] = "html";

  FlightData *Fi = new FlightData(client);
  
  if (request->hasParam("format")) {
    String fmt = request->getParam("format")->value();
    if (fmt) {
      fmt.toCharArray(s,20);
    }
  }
  
  switch (stringhash(s)) {
    case "txt"_:   logformat = LOG_TXT ; cts = "text/plain"; break;
    case "csv"_:   logformat = LOG_CSV;  cts = "text/csv"; break ;
    case "html"_:  logformat = LOG_HTML; cts = "text/html"; break;
    case "json"_:  logformat = LOG_JSON; cts = "application/json"; break;
    default: logformat = LOG_HTML ; cts = "text/html"; 
  }
  Serial.print(F("Log request format = ")); Serial.println(cts);

  if (! Fi->readBegin(logformat)) {
      request->send(200,"text/plain","No log data available");
      return;
  }
  
  AsyncWebServerResponse *response = request->beginChunkedResponse(cts, [Fi](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
  //Write up to "maxLen" bytes into "buffer" and return the amount written.
  //index equals the amount of bytes that have been already sent
  //You will be asked for more data until 0 is returned. 
  //Keep in mind that you can not delay or yield waiting for more data!

  // BUG: chunk called again after 0 returned !! how to solve..?
  
    size_t r = 0;
    //Serial.println("Chunk callback: " + String(maxLen));
    
    r = Fi->readLog((char *)buffer,maxLen);
    if (r == 0) {
      delete Fi;
      Serial.printf("ReadLog complete, bytesread %d of %d total %d\n", r, maxLen, index );
    }
    Serial.printf("Chunk read complete, bytesread %d of %d , sofar %d \n", r, maxLen, index);
    return r;
  
  },processor);

  response->addHeader("Server","Airsound ESP Web Server");
  if (logformat == LOG_CSV) {
    response->addHeader("Content-Disposition", "attachment;filename=soundlog.csv");
  }
  // here we start the chunking
  request->send(response);
  
}

ladnene, 2021-05-08:
weird , when i used chunked response with processor i don't remember it calls the callback after returning 0,  
i suggest you test these solutions:
*first solution:
add  mutable specifier to the lambda expression like this : 

  AsyncWebServerResponse *response = request->beginChunkedResponse(cts, [Fi](uint8_t *buffer, size_t maxLen, size_t index) **mutable** -> size_t {...}

it will allows the body to modify the objects captured by copy which is in your case the  [Fi] , i know it shouldn't be needed as you are using a pointer (reference) not a value but try it.

*second solution:
    if (r == 0) {
      delete Fi;
      Serial.printf("ReadLog complete, bytesread %d of %d\n", r, maxLen);
      return 0;
    }
i know it will return zero here "return r;" but like it calls it again i'am not so sure of anything anymore, it might need zero integer type to be returned so it don't call it again and not a size_t .

*third solution solution:
check the Fi pointer is not null before reading log from it which is causing the crash
```
if (Fi != null)
{
     r = Fi->readLog((char *)buffer,maxLen);
    if (r == 0) {
      delete Fi;
      Serial.printf("ReadLog complete, bytesread %d of %d\n", r, maxLen);
    }
}
```
like this if for any reason it needs to call the callback few times again after returning 0 (which shouldn't happen) it won't crash

MichielfromNL, 2021-05-10:
Thanks!

I checked all three solutions individually, the combination of #1 and #3 id the job.  mutable and checking the deleted instance .

I guess that the bug is caused by the expanded size of the response after processor adds data.  
If I call my Log class to produce "%free" data, there is no problem. See below

HTML with and %TOKEN%  :
Log request format = text/html
Chunk callback: maxLen 5559 index 0
Done, EOF 
Chunk read complete, read /return 2010 of 5559
Chunk callback: maxLen 3557 index 2010
ReadLog complete, bytesread 0 of 3557
Chunk read complete, read /return 0 of 3557
Chunk callback: maxLen 3549 index 2010     <-- the extra call after close
Chunk read complete, read /return 0 of 3549

JSON without %TOKENS%   :
Log request format = application/json
Chunk callback: maxLen 5552 index 0
Done, EOF 
Chunk read complete, read /return 1250 of 5552
Chunk callback: maxLen 5736 index 1250
ReadLog complete, bytesread 0 of 5736
Chunk read complete, read /return 0 of 5736
<-- no extra call when there is no %xx% token to expand in the data  

Thanks for the help!

closed 2021-07-26 by stale[bot]