me-no-dev/ESPAsyncWebServer #971: Add ESPAsyncWebServer to the Arduino Library Manager

dok-net, 2021-04-21:
Subsequently to issue https://github.com/me-no-dev/ESPAsyncTCP/issues/168,
the library.properties file should probably be updated to list the exclusively supported architectures in advance,
plus the automatic dependency on ESPAsyncTCP should be declared as well.
Instructions for the publishing process are [here](https://support.arduino.cc/hc/en-us/articles/360012175419-How-can-I-add-my-library-to-the-Library-Manager-)

zekageri, 2021-04-21:
Currently this webserver has a lot of issues. Ofc this is a really really amazing piece of art but dozens of users reporting crashes by websockets, events and http requests. Do you think it would be wise if it will be in the library manager?

dok-net, 2021-04-21:
It's an open issue nevertheless.
But for the time being, you are right, and it would take a release before announcing it to the library manager support.

closed 2021-07-08 by stale[bot]

s-light, 2022-01-12:
> It's an open issue nevertheless. But for the time being, you are right, and it would take a release before announcing it to the library manager support.

do you think this will happen this year?

iam a little bit skeptical of what the status of this project is.....
(there are 59 outstanding PullRequests - my feeling is that it is a little bit stale..)

per1234, 2024-12-17:
There is a new incarnation of the request at https://github.com/me-no-dev/ESPAsyncWebServer/issues/1433