me-no-dev/ESPAsyncWebServer #972: HTTPS with ESPAsyncWebServer (suggestion / functionality)

GabenRulez, 2021-04-22:
I would love to see HTTPS support (even just self-signed SSL certificates) with this library. It would definitely make my life easier :)

zekageri, 2021-04-23:
Yeaah, it would be really awesome.
Some folks actually started doing it or at least trying out things. 
[See it in here](https://github.com/me-no-dev/AsyncTCP/pull/90)

I have made a node js server within localhost and port forwarded it with https. 
I could do the same with my esp32 if the lib would support the https handsake. I would be more than happy.

proddy, 2021-04-24:
if you're on ESP32 there are easier solutions

zekageri, 2021-05-04:
> if you're on ESP32 there are easier solutions

For https?

proddy, 2021-05-04:
> > if you're on ESP32 there are easier solutions
> 
> For https?

https://github.com/fhessel/esp32_https_server

zekageri, 2021-05-04:
Oh, i knew about that, but for this i have to port my entire sketch because it is heavily based on asyncweb. :/

closed 2021-07-22 by stale[bot]