me-no-dev/ESPAsyncWebServer #978: Esp crash when websocket enabled and call webServer.reset()

imojams, 2021-04-30:
When wifi reconnect, esp crash

Code
[AsyncWebWsBug.log](https://github.com/me-no-dev/ESPAsyncWebServer/files/6406829/AsyncWebWsBug.log)

Stacktrace
```
10:41:17.840 -> assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c", line 267, function: heap_caps_free
10:41:17.840 -> abort() was called at PC 0x4010605b on core 1
10:41:17.840 -> 
10:41:17.840 -> Backtrace: 0x4008c534:0x3ffb9330 0x4008c765:0x3ffb9350 0x4010605b:0x3ffb9370 0x40084b4b:0x3ffb93a0 0x40085019:0x3ffb93c0 0x4000bec7:0x3ffb93e0 0x40138569:0x3ffb9400 0x400d1b00:0x3ffb9420 0x40145ecf:0x3ffb9440 0x400d81b7:0x3ffb9460 0x400d8421:0x3ffb9480 0x400d140a:0x3ffb94c0 0x400d15c7:0x3ffb9510 0x400d892e:0x3ffb9550 0x400d89fd:0x3ffb9640 0x40088c7d:0x3ffb9670
10:41:17.874 -> 
10:41:17.874 -> Rebooting...

Decoding stack results
0x4008c534: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 155
0x4008c765: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 170
0x4010605b: __assert_func at ../../../.././newlib/libc/stdlib/assert.c line 63
0x40084b4b: heap_caps_free at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/heap_caps.c line 267
0x40085019: _free_r at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/syscalls.c line 42
0x40138569: operator delete(void*) at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libstdc++-v3/libsupc++/del_op.cc line 46
0x400d1b00: AsyncWebSocket::~AsyncWebSocket() at C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESPAsyncWebServer\src\AsyncWebSocket.cpp line 871
0x40145ecf: std::_Function_handler   >::_M_invoke(const std::_Any_data &, AsyncWebHandler * const&) at C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESPAsyncWebServer\src\WebServer.cpp line 36
0x400d81b7: std::function ::operator()(AsyncWebHandler* const&) const at c:\users\dev\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d8421: AsyncWebServer::reset() at C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\ESPAsyncWebServer\src/StringArray.h line 166
0x400d140a: setupWebServer() at C:\Users\dev\Desktop\esp32\AsyncWebWsBug/AsyncWebWsBug.ino line 82
0x400d15c7: WiFiEvent(system_event_id_t) at C:\Users\dev\Desktop\esp32\AsyncWebWsBug/AsyncWebWsBug.ino line 73
0x400d892e: WiFiGenericClass::_eventCallback(void*, system_event_t*) at C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src\WiFiGeneric.cpp line 436
0x400d89fd: _network_event_task(void*) at C:\Users\dev\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.4\libraries\WiFi\src\WiFiGeneric.cpp line 61
0x40088c7d: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```

closed 2021-07-14 by stale[bot]

davidkleng, 2021-11-03:
Same problem.

stevesch, 2022-01-29:
An apparent fix if you're using an AsyncWebSocket follows.

It appears that the server code is somewhat asymmetric, in that it expects you to allocate the sockets, but it deletes them... kind of nasty (perhaps there is a more intended setup, but I didn't see any more "correct" examples).

>>> If you're doing something like this,
AsyncWebServer server(80);
AsyncWebSocket someSocket("/something");

// connect:
server.addHandler(&someSocket);
. . .
// disconnect:
server.removeHandler(&someSocket);

>>> instead do this:

AsyncWebServer server(80);
AsyncWebSocket* someSocketPtr = 0; // USE DYNAMICALLY ALLOCATED SOCKET

// connect:
if (!someSocketPtr)
{
  someSocketPtr = new ("/something");
  server.addHandler(someSocketPtr);
}
. . .
// disconnect:
if (someSocketPtr) {
  // removeHandler INTERNALLY CALLS 'delete' ON YOUR OBJECT:
  server.removeHandler(someSocketPtr);
  // do not delete someSocketPtr here-- removeHandler already did this
  someSocketPtr = 0; // just clear the pointer
}