me-no-dev/ESPAsyncWebServer #984: Crashing on downloading static files seems size dependent

ullix, 2021-05-08:
I have static, binary log files which I download via WiFi. Works well most of the time, but then there are perpetual crashes. It seems the success depends on the size of the files. Up to 64k mostly ok, 64k ... 128k ok for half the trials, beyond that size almost never ok.

I am using the LittleFS file system. The files are build up of 32 bytes binary records, and may eventually grow >2MB. I use this command for downloading as static file:

```
// file system definition:   fs::LITTLEFSFS  * myFS       = &LITTLEFS;
server.serveStatic("/log.cam",      *myFS, "/log.cam");
```
I also tried this. Works just as well, but no improvement:
```
server.on("/log.cam", HTTP_GET, [] (AsyncWebServerRequest *request) {request->send(*myFS, "/log.cam", application/octet-stream");});
```

The download is triggered from a website, either as a straight link to download the binary file, or as Javascript code inside a function:

```
Link:   "<a href='/log.cam' >CAM</a>"
...
JS:   const fetchdata     = await fetch('/log.cam', {cache: "no-store"});
```

Your ReadMe seems to suggest that chunked responses are NOT needed for static files? So, I did not use that here. Am I forgetting any settings, or do I also need to use chunked response  when the static file is beyond a certain size? Which size?

ullix, 2021-05-09:
Upon further investigations I conclude that this is a bug. For me a very serious one as it drastically reduces download speed by a factor of more than 10!

As first test I also coded a chunked response, but as expected, it showed the same failures as reported above. However, this chunk code now allowed to define the chunk size, and this made a huge difference!

I created a 300k (307200 B) file and downloaded that in various configurations. In all tests it always crashed with a wdt message when the download touched 100000; I observed only the two values of 100030 and 101459. Message was:

```
E (144357) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (144357) task_wdt:  - async_tcp (CPU 1)
E (144357) task_wdt: Tasks currently running:
E (144357) task_wdt: CPU 0: IDLE0
E (144357) task_wdt: CPU 1: loopTask
E (144357) task_wdt: Aborting.
abort() was called at PC 0x401058a4 on core 0
```

I used various ESP32-WROOM-32E (all latest revision 3) with Flash of 4MB, 8MB, and 16MB. All gave the same result.

Using the code included below I tested for the impact of chunk size, and got a big surprise:

- 1028 Bytes chunk size : all downloads worked. Even a file of 1.2MB could be downloaded correctly!

- 1029 Bytes chunk size : only 1 Byte more and all downloads failed when the filesize was 100k or greater! Files of size 90k so far could be downloaded without problems.

I tested the file systems FFat and LittleFS. Outcome was identical.

I noticed that the download was very slow, slower by about an order of magnitude!

So I used a file of 90k, which could be downloaded under all conditions, and measured download times:

- 1028 Bytes chunk size :  3.1 sec
- 1029 Bytes chunk size :  0.4 sec
- (set by async server)    :  0.2 sec   (varied between 5600 and 1400 Bytes during a download)

Using 1028 Bytes makes it 15 times slower, but is the only option for big files, when you need speed most!

And one last surprise: removing the print options in  chunk-code made it even slower, not faster!

Here is the relevant code:
```
// File system
// either:
    fs::F_Fat       * myFS       = &FFat;
// or:
    fs::LITTLEFSFS  * myFS       = &LITTLEFS;

// ESPAsyncWebServer setup
    server.on("/log.cam",       HTTP_GET, [] (AsyncWebServerRequest *request) { staticDataType = CAM;
                                                                                request->sendChunked(text_plain, onStaticDownLoad);});

// chunk code
size_t onStaticDownLoad(uint8_t *buffer, size_t maxLen, size_t index){

    static  char *   filename;                 // file to be sent as chunks
    static  File     chunkfile;                // file handle of filename
    static  size_t   dnlLen     = 1028;        // limit to buffer size
            uint32_t countbytes = 0;           // bytes to be returned

    if (index == 0)  {
        if (staticDataType == CAM)  filename    = LOGCAM;
        else                        filename    = LOGCPS;
        samm->printf("onStaticDownLoad: START filename: %s index: %6i maxLen: %5i\n", filename, index, maxLen);
        chunkfile       = myFS->open(filename);
    }

    chunkfile.seek(index);
    countbytes = chunkfile.read(buffer, min(dnlLen, maxLen));

    if (countbytes == 0) {
        chunkfile.close();
    }
    samm->printf("onStaticDownLoad: f: %s index: %6u maxLen: %5u countb: %5u \n"
                    , filename
                    , index
                    , maxLen
                    , countbytes
                );

    return countbytes;
}

```

ullix, 2021-05-20:
This is still a major problem for me, so I investigated further. To simplify I only used the webserver with command 
```
server.serveStatic("/bindataST",      *myFS, "/data.cam");
```

and only on ESP32-WROOM-32E (revision3, the latest models) with 4MB Flash with the LittleFS file system. I created binary data files of various sizes. (While saved in binary mode, all bytes were printable ASCII, so the file content was legible to humans).

The data shown below were compiled with Arduino 2.0 beta 7, and ESP core 1.0.6, but I repeated some tests with Arduino 1.8.13 and ESP core 1.0.4 and got same results.

I automated this by creating an auto-refreshing website containg JavaScript, by which a `serveStatic` function of the webserver was called, and logged to the Serial Terminal.

Up to a file size of 150 Kibibytes all downloads were always successful. In an overnight run with a 90KB file all 15833 downloads were successful, not a single failure.

However, with only a few bytes more, 160 Kibibytes, the number of Watch-Dog-Timer induced crashes increased progressively. All were of the same type:

```
E (69589) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (69589) task_wdt:  - async_tcp (CPU 1)
E (69589) task_wdt: Tasks currently running:
E (69589) task_wdt: CPU 0: IDLE0
E (69589) task_wdt: CPU 1: loopTask
E (69589) task_wdt: Aborting.
abort() was called at PC 0x4014246c on core 0
```

The successful downloads decreased to as low as 58%, i.e. almost every 2nd download attempt resulted in a crash and reboot! Somewhat strangely, beyond a file size of ~300K the success rate increased a little bit, and was 70% at 1.4MB,  the biggest file possible in this setting.

[image]

[image]

As I had mentioned on previous posts, one can mitigate this WDT problem by using chunked downloads. However, this slows down the transfer drastically by 40...50 times! Here are two examples done subsequently:

This shows the web page result for serveStatic (picking one when it worked!):
```
Binary Data in Human-Readable Format
#Total: 1433600 bytes, 1400 kB, 44800 records, dur: 2834 ms, speed: 506 kB/s
1 0123456789ABCDEFGHIJKLMNOPQRSTUV
2 0123456789ABCDEFGHIJKLMNOPQRSTUV
```
A speed of 506 kB/s is very nice; top-speed observed was even a fantastic ~800kB/s!

With a chunked download you need to limit the maxLen of the chunk to 1028 bytes, or you face again the WDT crash issue. With that limit this is what you get:

```
Binary Data in Human-Readable Format
#Total: 1433600 bytes, 1400 kB, 44800 records, dur: 124706 ms, speed: 11 kB/s
1 0123456789ABCDEFGHIJKLMNOPQRSTUV
2 0123456789ABCDEFGHIJKLMNOPQRSTUV
```
Instead of 3 sec it now takes over 2 min, a speed of only 11 kB/s!

I am building a device which logs data, and I will use even the bigger 8MB and 16MB ESP variants and files filling the Flash. This webserver cannot handle this!

BlueAndi, 2021-05-23:
You download the files in the context of the web server (Async TCP task). Did you try to move it to the loop context?

ullix, 2021-05-23:
@BlueAndi How can I do this? I think it is not possible, but please, correct me.

My webserver initialization gets this code:

```
AsyncWebServer      server(80);
void initWebServer() {
    ...
    server.serveStatic("/bindataST",      *myFS, "/data.cam");
    server.begin(); 
}
```

Then from a website I call `<ESP-IP>/bindataST`. That's it. Which part can I relocate?

BlueAndi, 2021-05-23:
You can't do this with serviceStatic(), but you can handle the request by your own.
Please note, below is just a rough principle I could imagine, which would be worthful to check. To be task safe, you can later use a freeRTOS queue, instead of a deferredRequest variable.

```

static AsyncWebServerRequest* deferredRequest = nullptr;

static void downloadPage(AsyncWebServerRequest* request)
{
    deferredRequest = request;
}

...
void setup()
{
    ...
    server.on("/bindataST/data.cam", HTTP_GET, downloadPage);
    server.begin();
    ...
}

...
void loop()
{
    ....
    if (nullptr != deferredRequest)
    {
        deferredRequest->send(LITTLEFS, "/bindataST/data.cam", String(), true);
        deferredRequest = nullptr;
    }
    ...
}

```

zekageri, 2021-05-23:
Wow. This is gold

zekageri, 2021-05-25:
### This seems the common async server crash problem to me.
Can you replace the ESPAsyncWebServer lib with  [this forked version by Depau](https://github.com/Depau/ESPAsyncWebServer/tree/partial-header) version and try again?

I'm currently testing this fork and it seems very stable

ullix, 2021-05-25:
I'll try. Looking at both ReadMEs, the Depau version seems to be taken verbatim from me-no-dev. Where is the difference in the code? Which part is changed?

One other thing bothers me: Both parties offer their code is explicitly for ESPs. I learned that neither PROGMEM nor the F() macro  make any sense on these CPUs (https://www.esp32.com/viewtopic.php?f=19&t=20595 ). Yet, both offerings expressly advertise for PROGMEM and for F(). Are they not aware of this? I don't believe that. So, why this nonsense?

zekageri, 2021-05-25:
[Here is the PR by Depau](https://github.com/me-no-dev/ESPAsyncWebServer/pull/952) He fixed some linked list problems and things like that.

Abot the PROGMEM and F() things i don't know. Probably compatibility solutions between esp8266 and esp32 and things like that. There are so many esps around, you never know what was Me_no_dev's intention with these.

zekageri, 2021-05-25:
> I'll try. Looking at both ReadMEs, the Depau version seems to be taken verbatim from me-no-dev.

Yes. It is a fork from me-no-dev's lib. The readme and almost everything is the same except the linked list and some other fixes.

Pablo2048, 2021-05-25:
> One other thing bothers me: Both parties offer their code is explicitly for ESPs. I learned that neither PROGMEM nor the F() macro make any sense on these CPUs [...]

The reason for this is compatibility - ESP8266 requires using PROGMEM/F() if you don't want to waste the RAM. Definitely NOT a nonsense...

ullix, 2021-05-25:
Getting the Depau lib is not very promising, so far:

I downloaded the ZIP which gave me a file of this name: `ESPAsyncWebServer-partial-header.zip`. The 'partial' is irritating, but I tried several times and Firefox tells me the download had correctly completed.

The tried to install in Ar1.8.13 and got this message:
[image]

I tried Ar2.0beta7 and got this:
[image]

Then I tried `$ gh repo clone Depau/ESPAsyncWebServer` and got this:
`ERROR: ld.so: object 'libgtk3-nocsd.so.0' from LD_PRELOAD cannot be preloaded (failed to map segment from shared object): ignored.`

Now, what?

BlueAndi, 2021-05-25:
> What I find strange is the async_tcp (CPU 0/1) message. What does it mean? Is the WDT triggered from both CPU cores at exactly the same time? Out of a total of 293 crashes, 292 were of this (CPU 0/1) type, and only 1 of (CPU 1) type, none of (CPU 0) type.
> 
> So, while greatly improved, this is not the final solution. What else can I do?

The ```async_tcp (CPU 0/1)``` sounds for me like, the loopTask is calling the send() function, which try to interact with the LwIP stack. But the LwIP stack is already waiting for the async_tcp queue, but the queue might be full. This causes the deadlock, just a rough guess.

To debug it down, I propose the following actions, each to be executed independently:
* Increase the async_tcp queue size: https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.cpp#L98
Expected result: The magic border of 300k moves higher.
* To see whether its just really a timing issue, disable the watchdog of the async_tcp task by global define the following preprocessor defines in e.g. the platformio.ini
```cpp
#define CONFIG_ASYNC_TCP_RUNNING_CORE 0
#define CONFIG_ASYNC_TCP_USE_WDT 0
```

> One other thing bothers me: Both parties offer their code is explicitly for ESPs. I learned that neither PROGMEM nor the F() macro make any sense on these CPUs [...]

Just a note to PROGMEM, it depends on the MCU architecture (harvard, von neumann) and to say the truth, there might be exceptions depended on the used adresses.

zekageri, 2021-05-25:
> Getting the Depau lib is not very promising, so far:
>
> I downloaded the ZIP which gave me a file of this name: `ESPAsyncWebServer-partial-header.zip`. The 'partial' is irritating, but I tried several [...]

You have to unzip the lib.

ullix, 2021-05-25:
> You have to unzip the lib.

I don't believe so. Both Ar1.8 and 2.0 have an explicit option to install ZIP libs. Like here Ar2.0

[image]

ullix, 2021-05-25:
@BlueAndi I tried your 2nd option first and put this
```
#define CONFIG_ASYNC_TCP_RUNNING_CORE 0
#define CONFIG_ASYNC_TCP_USE_WDT 0
```
at the top of the code. Compiled and ran, made a 300k file and downloaded. Out of 10 attempts to download, only 1 succeeded, and 9 crashed. All for WDT, and all with `- async_tcp (CPU 0/1)`. That is the worst-ever result.

I can only use IDEs Ar1.8 or Ar2.0. I have Platformio installed, but so far have troubles to run it (some installation problem).

What can be done?  I can offer my test sketch, if you want to try it.

BlueAndi, 2021-05-25:
Hmm ... I don't believe that the task watchdog was deactivated. I use platformio inside VSCode, as it offers more options.
Provide your test sketch and I will try during the week.

ullix, 2021-05-27:
I am still seeing these annoying crashes, though progress has been made. I am showing an encouraging update, and the code at the end.

Here is the current stand as a graphics in the exact scale as above (https://github.com/me-no-dev/ESPAsyncWebServer/issues/984#issuecomment-845206395). 

[image]

Much, much better, and basically all due to BlueAndi's suggestion for deferring the download execution into the Loop. But still some 1% crashes, which strangely seem to be largely around the 300k.

Here a picture on the download speed:

[image]

The smaller files may be impacted by the download overhead, but from 500k onwards 700kB/s are reached and held. I have seen better - up to 800kB/s -  with this chip, but 700 would be quite alright.

These data were obtained with the code, I am attaching. Please, see the READ.ME for usage instructions. I ran it under Arduino 1.8.13, Arduino2.0beta7, and the uecide IDE. Let me know of any issues.

[TESTPUB_async_server.zip](https://github.com/me-no-dev/ESPAsyncWebServer/files/6554548/TESTPUB_async_server.zip)

zekageri, 2021-06-01:
My async server prject running 4 days straight for now ( with continuous  up and downloads ). Can you try replace me_no_dev's ESPAsyncWebServer lib with [this forked one from Depau](https://github.com/Depau/ESPAsyncWebServer/tree/wi-se-patches)?

My below link was not the full fixed fork

gnalbandian, 2021-06-01:
Folks, how are you.

Many of the problems with this library reside in the undelying use of AsyncTCP or ESPAsyncTCP library aswell.
There many forks that have already got this working steadily.

The combination I' currently using is:

Lorol's ESPAyncWebServer fork: [Lorol's](https://github.com/lorol/ESPAsyncWebServer)

Adam5Wu ESPAsyncTCP/ [Adams](https://github.com/Adam5Wu/ESPAsyncTCP)

Also @philbowles have made his own version of this libraries and seems to work fine.
Please check: https://github.com/philbowles/PangolinMQTT

ullix, 2021-06-01:
> Can you try replace me_no_dev's ESPAsyncWebServer lib with [this forked one from Depau](https://github.com/Depau/ESPAsyncWebServer/tree/wi-se-patches)?

@zekageri : 
This time it worked well to download the ZIP and install.

I ran it with the code I had published above. Unfortunately, it crashed as happily as the original me-no-dev code! At best, every second download crashed, but on average it was closer to 2 out of 3 crashing!

So this lib is not an improvement.

Am I correct that your 4-day-running project did not use my code, and that the downloaded files were smaller?

The sweet spot for bad behavior seems to be a 300k file. I make a binary file of records with 32 bytes of printable ASCII chars (makes it easier to read a "binary"), 9600 records, 307200 bytes.

I semi-automate the downloads by using HTML to auto-refresh the web page every 10 sec, and then count reboots and successful downloads from the Serial Monitor log.

I observed in some trials that having Core Debug Level to anything but 'None' made it worse; though that also was inconsistent. Anyway, I suggest to set Core Debug Level=None, and also comment out the `esp_log_level_set("*", ESP_LOG_VERBOSE)`. It remains confusing.

ullix, 2021-06-01:
> Lorol's ESPAyncWebServer fork: [Lorol's](https://github.com/lorol/ESPAsyncWebServer)

@gnalbandian I tried this Lorol version of the webserver, but nope, same bad performance as with the others. A successful download about once in three trials!

Have you guys really tried to download 300k files? See my files comments here  https://github.com/me-no-dev/ESPAsyncWebServer/issues/984#issuecomment-852263756 and in the source code, attached here: https://github.com/me-no-dev/ESPAsyncWebServer/issues/984#issuecomment-849659026

Am I the only one with highly reproducible problems in downloading big files?

gnalbandian, 2021-06-01:
Have you replaced ESPAsyncTCP library aswell?

BlueAndi, 2021-06-01:
> Have you replaced ESPAsyncTCP library aswell?

For esp32 AsyncTCP must be replaced, ESPAsyncTCP is only for esp8266.

Funny ... I have my own fork of both AsyncTcp and ESPAsyncWebServer. :-D

BlueAndi, 2021-06-01:
@ullix Just to be sure, but you don't have the TESTPUB_async_server.ino and etc. as a vscode + platformio project available?

BlueAndi, 2021-06-01:
And again to be sure, can you ensure that the downloadPage() is not called during a already running download?

Currently it is not ensured:
```
static void downloadPage(AsyncWebServerRequest* request){

    Serial.println("2 webserver: downloadPage");
    deferredRequest = request;
}
```

ullix, 2021-06-03:
In increasing desperation I wanted to make sure that the choice of file system does not play an unexpected role. So in 3 subsequent runs I tested FFat, LittleFS, and SPIFFS. (Easy to do in my testcode, attached here: https://github.com/me-no-dev/ESPAsyncWebServer/issues/984#issuecomment-849659026 just change the FS in lines 28-30, and make sure to have matching partitions). In all runs I tried to download a 300k file.

```
    // FS: FFat:      File: 300.0 kB, Success:  25.0%
    // FS: LittleFS   File: 300.0 kB, Success:  26.8%
    // FS: SPIFFS     File: 300.0 kB, Success:  19.3%
```

This time only 1 in 4 or even 1 in 5 succeeds, all other attempts crash with WDT. FFat and LittleFS are basically the same. SPIFFS is a bit further behind, but this is not an option anyway, because I found SPIFFS to be a 'Flash Memory Catastrophe' as shown here: https://forum.arduino.cc/t/esp32-mit-spiffs-eine-speicher-katastrophe/621573

I am not seeing any further option as to change the Watch Dog Timer duration. Can anyone tell me how to do this in the Arduino framework? If I have to do it by hard-coding somewhere in the ESP* files, that'd be ok too.

BlueAndi, 2021-06-05:
I used my esp32 project and downloaded your TEST300k.cam file 400 times without any reset under following conditions:
* FS: SPIFFS
* Request is handled in AsyncTcp task context.
* No pause between each download.

Hmm ...

ullix, 2021-06-05:
> Hmm ...

Hmmm. So true! This really freaks me out. Could you please say which OS you are using, and what the precise type of your ESP is?

Is there a chance I could get your code for testing?

I have searched for the file where all the Watch Dog Timers are defined, and it seems to be 
`~/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/config/sdkconfig.h`

I looked up all #defines which had 'WDT' in them, and changed and then tested all values in 3x steps, like x1, x3, x10, x30. There was some minor response on some, but nothing ever near 100%. What I think being the most likely candidate is:

```
#define CONFIG_TASK_WDT 1
```
I guess this is in units of seconds? I tried 3, 10, 30, with success rates of 59%, 81%, 78%, 55%. It does not look like the 'WDT crashes' do even have anything to do with that WDT setting. Is that possible?

BlueAndi, 2021-06-06:
I used the [TTGO T-Display esp32](http://www.lilygo.cn/prod_view.aspx?TypeId=50033&Id=1126&FId=t3:50033:3) module for test and my current "unstable" branch of [Pixelix](https://github.com/BlueAndi/esp-rgb-led-matrix/tree/PluginDevelopment).
For the test I removed some images from /data/images folder, placed the test file to /data and used the following code snippet to the /src/Web/Pages::Init() function.
```cpp
    (void)srv.on("/TEST300k.cam", HTTP_GET, 
        [](AsyncWebServerRequest* request)
        {
            request->send(SPIFFS, "/TEST300k.cam", String(), true);
        })
        .setAuthentication(webLoginUser.c_str(), webLoginPassword.c_str());
```

The task WD you are searching for is in the AsyncTcp:
https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.cpp#L188-L207

Typically you can not just change the sdkconfig.h, because the code behind is already compiled and linked as a library. IMHO the Arduino IDE uses the library, equal to platformio.

I will create a simple platformio example, for easier testing and come back again.

BlueAndi, 2021-06-06:
@ullix Can you try this one? https://github.com/BlueAndi/WebServerTest

ullix, 2021-06-08:
> Typically you can not just change the sdkconfig.h, because the code behind is already compiled and linked as a library. IMHO the Arduino IDE uses the library, equal to platformio.

To be sure of correct understanding: I made the various WDT changes in the file sdkconfig.h itself, not in any of my *.ino files! Unfortunately, Arduino has no 'recompile everything' command, but from what I see, Arduino does compile everything when some of those files gets changed. Correct me if this is wrong, but the fact that I did see responses to my changes, though not enough, suggests that my changed got compiled. 

> The task WD you are searching for is in the AsyncTcp:
> https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.cpp#L188-L207

This `CONFIG_ASYNC_TCP_USE_WDT` is only the flag to determine whether a WDT should be used or not. A WDT duration cannot be found in this lib, and I also do not see which WDT will be used. I guess it is one setting in`sdkconfig.h`, but which one?

In `AsyncTCP.h` I found this segment related to WDT and core:
```
//If core is not defined, then we are running in Arduino or PIO
#ifndef CONFIG_ASYNC_TCP_RUNNING_CORE
#define CONFIG_ASYNC_TCP_RUNNING_CORE -1 //any available core
#define CONFIG_ASYNC_TCP_USE_WDT 1 //if enabled, adds between 33us and 200us per event
#endif
```

So I tried - all changes done within file  `AsyncTCP.h`  - all 4 possible combinations of core=0, core=1, WDT=0, WDT=1 with this outcome:

- If WDT=0 (i.e. switched off) then up to a handful of downloads are possible, but then the CPU is completely blocked no matter what core is used; does not even respond to input via serial monitor; does not recover even after > 10min.
- With WDT=1 and core=0 it crashes at almost every download
- With WDT=1 and core=1 it works about the same as observed before, with a success rate of 60+%

Looks like the problems originate even before AsnycTCP?

BlueAndi, 2021-06-08:
> To be sure of correct understanding: I made the various WDT changes in the file sdkconfig.h itself, not in any of my *.ino files! Unfortunately, Arduino has no 'recompile everything' command, but [...]

If a change in the ```sdkconfig.h``` causes changes in header files, yes. If it causes changes in the source files, no. :-) It may cause strange behaviour in case it will relate to changes in headers, combined with linked libs. Platformio e.g. uses the linked libraries from https://github.com/espressif/arduino-esp32. I am not sure about how the Arduino IDE deals with it.

> This CONFIG_ASYNC_TCP_USE_WDT is only the flag to determine whether a WDT should be used or not. A WDT duration cannot be found in this lib, and I also do not see which WDT will be used. I guess it is one setting insdkconfig.h, but which one?

https://docs.espressif.com/projects/esp-idf/en/v3.3.5/api-reference/kconfig.html#config-task-wdt-timeout-s
Note, may depend on your underlying IDF version.

> If WDT=0 (i.e. switched off) then up to a handful of downloads are possible, but then the CPU is completely blocked no matter what core is used; does not even respond to input via serial monitor; does not recover even after > 10min.

I assume that its getting blocked, because queue handling blocks infinite.
https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.cpp#L106-L116

ullix, 2021-06-08:
> @ullix Can you try this one? https://github.com/BlueAndi/WebServerTest

I did try, and the results are as bad as before. 

I used 2 devices, both included as 'ESP32 Dev Module':

1. a 'ESP32-WROOM-32E' (the very latest ESP version) with 16MB, of which only 4 MB were partitioned for SPIFFS.
2. a 'ESP-WROOM-32', the oldest one I have in my box, with 4MB, all partitioned for SPIFFS.  

I did manually trigger the downloads on Chromium. The browser shows success in its bottom line like this:

[image]
Out of my first 6 downloads 3 failed with "network error", which always goes hand-in-hand with a WDT triggering a crash. The Serial Monitor shows the typical result:
```
10:56:59.411 -> [I][main.ino:271] loop(): 2 - /testData/TEST300k.cam requested (uc2).
10:56:59.411 -> E (256068) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
10:57:04.807 -> E (256068) task_wdt:  - async_tcp (CPU 0/1)
10:57:04.807 -> E (256068) task_wdt: Tasks currently running:
10:57:04.807 -> E (256068) task_wdt: CPU 0: IDLE0
10:57:04.807 -> E (256068) task_wdt: CPU 1: loopTask
10:57:04.829 -> E (256068) task_wdt: Aborting.
10:57:04.829 -> abort() was called at PC 0x40140d5c on core 0
```
Your uc2 is download in loop context, while uc1 is download in webserver context. I tried this too, it is neither better nor worse.

Instead of downloading my data file and saving on the desktop, one can also download and display the file in the browser window. As my "binary" file consists of only printable-ASCII, it does display well on the screen.

Yet, no improvement. Roughly half the downloads are successful, the other half results in a WDT crash when approx half of the file has loaded.

Any ideas?

BlueAndi, 2021-06-08:
Strange ... I used now my DevKit-v1 board:
[image]

[image]

[image]

Can you erase the whole flash and program it again?

ullix, 2021-06-08:
This is what I can offer:

[image]

Why are all speeds == 0?

I can't use PIO, what Flash erasing procedure do you have on mind? All I can do is partitioning and formatting.

BlueAndi, 2021-06-08:
IMHO you can erase the flash with Arduino IDE too. See tools menu -> Erase flash -> All
As long as we don't use the same setup, the investigation is quite hard.

ullix, 2021-06-08:
Nope. Neither Arduino 1.8.13 nor 2.0 beta7 have such a menu option for ESP32. I believe it is something Board-specific. However, few things can be simpler that erasing the Flash on an ESP32:

```
To erase the entire flash chip (all data replaced with 0xFF bytes):
      esptool.py erase_flash

```
It finds everything else on its own. I did that, and look at the outcome:

```
ullix@crucialmx200:~/Code/Arduino/main$ ./test.sh uc1 10
using file:  TEST300k.cam
1 Successful: 0,792s 0kb/s
2 Successful: 0,747s 0kb/s
3 Successful: 0,734s 0kb/s
4 Successful: 0,751s 0kb/s
5 Successful: 0,748s 0kb/s
6 Successful: 0,767s 0kb/s
7 Successful: 0,760s 0kb/s
8 Failed.
9 Successful: 0,749s 0kb/s
10 Successful: 0,722s 0kb/s
Successful downloads: 9 out of 10
Failed downloads    : 1 out of 10.
```
So, something bad in the Flash. Who would have thought; given no error message of anything. I'll look into details tomorrow.

BlueAndi, 2021-06-08:
I know that there is sometimes bad data stored, which is used by some of the underlying arduino/idf components. The wifi is such a component. Original as I started my project Pixelix, I investigated two weeks to catch a problem in wifi ap mode. The solution was easy (if you know it), just erasing the flash and disable storing wifi configuration in nvs.

ullix, 2021-06-10:
Good thing I celebrated right away, because one day later there was nothing left to celebrate :-((.

Testing the download of files 50k and 90 always resulted in 100% success.

Actually, not quite true: sometimes the test code stalled forever. But after inserting a 1sec sleep between subsequent calls, all later runs went fine. Change to your test code:
```
    fi;
    sleep 1s
done
``` 
However, the 300k file, whether via uc1 or uc2 (i.e. by regular static call or by deferred call) had a success rate of something in the low 80%. No difference between the two. An overnight run of 1000 calls had a success of 83.3%; one in five downloads failed and resulted in a WDT triggered reboot.

I then repeated the erasure of the flash via `esptool.py erase_flash` and reran the tests. No improvement. Both uc1 and uc2 came out at the low 80+%.

So, a hiccup in the Flash is not the cause for these problems.

ullix, 2021-06-12:
I have retested this with file systems SPIFFS, LittleFS, and FFat, and the outcome is basically the same in each one. Beyond a file size of 130k the frequency of WDT triggered crashes increases up to a file size of ~300k. Using deferred processing vs. static command mitigates the problems sometimes, but not always, and never heals it.

On Arduinoe IDE 1.8.13 I have now installed this exception decoder https://github.com/me-no-dev/EspExceptionDecoder
Among the many crashes I see only 2 different Backtraces:

```
Backtrace: 0x400886e8:0x3ffbf890 0x40088965:0x3ffbf8b0 0x4013e2a8:0x3ffbf8d0 0x40086e79:0x3ffbf8f0 0x401503bb:0x3ffbc220 0x4013fba3:0x3ffbc240 0x4008b125:0x3ffbc260 0x40089976:0x3ffbc280
Decoding stack results
0x400886e8: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x40088965: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x4013e2a8: task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 174
0x401503bb: esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c line 492
0x4013fba3: esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 63
0x4008b125: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x40089976: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

Backtrace: 0x400886e8:0x3ffbf900 0x40088965:0x3ffbf920 0x4013e368:0x3ffbf940 0x40086e79:0x3ffbf960 0x4015047b:0x3ffbc220 0x4013fc63:0x3ffbc240 0x4008b125:0x3ffbc260 0x40089976:0x3ffbc280
Decoding stack results
0x400886e8: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x40088965: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x4013e368: esp_task_wdt_init at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c line 206
0x4015047b: nvs::HashList::HashList() at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/nvs_flash/src/intrusive_list.h line 33
0x4013fc63: esp_deregister_freertos_idle_hook_for_cpu at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c line 113
0x4008b125: prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c line 3382
0x40089976: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```

I really don't know what conclusions to draw from it. Can somebody help?

Is there any obvious thing to look at and investigate further?

zekageri, 2021-06-12:
I have managed to drastically improve the performance of serving the files directly from LITTLEFS with deferred requests with queues.
Btw i dont use serveStatic(). 

Here is an example how i did it.

```C

// define the queue handler.
QueueHandle_t downloadQueue;

static const inline void Init_FileServing_Task(int Stack_Depth,int Core,int Priority){
    xTaskCreatePinnedToCore(fileServing_Task,"fileServing_Task",Stack_Depth,NULL,Priority,&fileServing_Task_Handle,Core);
}
void setup(){
  // First create a task just for serving the files.
  Init_FileServing_Task(6000,1,2);
  // Register the events before server begin
  downloadEvent();
}

void loop(){
  // Bunch of other function call in here...
  /* And i have like 6 or 7 task running at once. 
      One task out of the 6-7 is running on core 0 and it is a
      heavy continous modbus communication task with hardware UART  */
}

static const inline void downloadEvent(){
    server.on("/DownFile", HTTP_GET, [](AsyncWebServerRequest *request){
       // Put the request in the queue immidiately
      if( xQueueSend( downloadQueue , ( void * ) &request, ( TickType_t ) 10 ) != pdPASS ){
          // if for some reason (queue full etc..) we can't put the request to the queue, we send 500 response.
         request->send(500); // ( internal server error )
      }
    });
}

static const inline void deferredDownload_Handle(){
 // in the handle situation we proceed if we got anything in the queue.
 if( uxQueueMessagesWaiting( downloadQueue ) ){
    AsyncWebServerRequest* deferredDownload;
    if( xQueueReceive( downloadQueue, &( deferredDownload ), ( TickType_t ) 10 ) ){
      AsyncWebParameter* p = deferredDownload->getParam(0);
      String File = p->value();
      if( ( LITTLEFS.exists(File) && adminModeIsOn ) || ( LITTLEFS.exists(File) && isFileAllowed(File) ) ){
        deferredDownload->send(LITTLEFS, File, "text/plain", true);
      }else{
        AsyncWebServerResponse *response = deferredDownload->beginResponse(204, "text/plain", "No such file.");
        response->addHeader("Access-Control-Allow-Origin","*");
        deferredDownload->send(response);
      }
    }
  }
}

void fileServing_Task( void * parameter ){
    // create a queue with 10 item spaces. It is because the TCP socket can hold only 10 tcp requests at a time.
    downloadQueue = xQueueCreate( 10, sizeof( AsyncWebServerRequest* ) );
   // ever is defined as --> ( ;; )
    for ever {
        if(!Firmware_Is_Updating){
            deferredDownload_Handle();
        }
        vTaskDelay(1);
    }
}
```

This way the files served correctly. I wrote a js test sketch where i can define the file, the count of the downloads and the delay between downloads and i can select if the download request should start one after the other or to wait for the previous download response. And it seems very very stable. I did some tests with this, but not too much to be able to form a conclusion of it yet. I have no time for it now but i will test it further. For now, i got 1000 downloads and it was all success if the delay between downloads at least 30ms. The downloaded files was 10kb to 150kb. I haven't tested with bigger files yet, but i will because i will reach a point where i must serve 200 or even 300kb files so it is a must for me too.

ullix, 2021-06-12:
@zekageri I wish you good luck, but so far I am not impressed. 

With a file size of 130k I get 100% success consistently, no matter which of the two approaches (deferred or static) I take. With 150k I still get between 95% and 99% success.

The "sweet spot" for problems are file sizes around 300k. I use the 300k file as attached here: https://github.com/me-no-dev/ESPAsyncWebServer/issues/984#issuecomment-852836070 as the bad guy for testing. 

Success rate tends to become a little bit better for even bigger files. Why? No idea.

lbernstone, 2021-06-12:
If you know you have files that regularly will take longer than 5 seconds to deliver, why not just increase the wdt timeout?  It is not easy to see exactly where to apply this for serveStatic, but could certainly be done on your delayed response.
```
#include <esp_task_wdt.h>
...
esp_task_wdt_init(10,0);
// deliver the file
esp_task_wdt_init(5,1);
```

zekageri, 2021-06-13:
Its not about the wdt. The server freezes. If you turn off the wdt completely the result is the same

zekageri, 2021-06-13:
Btw iam using an esp32 wrover 8mb bare chip with psram

ullix, 2021-06-13:
@lbernstone I have already pursued the idea of lengthening or even switching-off the WDT, as I showed here:  https://github.com/me-no-dev/ESPAsyncWebServer/issues/984#issuecomment-855256610 and https://github.com/me-no-dev/ESPAsyncWebServer/issues/984#issuecomment-856509778

The result was a marginal improvement, if any at all!

However, your code suggestion is very clear and simple so I tried it in the uc2 (deferred handling in the loop) case.  The results for downloading the 300k file (the baddest guy) for a good dozen times were all over the place, with successes in 60 ... 100(!) out of 100 tries. Whenever it failed, the ESP rebooted with the usual message:
```
11:26:43.118 -> E (21774) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
11:26:48.382 -> E (21774) task_wdt:  - async_tcp (CPU 0/1)
11:26:48.415 -> E (21774) task_wdt: Tasks currently running:
11:26:48.448 -> E (21774) task_wdt: CPU 0: IDLE0
11:26:48.472 -> E (21774) task_wdt: CPU 1: loopTask
```
Given that the panic flag in `esp_task_wdt_init(10,0);` was set to zero, I am surprised that it even crashed with WDT anyway? Shouldn't that have been switched off by this command?  (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/wdts.html#_CPPv417esp_task_wdt_init8uint32_tb)

As a WDT of 10 sec did not seem enough, I increased it to 30 sec. Yet, no improvement! Increasing to 50sec, 100sec: almost every download attempt was a failure!

I have already wondered whether the length of processing triggering the WDT is the prime cause for the crash, and the WDT may be some secondary effect to quite a different problem. Maybe "the server freezes" as @zekageri suggests - well, what exactly do you mean by that? Could you put some more meat on the bones of that statement?

zekageri, 2021-06-13:
The async tcp task freezing for some reason. In my case, if i disable the wdt, my esp does not crash but the server is completely unreachable. In that case, all the other tasks are working properly.

ullix, 2021-06-14:
I continued playing with that WDT switch-off / switch-on as suggested by @lbernstone here https://github.com/me-no-dev/ESPAsyncWebServer/issues/984#issuecomment-860080923. However, no matter which condition I used, the result was always somewhere between 70% and 100% success. Unfortunately, a randomly occurring occasional 100% is of little value when your first N trials all results in a crash of your app.

On first glance this code for deferred handling seemed meaningful:
```
esp_task_wdt_init(10, 0);
gDeferredReq->send(*myFS, path.c_str(), String(), TREATMENT);   // Download or Display
esp_task_wdt_init(5, 1);
```
The duration for this code snippet is 16+/-1 millisecond irrespective of the file size being 90k or 900k. However, the total download time for such files is 200ms for 90k, and 1000ms for 900k, much longer than 16ms! Obviously, this `send()` piece of code is only putting the action into some queue, and our WDT setting doesn't impact this queue. An effect can thus not be expected.

Trying a remedy by switching the WDT permanently to:
```
esp_task_wdt_init(10, 0);
```
cannot be used, as this after the first WDT results in a completely hung CPU which goes from one WDT to the next, and does not allow anything else.

Lengthening the WDT period but keeping the panic flag set:
```
esp_task_wdt_init(10, 1);
or:
esp_task_wdt_init(30, 1);
```
has no discernible benefit over not using this command at all. For some odd reason the length set to the WDT period does not seem to matter.

I agree with @zekageri that the `async_tcp` seems to be the likely candidate for this problem. How can we get more insight into this piece of the server?

zekageri, 2021-06-14:
The async task is [in here](https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.cpp#L188). Maybe it is better to run it on the core 1? I will try it.

ullix, 2021-06-21:
Well, hold on to your Lederhosen, I'd venture to say that the whole thing has nothing to do with a bug in ESPAsncyWebServer, nor in AsyncTCP, but is due to a bug in the WiFi code of either the ESP lib, or in the code of certain routers - in my case the FritzBox 7490 - or both.

Really running out of ideas, I remembered one of my other major problems, which was that a WiFi connection to the router needed a "double-hit", i.e. you had to issue twice the `WiFi.begin()` command in order to get a single connection. This has a very lengthy discussion here, beginning more than 2 years ago: https://github.com/espressif/arduino-esp32/issues/2501

Typical in the discussion was that some people did see this problem, and some not at all. At some time I discovered that there are "Good" and "Bad" routers for this effect: Most of the time - but not exclusively - the bad routers were FritzBoxes 74XX and 75XX, while the good ones were various others, including an Android Smartphone configured as WiFi router, but also e.g an older FritzBox 7272.

These observations were all based on the ESP-core code in version 1.0.4. It seemingly got solved with ESP-core 1.0.6. With the latter a "single-hit" was sufficient for a connection even on a "bad" router. However, it came at the expense of a much longer connection time of now >2sec, while previously  the complete double-hit concluded at under 0.5sec!

On a hunch I took out my other routers to compare with my standard FritzBox 7490: One was again my Smartphone, the other the FritzBox 7272. 

And guess what: doing some 10000 downloads of my "baddest" 300k file, those two routers failed only 5 times (!), and of those 5 only 1 resulted in a WDT triggered crash and reboot, while the other 4 were timeouts. Timeouts in that low frequency are nothing unexpected; my concerns are only when they come with a crash.

How about your routers, can you confirm this observation?

zekageri, 2021-06-21:
Oh. Thats really interesting. But if no other device on the market fails with those routers you know maybe the esp is the bad here not the routers. If the handshake of the request goes out of time or something happens beetween handshakes the esp should not crash but indicate a connection close or something like that. :D

I can test with other routers. In my case i'm using ethernet with a LAN8720 chip so the connection is as stable as it can but the crash is still there. My expection decoder complains about an other task in my esp but if i uncomment that task the crash happens with the common asynctcp crash.

My router is a TP-Link device right now. And the thing is that it does not matter if i go from localhost or i want to reach the esp from outside my network. The crash happens no matter what. So i think its not the routers.

My esp is already port forwarded with a ddns server, and if i indicate a connection from a mobile network the crash happens too.

ullix, 2021-06-23:
This thing is getting weirder every minute.

My code has so far been run under Arduino 1.8.13 and Arduino 2.0.0 Beta 7. I have now converted the code so that it runs under PlatformIO as extension to VScode. 

And, surprise, the code compiled with Arduino keeps crashing, while the PIO code runs just fine! On the "bad" router FritzBox 7490! 

For now some 2000+ downloads not a single failure (timeout) let alone a crash. For the downloads I used the webserver command `server.serveStatic(uri, FS, path);` which we initially considered to be the most critical one among the serve-options (now we know it does not matter).

Here the first 20 downloads for each of the two conditions:

```
# This output generated with Arduino compiled code:
$ ./test.sh uc3 1000
using file:  TEST300k.cam
13:16:20 uc3 1 Successful: 0.485s 633784.000B/s 307200 Bytes Totals: S: 1 F:0  -->  %100.00
13:16:20 uc3 2 Successful: 0.386s 795968.000B/s 307200 Bytes Totals: S: 2 F:0  -->  %100.00
13:16:21 uc3 3 Timeout___: 19.991s 9545.000B/s 190825 Bytes Totals: S: 2 F:1  -->  %66.66
13:16:41 uc3 4 Successful: 0.476s 645073.000B/s 307200 Bytes Totals: S: 3 F:1  -->  %75.00
13:16:41 uc3 5 Successful: 0.395s 778229.000B/s 307200 Bytes Totals: S: 4 F:1  -->  %80.00
13:16:42 uc3 6 Successful: 0.398s 772468.000B/s 307200 Bytes Totals: S: 5 F:1  -->  %83.33
13:16:42 uc3 7 Timeout___: 19.992s 8324.000B/s 166413 Bytes Totals: S: 5 F:2  -->  %71.42
13:17:02 uc3 8 Successful: 0.465s 661246.000B/s 307200 Bytes Totals: S: 6 F:2  -->  %75.00
13:17:03 uc3 9 Successful: 0.404s 760224.000B/s 307200 Bytes Totals: S: 7 F:2  -->  %77.77
13:17:04 uc3 10 Successful: 0.392s 782822.000B/s 307200 Bytes Totals: S: 8 F:2  -->  %80.00
13:17:04 uc3 11 Successful: 0.390s 787163.000B/s 307200 Bytes Totals: S: 9 F:2  -->  %81.81
13:17:05 uc3 12 Successful: 0.400s 768203.000B/s 307200 Bytes Totals: S: 10 F:2  -->  %83.33
13:17:05 uc3 13 Successful: 0.408s 752421.000B/s 307200 Bytes Totals: S: 11 F:2  -->  %84.61
13:17:06 uc3 14 Successful: 0.397s 773081.000B/s 307200 Bytes Totals: S: 12 F:2  -->  %85.71
13:17:06 uc3 15 Successful: 0.435s 706749.000B/s 307200 Bytes Totals: S: 13 F:2  -->  %86.66
13:17:07 uc3 16 Timeout___: 19.991s 9258.000B/s 185081 Bytes Totals: S: 13 F:3  -->  %81.25
13:17:27 uc3 17 Successful: 0.373s 823548.000B/s 307200 Bytes Totals: S: 14 F:3  -->  %82.35
13:17:27 uc3 18 Successful: 0.451s 680544.000B/s 307200 Bytes Totals: S: 15 F:3  -->  %83.33
13:17:28 uc3 19 Successful: 0.396s 776075.000B/s 307200 Bytes Totals: S: 16 F:3  -->  %84.21
13:17:28 uc3 20 Successful: 0.408s 752987.000B/s 307200 Bytes Totals: S: 17 F:3  -->  %85.00
                            ======
Average Download Time for downloads: 0.42s (successful ones only)

#This output generated with PlatformIO code:
$ ./test.sh uc3 1000
using file:  TEST300k.cam
13:25:26 uc3 1 Successful: 0.889s 345577.000B/s 307200 Bytes Totals: S: 1 F:0  -->  %100.00
13:25:27 uc3 2 Successful: 0.898s 341997.000B/s 307200 Bytes Totals: S: 2 F:0  -->  %100.00
13:25:28 uc3 3 Successful: 1.031s 297952.000B/s 307200 Bytes Totals: S: 3 F:0  -->  %100.00
13:25:29 uc3 4 Successful: 0.917s 334955.000B/s 307200 Bytes Totals: S: 4 F:0  -->  %100.00
13:25:30 uc3 5 Successful: 0.915s 335816.000B/s 307200 Bytes Totals: S: 5 F:0  -->  %100.00
13:25:31 uc3 6 Successful: 0.899s 341661.000B/s 307200 Bytes Totals: S: 6 F:0  -->  %100.00
13:25:32 uc3 7 Successful: 0.899s 341897.000B/s 307200 Bytes Totals: S: 7 F:0  -->  %100.00
13:25:33 uc3 8 Successful: 0.891s 344684.000B/s 307200 Bytes Totals: S: 8 F:0  -->  %100.00
13:25:34 uc3 9 Successful: 0.916s 335198.000B/s 307200 Bytes Totals: S: 9 F:0  -->  %100.00
13:25:35 uc3 10 Successful: 0.936s 328221.000B/s 307200 Bytes Totals: S: 10 F:0  -->  %100.00
13:25:36 uc3 11 Successful: 0.912s 336882.000B/s 307200 Bytes Totals: S: 11 F:0  -->  %100.00
13:25:37 uc3 12 Successful: 0.896s 343041.000B/s 307200 Bytes Totals: S: 12 F:0  -->  %100.00
13:25:38 uc3 13 Successful: 0.927s 331374.000B/s 307200 Bytes Totals: S: 13 F:0  -->  %100.00
13:25:39 uc3 14 Successful: 0.869s 353447.000B/s 307200 Bytes Totals: S: 14 F:0  -->  %100.00
13:25:40 uc3 15 Successful: 0.949s 323695.000B/s 307200 Bytes Totals: S: 15 F:0  -->  %100.00
13:25:41 uc3 16 Successful: 0.893s 343818.000B/s 307200 Bytes Totals: S: 16 F:0  -->  %100.00
13:25:42 uc3 17 Successful: 0.950s 323290.000B/s 307200 Bytes Totals: S: 17 F:0  -->  %100.00
13:25:43 uc3 18 Successful: 0.878s 349813.000B/s 307200 Bytes Totals: S: 18 F:0  -->  %100.00
13:25:44 uc3 19 Successful: 0.904s 339968.000B/s 307200 Bytes Totals: S: 19 F:0  -->  %100.00
13:25:45 uc3 20 Successful: 0.908s 338242.000B/s 307200 Bytes Totals: S: 20 F:0  -->  %100.00
                            ======
Average Download Time for downloads: 0.91s (all were successful)
```

Noteworthy is that the download times for the PIO code are almost twice as long as the Arduino code (0.91s vs. 0.42s). So, something is clearly different between the two. But what? 

Both IDEs use ESP-core 1.0.6, and whenever I had a choice, I have taken only the defaults in each IDE. In particular, with my last observation of router dependence I had suspected the WiFi lib to be the problem. But this is version 1.0 in both cases.

BlueAndi, 2021-06-23:
Do they both have the same configuration (sdkconfig)? The difference in the wifi throughput sounds that there are different configurations.

zekageri, 2021-06-24:
Iam using PIO from the start with Ethernet and i got that problem from the start. So whats up with that?

ullix, 2021-06-24:
@BlueAndi Good idea. But meld tells me they are identical to the last bit! :-/

Just in case, this is the location of the two sdkconfig.h files:
```
/home/ullix/.arduino15/packages/esp32/hardware/esp32/1.0.6/tools/sdk/include/config/sdkconfig.h
/home/ullix/.platformio/packages/framework-arduinoespressif32/tools/sdk/include/config/sdkconfig.h
```
Arduino uses the 1.0.6 ESP-core version (see the path), and so does pio:
[image]

I ran the pio code overnight and just ended it with 37672 successful downloads and not a single timeout failure, let alone a crash.
```
10:01:49 uc3 37672 Successful: 0.897s 342344.000B/s 307200 Bytes Totals: S: 37672 F:0  -->  %100.00
```
Yeah, whats up with that, tell me!

zekageri, 2021-06-24:
I tested on PIO with an ESP_WROVER_32 ( 8mb ) with PSRAM on three routers ( Two not identical TPLINK routers and a Cisco router ) on localhost and with port forwarded ips trought a ddns and in every case my esp restarted if i requested a file that is largen then 120kb. The download request is goes to a queue and handled on a separate task on core 1 with or without WDT. If wdt is enabled the esp crashes, if not the asyncTCP task freezes and no request gets processed until i physically restart the module.

Ullix how is your file server function looks like? Do you just serve the file inside the callback or do you defer the request in some form?

ullix, 2021-06-25:
@zekageri I am almost happy to see that someone else has the very same struggles that I have ;-)). It also looks like you had the same ideas that I had, and also failed in the same way.

Now that I am converting to pio I face additional issues; hopefully the payback will come. But at least it seems that my latest code works reproducible on multiple ESPs even with the FritzBox7490! I'd be happy to share; is it sufficient just to zip the project folder and provide that?

> re server functions:

I am using 3 variants. It does not matter which one is being used, because either all 3 work, or all 3 fail. They are used as uc1, uc2, uc3. From my webserver code:
```
/* Download test data in different ways:
  *    Use case 1 (uc1): Download in the webserver context.
  *    Use case 2 (uc2): Download in main loop context.
  *    Use case 3 (uc3): Download from serveStatic()
  */

/* uc1   ---    */ 
    gWebServer.on("/download/uc1/*", HTTP_GET, [](AsyncWebServerRequest* request){
            auto t0 = micros();

            cSF(path, 100);
            path.printf("%s/%s", TEST_DATA_PATH, (request->url().substring(request->url().lastIndexOf("/") + 1U)).c_str());

            ++gNumberOfDownloads;
            if (nullptr != request){
                request->send(*myFS, path.c_str(), String(), TREATMENT);      // Download or Display
            }
            else{
                log_e("uc1 -- #%u - %s  REQUEST is NULLPTR!", gNumberOfDownloads, path.c_str());               
            }

            auto t1 = micros();

            float dur = (t1 - t0)/1000.0f; // takes typically 16 ms independent of files size
            log_i("uc1 -- #%u - %s  duration: %0.1f ms,  running on core: %i", gNumberOfDownloads, path.c_str(), dur, xPortGetCoreID());             
        });

/* uc2   ---   deferred action to loop! */ 
    gWebServer.on("/download/uc2/*", HTTP_GET, [](AsyncWebServerRequest* request){gDeferredReq = request;});

/* uc3   ---   serveStatic! */ 
    String uri  = "/download/uc3/TEST" + String(FILESIZE) + "k.cam";
    String path = "testData/TEST"      + String(FILESIZE) + "k.cam";
    gWebServer.serveStatic(uri.c_str(), *myFS, path.c_str());
```
uc2 is then handled in `loop()` basically the same way as uc1:
```
gDeferredReq->send(*myFS, path.c_str(), String(), TREATMENT);   // Download or Display
gDeferredReq = nullptr;

```
The only advantage of uc1 and uc2 is that one can get e.g. how long it takes, namely 16ms independent of file size. This suggests to me that the job is handed into a queue, and ESPAsyncWebserver is done with it. Thus it is immaterial which of the ucX*s is being used.

The queue may be AsyncTCP, but I guess it is WiFi.

zekageri, 2021-06-25:
I see. There is no difference on my implementation. Thats really sad. Maybe i run out of stack or something like that. Thanks for the sneak peak to your code

ullix, 2021-06-28:
Just to keep the weirdness going: it now looks like the manufacturer of the Flash has something to do with the reliability of the downloads.

I looked at my collection of ESP32s - they are all of the `ESP32Dev` type - uploaded the same code on each and counted failures-to-download, using the uc1, uc2, uc3 test cases explained above.

Before any upload I always first erased the Flash by `esptool.py erase_flash` (esptool was version 3.0). This provides this output  (device is the 3rd entry in the table below):
```
~$ esptool.py erase_flash
esptool.py v3.0
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: c8:2b:96:ba:60:e8
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 8.5s
Hard resetting via RTS pin...
```
In addition I downloaded and recorded the info about the Flash obtained by `esptool.py  flash_id`.  The Flash related lines in this case are:
```
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
```
The data are summarized in the table. "Label" is the verbatim text etched into the ESP-module metal case. The other numbers come from the esptool output. "Performance" qualifies the downloads: green with exact 100% success over several 100 up to several 10000(!) downloads; yellow with a few (2...3) failures for several 100 downloads, and red with real problems, sometimes with failures every 2nd download.

[image]
Results:
- Old ESPs (revision 1) can be as good as the latest chip (revision 3). The worst of the pack is the latest chip!
- A 4MB Flash can be as good as a 16MB Flash
- 3 different Flash manufacturer were found, coded as c8, 20, and 68
- Manufacturer c8 and 20 are green, or at least yellow. 68 is in the deep red

Googling for the manufacturers gave inconsistent results. Can anybody help?

One thing worth mentioning is the duration of chip erasing displayed in the last column. Shown are durations in sec of 3 consecutive erasures. 

For manufacturer 20 all three erasures take the same time, albeit in one case twice as long as in the other cases, despite same size Flash of 4MB in all cases. For manufacturer c8 they become shorter with each run.  For manufacturer 68 the duration is basically zero after the first run.

Obviously the ESP firmware code (bootloader?) must be different depending on the type of Flash, but I don't see this mentioned anywhere, let alone giving the option of purchasing a specific type of Flash.

[111 more replies]

closed 2022-04-16 by stale[bot]