me-no-dev/ESPAsyncWebServer #990: ESP32 restarts at a websocket event

igortomiatti, 2021-05-21:
I have a function that is called by a webserver event.

This function takes around 5600ms to be executed, which causes the reset by watch dog time.

Is there any way to call this function and execute it in the same task as the loop()?

Best regards,
Igor.

zekageri, 2021-05-22:
Let us see the event

igortomiatti, 2021-05-25:
I solved the problem using another task. Follows the event code:

Before the solution:

```
void onEvent(AsyncWebSocket *server, AsyncWebSocketClient *client, AwsEventType type, void *arg, uint8_t *data, size_t len)
{
  StaticJsonDocument<96> receive;
  StaticJsonDocument<96> send;
  char response[96];
  if (type == WS_EVT_DATA)
  {
    AwsFrameInfo *info = (AwsFrameInfo *)arg;
    if (info->final && info->index == 0 && info->len == len)
    {
      if (info->opcode == WS_TEXT)
      {
        data[len] = 0;
        deserializeJson(receive, (char *)data);
        if (receive["comando"] == "scanSSID")
        {
          uint8_t n = WiFi.scanNetworks();
          send["comando"] = "arraySSID";
          JsonArray arraySSID = send.createNestedArray("arraySSID");
          for (uint8_t i = 0; i < n; i++)
          {
               if (WiFi.SSID(i) != "")
                   arraySSID.add(WiFi.SSID(i));
          }
        serializeJson(send, response);
        }

        // update only requested socket
        ws.text(client->id(), response);
      }
    }
  }
}
```

After the solution:

```
void scanSSID(void *parameter)
{
  StaticJsonDocument<500> send;
  char json[500];
  int *id = (int *)parameter;
  uint8_t n = WiFi.scanNetworks();
  send["comando"] = "arraySSID";
  JsonArray arraySSID = send.createNestedArray("arraySSID");
  for (uint8_t i = 0; i < n; i++)
  {
    if (WiFi.SSID(i) != "")
      arraySSID.add(WiFi.SSID(i));
  }
  serializeJson(send, json);
  ws.text(*id, json);
  vPortFree( (void *)id );
  vTaskDelete(NULL);
}

void onEvent(AsyncWebSocket *server, AsyncWebSocketClient *client, AwsEventType type, void *arg, uint8_t *data, size_t len)
{
  StaticJsonDocument<96> receive;
  StaticJsonDocument<96> send;
  char response[96];
  if (type == WS_EVT_DATA)
  {
    AwsFrameInfo *info = (AwsFrameInfo *)arg;
    if (info->final && info->index == 0 && info->len == len)
    {
      if (info->opcode == WS_TEXT)
      {
        data[len] = 0;
        deserializeJson(receive, (char *)data);
        if (receive["comando"] == "scanSSID")
        {
          int *id = (int *)pvPortMalloc(sizeof(int));
          *id = client->id();
          xTaskCreatePinnedToCore(
              scanSSID,   /* Function to implement the task */
              "scanSSID",    /* Name of the task */
              10000,      /* Stack size in words */
              (void *)id, /* Task input parameter */
              20,         /* Priority of the task */
              NULL,     /* Task handle. */
              1);         /* Core where the task should run */
          return;
        }

        // update only requested socket
        ws.text(client->id(), response);
      }
    }
  }
}
```

I don't know if this is the best solution, but now it is working correctly

zekageri, 2021-05-25:
### It is a good solution. 
**You could just store the client id in a uint32 and process the request in the loop.**

### Something like this:

```C
boolean sendToLatest = false;
uint32_t Last_Socket_ID;

void loop(){
  if(sendToLatest){
    sendToLatest = false;
    if(globalClient != NULL && globalClient->status() == WS_CONNECTED){
      if (ws.count() > 0){
        ws.text(Last_Socket_ID,"Msg to send.");
      }
    }
  }
}

  void onEvent(AsyncWebSocket *server, AsyncWebSocketClient *client, AwsEventType type, void *arg, uint8_t *data, size_t len)
{
  StaticJsonDocument<96> receive;
  StaticJsonDocument<96> send;
  char response[96];
  if (type == WS_EVT_DATA)
  {
    AwsFrameInfo *info = (AwsFrameInfo *)arg;
    if (info->final && info->index == 0 && info->len == len)
    {
      if (info->opcode == WS_TEXT)
      {
        data[len] = 0;
        deserializeJson(receive, (char *)data);
        if (receive["comando"] == "scanSSID")
        {
          sendToLatest = true;
          Last_Socket_ID = client->id();
        }
      }
    }
  }
}
```

closed 2021-08-13 by stale[bot]