me-no-dev/ESPAsyncWebServer #991: Basic HTTP Authentication ?!

feraldruid22, 2021-05-23:
```
  // Route for root / web page
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/html", index_html, processor);
  });
  server.setAuthentication("user", "pass");
```

```
esp32asyncweb:104:10: error: 'class AsyncWebServer' has no member named 'setAuthentication'
   server.setAuthentication("user", "pass");
```

Why is something so simple does not work?

neutralvibes, 2021-05-25:
`  // HTTP basic authentication
  server.on("/login", HTTP_GET, [](AsyncWebServerRequest *request){
    if(!request->authenticate(http_username, http_password))
        return request->requestAuthentication();
    request->send(200, "text/plain", "Login Success!");
  });`

closed 2021-08-13 by stale[bot]

Uinden, 2023-02-17:
Whoever have this problem, I found solution.
`AsyncWebServer` does not have `setAuthentication()` function. Its `AsyncWebHandler` who has it.
`AsyncWebServer.serveStatic()` returns `AsyncWebHandler` so example works.
`AsyncWebServer.on()` will also return `AsyncWebHandler`.
So to make this work you need to use `setAuthentication()` after `on()` like:
```cpp
server.on("/", HTTP_GET, handleRoot).setAuthentication("user", "pass");
server.on("/path2", HTTP_GET, handlePath2).setAuthentication("user", "pass");
server.on("/file.html", HTTP_GET, handleFile).setAuthentication("user", "pass");
```