me-no-dev/ESPAsyncWebServer #996: Guru Meditation Error: Core 0 panic’ed (IntegerDivideByZero). Exception was unhandled

Nehessy, 2021-06-04:
Hi, I am doing a final year project which is a weather station Web with DHT22 sensor for temperature and humidity and pressure with BMP180 sensor all on the Huzzah ESP32 I need help because when a client connects to the server the monitor shows this error and the ESP32 restarts:

`Guru Meditation Error: Core 1 panic'ed (IntegerDivideByZero). Exception was unhandled.
Core 1 register dump:
PC : 0x40147ec1 PS : 0x00060830 A0 : 0x800d1430 A1 : 0x3ffcf740
A2 : 0x00000000 A3 : 0x00000000 A4 : 0x00000002 A5 : 0x0000ff00
A6 : 0x00ff0000 A7 : 0xff000000 A8 : 0x00000000 A9 : 0x3ffcf720
A10 : 0x00ffffff A11 : 0x000000f8 A12 : 0x3ffc18b0 A13 : 0x0000ff00
A14 : 0x00ff0000 A15 : 0xff000000 SAR : 0x00000005 EXCCAUSE: 0x00000006
EXCVADDR: 0x00000000 LBEG : 0x4000c2e0 LEND : 0x4000c2f6 LCOUNT : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x40147ec1:0x3ffcf740 0x400d142d:0x3ffcf760 0x400d0c7a:0x3ffcf780 0x400d0d03:0x3ffcf7b0 0x40147d4d:0x3ffcf7d0 0x400d6858:0x3ffcf7f0 0x400d6ae1:0x3ffcf880 0x400d56fa:0x3ffcf8d0 0x400d423a:0x3ffcf910 0x400d4c41:0x3ffcf950 0x400d0a4f:0x3ffcf990 0x400d6f29:0x3ffcf9d0 0x400d4e05:0x3ffcfa20 0x400d4ec9:0x3ffcfa60 0x400d5119:0x3ffcfab0 0x400d7fa5:0x3ffcfad0 0x400d8021:0x3ffcfb10 0x400d85fe:0x3ffcfb30 0x4008a02e:0x3ffcfb60

Rebooting...`

This is what the ESP exception decoder shows me: 

`PC: 0x40147ec1: Adafruit_BMP085::computeB5(int) at C:\Users\pc\Documents\Arduino\libraries\Adafruit_BMP085_Library\Adafruit_BMP085.cpp line 91
EXCVADDR: 0x00000000

Decoding stack results
0x40147ec1: Adafruit_BMP085::computeB5(int) at C:\Users\pc\Documents\Arduino\libraries\Adafruit_BMP085_Library\Adafruit_BMP085.cpp line 91
0x400d142d: Adafruit_BMP085::readPressure() at C:\Users\pc\Documents\Arduino\libraries\Adafruit_BMP085_Library\Adafruit_BMP085.cpp line 163
0x400d0c7a: readBMPPressure() at C:\Users\pc\Documents\Arduino\dht22web2/dht22web2.ino line 45
0x400d0d03: processor(String const&) at C:\Users\pc\Documents\Arduino\dht22web2/dht22web2.ino line 147
0x40147d4d: std::_Function_handler ::_M_invoke(std::_Any_data const&, String const&) at c:\users\pc\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 1857
0x400d6858: AsyncAbstractResponse::_fillBufferAndProcessTemplates(unsigned char*, unsigned int) at c:\users\pc\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d6ae1: AsyncAbstractResponse::_ack(AsyncWebServerRequest*, unsigned int, unsigned int) at C:\Users\pc\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebResponses.cpp line 315
0x400d56fa: AsyncAbstractResponse::_respond(AsyncWebServerRequest*) at C:\Users\pc\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebResponses.cpp line 261
0x400d423a: AsyncWebServerRequest::send(AsyncWebServerResponse*) at C:\Users\pc\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebRequest.cpp line 725
0x400d4c41: AsyncWebServerRequest::send_P(int, String const&, char const*, std::function ) at C:\Users\pc\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebRequest.cpp line 804
0x400d0a4f: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ) at C:\Users\pc\Documents\Arduino\dht22web2/dht22web2.ino line 171
0x400d6f29: AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at c:\users\pc\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d4e05: AsyncWebServerRequest::_parseLine() at C:\Users\pc\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebRequest.cpp line 581
0x400d4ec9: AsyncWebServerRequest::_onData(void*, unsigned int) at C:\Users\pc\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebRequest.cpp line 123
0x400d5119: std::_Function_handler   >::_M_invoke(const std::_Any_data &,  ,  ,  ,  ) at C:\Users\pc\Documents\Arduino\libraries\ESPAsyncWebServer\src\WebRequest.cpp line 76
0x400d7fa5: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at c:\users\pc\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400d8021: AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, signed char) at C:\Users\pc\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp line 1210
0x400d85fe: _async_service_task(void*) at C:\Users\pc\Documents\Arduino\libraries\AsyncTCP\src\AsyncTCP.cpp line 162
0x4008a02e: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
`

This is my Arduino codes:

`#include <Wire.h>
#include <Adafruit_BMP085.h>
#include "WiFi.h"
#include "ESPAsyncWebServer.h"
#include <Adafruit_Sensor.h>
#include <DHT.h>

Adafruit_BMP085 bmp;

const char* ssid = "iPhone de Mouya";
const char* password = "1234567890";

#define DHTPIN 27     
#define DHTTYPE    DHT22

DHT dht(DHTPIN, DHTTYPE);

AsyncWebServer server(80);

String readDHTTemperature() {
  float t = dht.readTemperature();
  if (isnan(t)) {    
    Serial.println("Failed to read from DHT sensor!");
    return "--";
[... 131 lines cut ...]

  // Route for root / web page
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/html", index_html, processor);
  });
  server.on("/temperature", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/plain", readDHTTemperature().c_str());
  });
  server.on("/humidity", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/plain", readDHTHumidity().c_str());
  });
  server.on("/pressure", HTTP_GET, [](AsyncWebServerRequest *request){
    request->send_P(200, "text/plain", readBMPPressure().c_str());
  });
 

  // Start server
  server.begin();
}
 
void loop(){
  
}`

if you can help me it is really important for my degree.

Pablo2048, 2021-06-04:
The problem is obvious IMHO - read carefully this https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember - especially **You can not use yield or delay or any function that uses them inside the callbacks** then take a look into libraries you are using. ( https://github.com/adafruit/Adafruit-BMP085-Library/blob/2653740d262571075411c052bc5ad03892f1538b/Adafruit_BMP085.cpp#L106 , etc...). 
BTW pay attention about text formatting in the github next time. Otherwise it's a hardly readable mess.

Nehessy, 2021-06-05:
> The problem is obvious IMHO - read carefully this https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember - especially **You can not use yield or delay or any function that uses [...]

Thanks for the advice. So what can I do to fix this?

Pablo2048, 2021-06-06:
The solution is simple - do all the sensor readings in the main loop lets say every 2 seconds and put the results to the global variables. Then in the callback just format these values to return strings. And no, **don't use simple delay(2000)** in the main loop between the measurements - you have to think about this as the timing is better to be done without using delay.

closed 2021-08-21 by stale[bot]