me-no-dev/ESPAsyncWebServer #997: Basic Auth not working on ESP8266 (ESP1)

mexusbg, 2021-06-04:
```
...
const char *http_username = "admin";
const char *http_password = "admin";
AsyncWebServer server(80);
...
```

In setup: 
```
...
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request)
            {
              if (!request->authenticate(http_username, http_password))
                return request->requestAuthentication();
              request->send(200, "text/plain", "OK");
              lampOn = true;
              toggle = false;
            });
   server.begin();
...
```

In POSTMAN I get error 401. The same code worked on ESP32 a month ago. Didn't test again with ESP32

mexusbg, 2021-06-04:
Looks like it defaults to Digest instead of Basic. Any fix?

feraldruid22, 2021-06-06:
I would be happy to hear that as well as I have developed an Android app connecting to this and just made the server side code ESP8266 backward compatible as well and run into the same issue.

With curl a quick fix is:

`curl -v --anyauth --connect-timeout 15 -u user:pass http://url`

for curl but as I said I would prefer using the same auth method for both instead of having to rewrite/recompile my Android app now.

feraldruid22, 2021-06-13:
Oh man this is straight out retarded.
The proper solution should be:

    if(!request->authenticate(www_username, www_password))
        return request->requestAuthentication(NULL,false);

but this seems to be broken on the ESP8266 because I get:

```
< HTTP/1.1 401 Unauthorized
< Content-Length: 0
< Connection: close
* Authentication problem. Ignoring this.
< WWW-Authenticate: Basic realm="Login Required"
< Accept-Ranges: none

```

So it is not using basic by default because its broken by default.

feraldruid22, 2021-06-29:
I have checked out the forks here:

https://github.com/me-no-dev/ESPAsyncWebServer/network

Sadly none of them fixed this.

githrdw, 2021-09-05:
>Looks like it defaults to Digest instead of Basic. Any fix?

Had to dive into the source code to find out how to enable base64/basic authentication.
Turns out you have to pass only one pararmeter, the base64 hash, so you have to encode that yourself (`username`:`password`).

https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/ESPAsyncWebServer.h#L224-L228

I think the above comment should also be in the readme

closed 2022-04-16 by stale[bot]