me-no-dev/ESPAsyncWebServer #1003: deferred file download.

zekageri, 2021-06-10:
### Hi !

**I have managed to defer some request in my code. Now i want to defer the file downloads by creating a queue for it.
All the other deferred requests are working, except the queue file download throws an error.**

**This is my approach:**

```C
QueueHandle_t downloadQueue;

void setup(){
  // Async webserver init and FS init and stuffs like that...
  downloadQueue = xQueueCreate( 10, sizeof( AsyncWebServerRequest ) );
}

void loop(){
  deferredDownload_Handle();
}

static const inline void deferredDownload_Handle(){
    if(downloadQueue == NULL)return;
    static AsyncWebServerRequest* deferredDownload  = nullptr;
    xQueueReceive(downloadQueue, &deferredDownload, portMAX_DELAY);
    AsyncWebParameter* p = request->getParam(0);

    String File = p->value(); // Get the file name, it will always starts with " / ". The client did this.

    // Some checks if we can serve the file.
    if( ( LITTLEFS.exists(File) && adminModeIsOn ) || ( LITTLEFS.exists(File) && isFileAllowed(File) ) ){
      request->send(LITTLEFS, File, "text/plain", true);
    }else{
      AsyncWebServerResponse *response = request->beginResponse(204, "text/plain", "No such file: " + File);
      response->addHeader("Access-Control-Allow-Origin","*");
      request->send(response);
    }
}
```

**Crash dump i get:**

```C
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.        
Core 1 register dump:
PC      : 0x401c030b  PS      : 0x00060e30  A0      : 0x800eb8f5  A1      : 0x3ffb1dd0    
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffe7aa0  A5      : 0x00000000    
A6      : 0x3ffe8558  A7      : 0x3ffb1f7f  A8      : 0x8009071a  A9      : 0x3ffb1d90    
A10     : 0x00000000  A11     : 0x3ffc2038  A12     : 0x00000001  A13     : 0x00000001    
A14     : 0x00060020  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c    
EXCVADDR: 0x000000a4  LBEG    : 0x4008ca80  LEND    : 0x4008ca9c  LCOUNT  : 0x00000000    

ELF file SHA256: 0000000000000000

Backtrace: 0x401c030b:0x3ffb1dd0 0x400eb8f2:0x3ffb1df0 0x40111298:0x3ffb1fb0 0x40090932:0x3ffb1fd0
  #0  0x401c030b:0x3ffb1dd0 in AsyncWebServerRequest::getParam(unsigned int) const at lib\ESPAsyncWebServer-wi-se-patches\src/WebRequest.cpp:797
  #1  0x400eb8f2:0x3ffb1df0 in loop() at src/Own_Headers/ModBus.h:290 (discriminator 2)   
      (inlined by) loop() at src/main.cpp:39 (discriminator 2)
  #2  0x40111298:0x3ffb1fb0 in loopTask(void*) at C:\Users\Pc\.platformio\packages\framework-arduinoespressif32\cores\esp32/main.cpp:23
  #3  0x40090932:0x3ffb1fd0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)   

Rebooting...
ets Jul 29 2019 12:21:46
```

zekageri, 2021-06-10:
**Okay, i have managed the download part and put all the other deferred requests in there own queue but now all the other deferred requests stopped working...**

```C
// Handle for the task
xTaskHandle fileServing_Task_Handle;

// queue handles for the different queues.
QueueHandle_t downloadQueue,loginQueue,mainPageQueue,logOutQueue;

// define the task to be able to reach everywhere in the program.
void fileServing_Task   ( void * parameter );

// define the route handling function to be able to reach everywhere in the program.
static const inline void routeHandling();

// Task init function.
static const inline void Init_FileServing_Task(int Stack_Depth,int Core,int Priority){
    xTaskCreatePinnedToCore(fileServing_Task,"fileServing_Task",Stack_Depth,NULL,Priority,&fileServing_Task_Handle,Core);
}

void setup(){
 // add routes before server begin
 routeHandling();

 // Init a task for all the deferred requests.
  Init_FileServing_Task(6000,1,2);
}

void loop(){
 // other things inside...
}

static const inline void deferredLogOut_Handle(){
    if(logOutQueue != NULL){
        AsyncWebServerRequest* deferredLogOut;
        xQueueReceive(logOutQueue, &deferredLogOut, portMAX_DELAY);
       
        // Process the request inside " deferredLogOut "
    }
}

static const inline void deferredMainPage_Handle(){
    if(mainPageQueue != NULL){
        AsyncWebServerRequest* deferredMainPage;
        xQueueReceive(mainPageQueue, &deferredMainPage, portMAX_DELAY);
       
       // Process the request inside " deferredMainPage"
    }
}

static const inline void deferredLogin_Handle(){
    if(loginQueue != NULL){
        AsyncWebServerRequest* deferredLogin;
        xQueueReceive(loginQueue, &deferredLogin, portMAX_DELAY);

       // Process the request inside " deferredLogin"
    }
}

static const inline void routeHandling(){

    server.on("/logOut", HTTP_GET, [](AsyncWebServerRequest *request) {
       // Put the request in the queue
        xQueueSend(logOutQueue, &request, portMAX_DELAY);
    });

    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
       // Put the request in the queue
        xQueueSend(mainPageQueue, &request, portMAX_DELAY);
    });

    server.on("/RemoteLogin", HTTP_POST, [](AsyncWebServerRequest *request){
       // Put the request in the queue
        xQueueSend(loginQueue, &request, portMAX_DELAY);
    });
   
    server.on("/DownFile", HTTP_GET, [](AsyncWebServerRequest *request){
       // Put the request in the queue
      xQueueSend(downloadQueue, &request, portMAX_DELAY);
    });
}

static const inline void deferredDownload_Handle(){
  if(downloadQueue != NULL){
    AsyncWebServerRequest* deferredDownload;
    xQueueReceive(downloadQueue, &deferredDownload, portMAX_DELAY);
    AsyncWebParameter* p = deferredDownload->getParam(0);
    String File = p->value();
    if( ( LITTLEFS.exists(File) && adminModeIsOn ) || ( LITTLEFS.exists(File) && isFileAllowed(File) ) ){
      deferredDownload->send(LITTLEFS, File, "text/plain", true);
    }else{
      AsyncWebServerResponse *response = deferredDownload->beginResponse(204, "text/plain", "Nincs ilyen file " + File);
      response->addHeader("Access-Control-Allow-Origin","*");
      deferredDownload->send(response);
    }
  }
}

// The actual file serving task for the deferred requests.

void fileServing_Task( void * parameter ){

    // Create the actual queues for the requests
    downloadQueue = xQueueCreate( 10, sizeof( AsyncWebServerRequest ) );
    loginQueue    = xQueueCreate( 10, sizeof( AsyncWebServerRequest ) );
    mainPageQueue = xQueueCreate( 10, sizeof( AsyncWebServerRequest ) );
    logOutQueue   = xQueueCreate( 10, sizeof( AsyncWebServerRequest ) );
    
    for ever {
       // check every queue one by one. If some of them is not null, process it.
        if(!Firmware_Is_Updating){
            deferredLogin_Handle();
            deferredMainPage_Handle();
            deferredLogOut_Handle();
            deferredDownload_Handle();
        }
        vTaskDelay(1);
    }
}

```

zekageri, 2021-06-10:
It turns out that if i use queues, the download part is sending empty responses for some reason.

zekageri, 2021-06-10:
I don't get it. The functions can't receive the requests inside the queues.

```C
    server.on("/logOut", HTTP_GET, [](AsyncWebServerRequest *request) {
        Serial.println("Put logout to queue");
        xQueueSend(logOutQueue, &request, portMAX_DELAY);
    });

    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
        Serial.println("Put main to queue");
        xQueueSend(mainPageQueue, &request, portMAX_DELAY);
    });

    server.on("/RemoteLogin", HTTP_POST, [](AsyncWebServerRequest *request){
        Serial.println("Put login to queue");
        xQueueSend(loginQueue, &request, portMAX_DELAY);
    });

static const inline void deferredLogOut_Handle(){
    if(logOutQueue != NULL){
        Serial.println("Receiving logout...");
        AsyncWebServerRequest* deferredLogOut;
        xQueueReceive(logOutQueue, &deferredLogOut, portMAX_DELAY);
       
        // Process the request inside " deferredLogOut "
    }
}

static const inline void deferredMainPage_Handle(){
    if(mainPageQueue != NULL){
        Serial.println("Receiving main...");
        AsyncWebServerRequest* deferredMainPage;
        xQueueReceive(mainPageQueue, &deferredMainPage, portMAX_DELAY);
       
       // Process the request inside " deferredMainPage"
    }
}

static const inline void deferredLogin_Handle(){
    if(loginQueue != NULL){
        Serial.println("Receiving login...");
        AsyncWebServerRequest* deferredLogin;
        xQueueReceive(loginQueue, &deferredLogin, portMAX_DELAY);

       // Process the request inside " deferredLogin"
    }
}

```

**I can see on the serial that the handlers put the requests inside the queue...
Serial:**

```C
Put download to queue
[W][AsyncTCP.cpp:982] _poll(): pcb is NULL
[W][AsyncTCP.cpp:982] _poll(): pcb is NULL
[W][AsyncTCP.cpp:982] _poll(): pcb is NULL
[W][AsyncTCP.cpp:982] _poll(): pcb is NULL
[W][AsyncTCP.cpp:1002] _poll(): rx timeout 4
Put download to queue
[W][AsyncTCP.cpp:1002] _poll(): rx timeout 4
Put main to queue
[W][AsyncTCP.cpp:1002] _poll(): rx timeout 4
[W][AsyncTCP.cpp:1002] _poll(): rx timeout 4
Put main to queue
[W][AsyncTCP.cpp:1002] _poll(): rx timeout 4
Put main to queue
[W][AsyncTCP.cpp:1002] _poll(): rx timeout 4
[W][AsyncTCP.cpp:1002] _poll(): rx timeout 4
Put main to queue
[W][AsyncTCP.cpp:1002] _poll(): rx timeout 4
Put main to queue
[W][AsyncTCP.cpp:1002] _poll(): rx timeout 4
[W][AsyncTCP.cpp:1002] _poll(): rx timeout 4
```

Pablo2048, 2021-06-10:
Just few thoughts:
1. do not store whole AsyncWebServerRequest object in the queue, but just the pointer (actually you are using COPY of the object, not the original one)
2. do not use another task for file serving - at least try to use loop() first, because LittleFS/SPIFFS is not thread safe IMHO

zekageri, 2021-06-10:
Like this?

```C
  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
        Serial.println("Put main to queue");
       // Request will become a pointer?
        if( xQueueSend( mainPageQueue, ( void * ) &request, ( TickType_t ) 10 ) != pdPASS ){
          Serial.println("Problem sending main to queue");
        }
    });
```

Pablo2048, 2021-06-10:
I see that the queues are created this way:
```
    // Create the actual queues for the requests
    downloadQueue = xQueueCreate( 10, sizeof( AsyncWebServerRequest ) );
    loginQueue    = xQueueCreate( 10, sizeof( AsyncWebServerRequest ) );
    mainPageQueue = xQueueCreate( 10, sizeof( AsyncWebServerRequest ) );
    logOutQueue   = xQueueCreate( 10, sizeof( AsyncWebServerRequest ) );
```
so sizeof(AsyncWebServerRequest), not the size of the pointer. Then yes - use void * ( more info https://www.freertos.org/a00117.html ). Difference between new task and 'Arduino' loop task is IMHO that the filesystem object is created within arduino task context so there can not be any race condition acessing the filesystem.

zekageri, 2021-06-10:
### I see. 
**So i must create the queues like this:**

```C
// Note the star after the AsyncWebServerRequest
downloadQueue = xQueueCreate( 10, sizeof( AsyncWebServerRequest* ) );
```

**Push to the quques like this:**

```C
 if( xQueueSend( loginQueue, ( void * ) &request, ( TickType_t ) 10 ) != pdPASS ){
   Serial.println("Problem sending logout to queue");
   // dont know what to do in here. maybe some error logs or something?
 }
```

**And receive like this:**

```C
if(loginQueue != NULL){
  AsyncWebServerRequest* deferredLogin;
  if( xQueueReceive( loginQueue, &( deferredLogin ), ( TickType_t ) 10 ) ){
    // Process request inside deferredLogin
  }
}
```

Pablo2048, 2021-06-10:
Exactly, give it a try and report back if that was a reason...

Pablo2048, 2021-06-10:
One more thing:
```cpp
 if( xQueueSend( loginQueue, ( void * ) &request, ( TickType_t ) 10 ) != pdPASS ){
   Serial.println("Problem sending logout to queue");
   // dont know what to do in here. maybe some error logs or something?
 }
```
In case of failure putting the pointer to the queue You can just return error 500 (internal server error) by using request->send(500);

zekageri, 2021-06-10:
**Thank you for your help.** 

The `request->send(500);` on fail was a really good idea.

It is working like this. 
But i kept the `fileServing_Task( void * parameter )` and it seems stable and surprisingly fast.
The reason was that i think i have no more stack space in the loop task.

**If my new task is on core 1 doesn't that mean that it is still inside the context on the loop task since it is the same core?**

### In case somebody needs some delicacy from these things i put some of the working code here:

```C

static const inline void Init_FileServing_Task(int Stack_Depth,int Core,int Priority){
    xTaskCreatePinnedToCore(fileServing_Task,"fileServing_Task",Stack_Depth,NULL,Priority,&fileServing_Task_Handle,Core);
}
void setup(){
}

void loop(){
// Bunch of other function call in here...
}

static const inline void deferredDownload_Handle(){
  if(downloadQueue != NULL){
    AsyncWebServerRequest* deferredDownload;
    if( xQueueReceive( downloadQueue, &( deferredDownload ), ( TickType_t ) 10 ) ){
      AsyncWebParameter* p = deferredDownload->getParam(0);
      String File = p->value();
      if( ( LITTLEFS.exists(File) && adminModeIsOn ) || ( LITTLEFS.exists(File) && isFileAllowed(File) ) ){
        deferredDownload->send(LITTLEFS, File, "text/plain", true);
      }else{
        AsyncWebServerResponse *response = deferredDownload->beginResponse(204, "text/plain", "Nincs ilyen file " + File);
        response->addHeader("Access-Control-Allow-Origin","*");
        deferredDownload->send(response);
      }
    }
  }
}

static const inline void deferredScript_Handle(){
   // This is a really big script file. Like 200-300kb minimized and gzipped. Contains bootstrap, jquery and own js codes.

    if(scriptQueue != NULL){
        AsyncWebServerRequest* deferredScript;
        if( xQueueReceive( scriptQueue, &( deferredScript ), ( TickType_t ) 10 ) ){
            AsyncWebServerResponse* response = deferredScript->beginResponse(LITTLEFS, "/BIG_SCRIPT.bundle.min.js", "text/javascript");
            response->addHeader("Content-Encoding", "gzip");
            deferredScript->send(response);
        }
    }
}

void fileServing_Task( void * parameter ){
    downloadQueue = xQueueCreate( 30, sizeof( AsyncWebServerRequest* ) );
    loginQueue    = xQueueCreate( 10, sizeof( AsyncWebServerRequest* ) );
    mainPageQueue = xQueueCreate( 10, sizeof( AsyncWebServerRequest* ) );
    logOutQueue   = xQueueCreate( 10, sizeof( AsyncWebServerRequest* ) );
    scriptQueue   = xQueueCreate( 10, sizeof( AsyncWebServerRequest* ) );
    for ever {
        if(!Firmware_Is_Updating){
            deferredDownload_Handle();
            deferredScript_Handle();
            deferredLogin_Handle();
            deferredMainPage_Handle();
            deferredLogOut_Handle();
        }
        vTaskDelay(1);
    }
}
```

Pablo2048, 2021-06-10:
Glad to hear that. Just few more suggestions ;-) :
1. there is no need to create 30 entries in the download queue - limit is the maximal number of open TCP sockets (I think that it is 10 for ESP32 and 5 for ESP8266)
2. Your checks if (downloadQueue != NULL) would be always true, because xQueueCreate allocate the memory. If You want to check if there is something in the queue use https://www.freertos.org/a00018.html#ucQueueMessagesWaiting .

zekageri, 2021-06-10:
> Glad to hear that. Just few more suggestions ;-) :
>
> 1. there is no need to create 30 entries in the download queue - limit is the maximal number of open TCP sockets (I think that it is 10 for ESP32 [...]

**Oh god. Thank for the `if( uxQueueMessagesWaiting( downloadQueue ) ){ }` link.** 

_The reason for the 30 entrie in the download queue is because the client can request more files at once.
(at least a few ms after each other ). And if there are more clients connections at once, i need the queue.
 I got a jquery doc onload handler on the client and if the doc loaded the client will request multiple files from the filesystem with ajax. And it will download them._ 

**It looks like this:**

```JS
jQuery(function() {

 /* connecting to websocket. ( its bugged and causes heap fragmentation or something similar on the server side because it is 
  not releasing the clients on DC, requires more investigation ) */
 WebSocketCn();

 // requesting a pretty large file and triggering the download route on the "server"
 getHwConf(); 
   
    // other huge file download
    getSavedUserProgram();
    // other huge file download
    getLanguage();

   // etc... etc...
}
```

Pablo2048, 2021-06-10:
:-) well, the problem is that the webserver itself can not create more than maximum available TCP sockets's AsyncWebServerRequests so...

zekageri, 2021-06-10:
I see. But tell me if my understanding is wrong.

Client 1 -> request x 5 -> queue ( processing ) -> releasing TCP socket
Client 2 -> request x5 -> Client1 requests still in queue, still processing. Client 2 request goes to queue -> releasing TCP socket for Client 2

Client 3 -> request x5 -> Client 1 request probably ended on the queue, client 2 still there , pushing Client 3 requiest to queue ->  releasing Client 3 TCP socket

Pablo2048, 2021-06-10:
There can be up to maximum number of TCP sockets AsyncWebServerRequests. Never more. After closing one TCP socket, after TCP_FIN_TIMEOUT the socket is released and available for new request. You can never have more than system limited TCP sockets AsyncWebServerRequests at one time. Just try it by yourelf - create web page with lets say 20 very small images inside, put it on the ESP32, direct browser to this page and use F12 (developers tools) and watch the requests and responses.

zekageri, 2021-06-10:
I see. I assume that it is becaue of the limited RAM. 
Can i force it to use external ram for the socket connections somehow?

Pablo2048, 2021-06-10:
Unfortunately not I'm afraid. The limit is:
https://github.com/espressif/arduino-esp32/blob/5ae3e836f9de6f3e850f8ead43e9e357475121af/tools/sdk/esp32/include/config/sdkconfig.h#L350

zekageri, 2021-06-10:
hmmm.  Strange that we can not do something about that tcp socket limit.

Thank you for all of your help. I really really appreciate it.
  
I close this issue for now.

closed 2021-06-10 by zekageri