me-no-dev/ESPAsyncWebServer #1006: Send 404 code and custom 404.html page with SPIFFS

MattiaPaletti, 2021-06-17:
How do i send both the 404 response code and the 404.html page via SPIFFS?

I have tried this without success:

`  server.onNotFound([](AsyncWebServerRequest *request){
    AsyncWebServerResponse *response = request->beginResponse(404); 
    response->addHeader("Location", "/404.html");
    response->addHeader("Cache-Control", "no-cache");
    request->send(response);
  });  
  server.on("/404.html", HTTP_GET, [](AsyncWebServerRequest *request){
      request->send(SPIFFS, "/404.html");
  });`

By doing so I only get the 404 code but not the custom 404 page, can someone help me?

Thanks in advance.

zekageri, 2021-06-17:
First of all. You should use [LITTLEFS by lorol](https://github.com/lorol/LITTLEFS) for esp projects. It is far better than SPIFFS and the SPIFFS is deprecated.

**Here is an example callback:**

```C
server.onNotFound([](AsyncWebServerRequest *request){
    AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/NotFound.html", "text/html");
    //response->addHeader("Content-Encoding", "gzip"); // --> uncomment if your file is GZIPPED 
    request->send(response);
});
```

MattiaPaletti, 2021-06-17:
By now I have done everything with SPIFFS and the rest works, the problem is how to send both the 404 code and the 404.html page

MattiaPaletti, 2021-06-17:
Yes, but in that way I see page 404 but the response code remains 200 and not 404

Pablo2048, 2021-06-17:
You can always set return code by calling `setCode()`, so the solution probably is to use zekageri's code and add `response->setCode(404);` just before `request->send(...)`.

zekageri, 2021-06-17:
**Oh i see. You are using some kind of an AJAX request and client side and want to distinguish the response in the callback?**

In that case as `Pablo2048 `said:

```C
server.onNotFound([](AsyncWebServerRequest *request){
    AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/NotFound.html", "text/html");
    response->setCode(404);
    request->send(response);
});
```

MattiaPaletti, 2021-06-17:
Thank you! `response->setCode(404);` is the solution!
The custom 404 page is shown and also the 404 error code, perfect.

> **Oh i see. You are using some kind of an AJAX request at client side and want to distinguish the response in the callback?**
> 
> In that case as `Pablo2048` said:
> 
> ```c
> server.onNotFound([](AsyncWebServerRequest *request){
>     AsyncWebServerResponse* response = request->beginResponse(LITTLEFS, "/NotFound.html", "text/html");
>     response->setCode(404);
>     request->send(response);
> });
> ```

closed 2021-06-17 by MattiaPaletti