me-no-dev/ESPAsyncWebServer #1007: Reboot esp32   E (33518) task_wdt:  - async_tcp (CPU 0/1)

rpavluchenko, 2021-06-19:
ESPAsyncWebServer serves http requests, responses are either SPIFFS files or json generated by a function.
At the first boot on any client (desktop or phone, different browsers), esp32 is reloaded with the following error:

E (33518) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (33518) task_wdt:  - async_tcp (CPU 0/1)
E (33518) task_wdt: Tasks currently running:
E (33518) task_wdt: CPU 0: IDLE0
E (33518) task_wdt: CPU 1: loopTask
E (33518) task_wdt: Aborting.
abort() was called at PC 0x4014dfcc on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40088a44:0x3ffbf9d0 0x40088cc1:0x3ffbf9f0 0x4014dfcc:0x3ffbfa10 0x40087159:0x3ffbfa30 0x40164f9b:0x3ffbc260 0x4014f8d7:0x3ffbc280 0x4008b481:0x3ffbc2a0 0x40089cd2:0x3ffbc2c0

Vscode / platformio / environment, framework - arduino
Libraries are up-to-date.

```
lib_deps =  bblanchon/ArduinoJson@^5.13.4
			https://github.com/me-no-dev/ESPAsyncWebServer.git
			https://github.com/me-no-dev/AsyncTCP.git
```

http.h
```
#include <Arduino.h>
//#include <AsyncTCP.h>  //Plugs into ESPAsyncWebServer.h
#include <ArduinoJson.h>
#include <AsyncJson.h>
#include <ESPAsyncWebServer.h>
```
http.cpp
```
#include "http.h"
#include <ArduinoJson.h>
#include "WString.h"

void HTTP_init(void)
{
    server.on("/configweb.json", HTTP_GET, handle_ConfigJSON); 
    server.on("/send", HTTP_GET, handle_request_parcer);
    server.begin();
    Serial.println("HTTP init");
}
```

The handle_request_parcer function gets a parameter and a value and sets some variables
```
void handle_request_parcer(AsyncWebServerRequest *request)
{
    int params = request->params();
    for (int i = 0; i < params; i++)
    {
        AsyncWebParameter *p = request->getParam(i);
        data_parcer(p->name().c_str(), p->value().c_str());
    }
    request->send(200, "text/plain", "OK");
}
```
It actually takes requests and processes them. I don't see a problem here, but maybe there is some kind of dependency that I don't understand.

Main part that creates problems ...

file_system.h
```
#ifndef file_system_h
#define file_system_h

#include "HTTP.h"
//#include <AsyncTCP.h>
#include "ESPAsyncWebServer.h"
#include <FS.h>
#include <SPIFFS.h>

void FS_init(void);
String getContentType(String filename);
bool handleFileRead(String path);
void listFilesInDir(File dir, int numTabs);

#endif
```

file_system.cpp
```
#include "file_system.h"

void FS_init(void) {
  bool result = SPIFFS.begin(false,"/spiffs",30);
  Serial.print("SPIFFS opened: ");
  Serial.println(result);

  SPIFFS.remove("/configweb.json");  // So it is necessary:)
  SPIFFS.remove("/wifi.json"); // So it is necessary:)

  {
    File root = SPIFFS.open("/");
    listFilesInDir(root, 1);
    Serial.println ("FS init");
  }
  

  server.serveStatic("/", SPIFFS, "/");
  // server.serveStatic ( " /bootstrap.min.css.gz " , SPIFFS, " /bootstrap.min.css.gz" );  
  // server.serveStatic ( " /bootstrap.min.js.gz " , SPIFFS, " /bootstrap.min.js.gz" );  
  // server.serveStatic ( " /bootstrap.min.js.map.gz " , SPIFFS, " /bootstrap.min.js.map.gz" );  
  // server.serveStatic ( " /jquery.min.js.gz" , SPIFFS, " /jquery.min.js.gz" );  
  // server.serveStatic ( " /fontello.css " , SPIFFS, " /fontello.css" );  
  // server.serveStatic ( " /fontello.eot " , SPIFFS, " /fontello.eot" );  
  // server.serveStatic ( " /fontello.svg " , SPIFFS, " /fontello.svg" );  
  // server.serveStatic ( " /fontello.ttf " , SPIFFS, " /fontello.ttf" );  
  // server.serveStatic ( " /fontello.woff" , SPIFFS, " /fontello.woff" );  
  // server.serveStatic ( " /fontello.woff2" , SPIFFS, " /fontello.woff2" );  
  // server.serveStatic ( " /oswald300.ttf" , SPIFFS, " /oswald300.ttf" );  
  // server.serveStatic ( " /oswald300.woff" , SPIFFS, " /oswald300.woff" );  
  // server.serveStatic ( " /oswald300.woff2" , SPIFFS, " /oswald300.woff2" );  
  // server.serveStatic ( " /style.css" , SPIFFS, " /style.css" );  

  server.onNotFound([](AsyncWebServerRequest *request){
    Serial.printf("NOT_FOUND: ");
    if(request->method() == HTTP_GET) Serial.printf("GET");
    else if(request->method() == HTTP_POST) Serial.printf("POST");
    else if(request->method() == HTTP_DELETE) Serial.printf("DELETE");
    else if(request->method() == HTTP_PUT) Serial.printf("PUT");
    else if(request->method() == HTTP_PATCH) Serial.printf("PATCH");
    else if(request->method() == HTTP_HEAD) Serial.printf("HEAD");
    else if(request->method() == HTTP_OPTIONS) Serial.printf("OPTIONS");
    else Serial.printf("UNKNOWN");
    Serial.printf(" http://%s%s\n", request->host().c_str(), request->url().c_str());
    if(request->contentLength()){
      Serial.printf("_CONTENT_TYPE: %s\n", request->contentType().c_str());
      Serial.printf("_CONTENT_LENGTH: %u\n", request->contentLength());
    }

    int headers = request->headers();
    int i;
    for(i=0;i<headers;i++){
      AsyncWebHeader* h = request->getHeader(i);
      Serial.printf("_HEADER[%s]: %s\n", h->name().c_str(), h->value().c_str());
    }

    int params = request->params();
    for(i=0;i<params;i++){
      AsyncWebParameter* p = request->getParam(i);
      if(p->isFile()){
        Serial.printf("_FILE[%s]: %s, size: %u\n", p->name().c_str(), p->value().c_str(), p->size());
      } else if(p->isPost()){
        Serial.printf("_POST[%s]: %s\n", p->name().c_str(), p->value().c_str());
      } else {
        Serial.printf("_GET[%s]: %s\n", p->name().c_str(), p->value().c_str());
      }
    }
    request->send(404);
  });
}

void listFilesInDir(File dir, int numTabs) 
{
  while (true) {
 
    File entry =  dir.openNextFile();
    if (! entry) {
      // no more files in the folder
      break;
    }
    for (uint8_t i = 0; i < numTabs; i++) {
      Serial.print('\t');
    }
    Serial.print(entry.name());
    if (entry.isDirectory()) {
      Serial.println("/");
      listFilesInDir(entry, numTabs + 1);
    } else {
      // display zise for file, nothing for directory
      Serial.print("\t\t");
      Serial.println(entry.size(), DEC);
    }
    entry.close();
  }
}
```

Subjectively, using parameters bool result = SPIFFS.begin (false, "/ spiffs", 30); made the work more stable, now I see reboots only if the browser cache is cleared and this is the first attempt to get data, in other cases everything works very stable.
But my application is designed in such a way that at the first boot you can (and should) get a PWA, and then use it, but getting it is just problematic!

I have read everything I can go through about this problem, but have not found or understood a solution.
I ask for help!

rpavluchenko, 2021-06-20:
While I was waiting for an answer, I continued experimenting and read that Backtrace decoder is implemented in the platform https://docs.platformio.org/en/latest/core/userguide/device/cmd_monitor.html#filters, I tried it in ini:
```
monitor_filters = log2file, default
```
and could not get an error in either of the previously described situations. Those. simply by specifying monitor_filters = log2file, default I cannot get an error, what this may be connected with is, of course, incomprehensible, and most importantly, this does not remove the main problem, but the behavior seems strange to me. Although my level of understanding the processes is more than amateur.

rpavluchenko, 2021-06-20:
Unfortunately, I don’t know what to do in this situation.
```
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
[W][AsyncTCP.cpp:949] _poll(): pcb is NULL
E (379149) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (379149) task_wdt:  - async_tcp (CPU 0/1)
E (379149) task_wdt: Tasks currently running:
E (379149) task_wdt: CPU 0: IDLE0
E (379149) task_wdt: CPU 1: loopTask
E (379149) task_wdt: Aborting.
abort() was called at PC 0x4014da74 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x40088a44:0x3ffbf9d0 0x40088cc1:0x3ffbf9f0 0x4014da74:0x3ffbfa10 0x40087159:0x3ffbfa30 0x40164a43:0x3ffbc260 0x4014f37f:0x3ffbc280 0x4008b481:0x3ffbc2a0 0x40089cd2:0x3ffbc2c0
  #0  0x40088a44:0x3ffbf9d0 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40088cc1:0x3ffbf9f0 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x4014da74:0x3ffbfa10 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x40087159:0x3ffbfa30 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x40164a43:0x3ffbc260 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
  #5  0x4014f37f:0x3ffbc280 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #6  0x4008b481:0x3ffbc2a0 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #7  0x40089cd2:0x3ffbc2c0 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
  ```

BbIKTOP, 2021-06-21:
After a few requests, it hangs here:
```c++
static esp_err_t _tcp_write(tcp_pcb * pcb, int8_t closed_slot, const char* data, size_t size, uint8_t apiflags) {
    if(!pcb){
        return ERR_CONN;
    }
    tcp_api_call_t msg;
    msg.pcb = pcb;
    msg.closed_slot = closed_slot;
    msg.write.data = data;
    msg.write.size = size;
    msg.write.apiflags = apiflags;
    tcpip_api_call(_tcp_write_api, (struct tcpip_api_call_data*)&msg);
    return msg.err;
}
```
``_tcp_write_api`` never called and everything just hangs

BbIKTOP, 2021-06-21:
Added some debug printfs:
```

Requested file /index.js
New FileAsyncChunkedResponse for [/index.js]
AsyncAbstractResponse::_ack: 1
AsyncAbstractResponse::_fillBufferAndProcessTemplates
/index.js (3FFD1BA0)) Chunk read, len=5618, index=0, buffer=1073575900 
/index.js (3FFD1BA0)) Chunk read 2, len=5618, index=0, buffer=1073575900 
/index.js (3FFD1BA0)) Chunk read 3, len=5618, index=0, buffer=1073575900 
AsyncChunkedResponse::_fillBuffer: got 3073 of requested 5618 bytes.
AsyncAbstractResponse::_fillBufferAndProcessTemplates: read 3073 bytes
AsyncAbstractResponse::_ack: 2
AsyncAbstractResponse::_ack: 3, outlen=3199, readlen=3073, headlen=118
AsyncAbstractResponse::_ack: 4
AsyncClient::write: 1
AsyncClient::add: _tcp_write: 3199
_tcp_write 1
_tcp_write_api 1
_tcp_write_api 2
_tcp_write_api 3
_tcp_write_api 4
_tcp_write_api 5
_tcp_write 2
AsyncClient::write: 2
AsyncAbstractResponse::_ack: 5

Requested file /tabs.js
New FileAsyncChunkedResponse for [/tabs.js]
AsyncAbstractResponse::_ack: 1
AsyncAbstractResponse::_fillBufferAndProcessTemplates
/tabs.js (3FFD59E4)) Chunk read, len=5618, index=0, buffer=1073575900 
/tabs.js (3FFD59E4)) Chunk read 2, len=5618, index=0, buffer=1073575900 
/tabs.js (3FFD59E4)) Chunk read 3, len=5618, index=0, buffer=1073575900 
AsyncChunkedResponse::_fillBuffer: got 1661 of requested 5618 bytes.
AsyncAbstractResponse::_fillBufferAndProcessTemplates: read 1661 bytes
AsyncAbstractResponse::_ack: 2
AsyncAbstractResponse::_ack: 3, outlen=1787, readlen=1661, headlen=118
AsyncAbstractResponse::_ack: 4
AsyncClient::write: 1
AsyncClient::add: _tcp_write: 1787
_tcp_write 1
_tcp_write_api 1
_tcp_write_api 2
_tcp_write_api 3
_tcp_write_api 4
_tcp_write_api 5
_tcp_write 2
AsyncClient::write: 2
AsyncAbstractResponse::_ack: 5

Requested file /drag.js
New FileAsyncChunkedResponse for [/drag.js]
AsyncAbstractResponse::_ack: 1
AsyncAbstractResponse::_fillBufferAndProcessTemplates
/drag.js (3FFD9F20)) Chunk read, len=5618, index=0, buffer=1073575900 
/drag.js (3FFD9F20)) Chunk read 2, len=5618, index=0, buffer=1073575900 
/drag.js (3FFD9F20)) Chunk read 3, len=5618, index=0, buffer=1073575900 
AsyncChunkedResponse::_fillBuffer: got 5618 of requested 5618 bytes.
AsyncAbstractResponse::_fillBufferAndProcessTemplates: read 5618 bytes
AsyncAbstractResponse::_ack: 2
AsyncAbstractResponse::_ack: 3, outlen=5744, readlen=5618, headlen=118
AsyncAbstractResponse::_ack: 4
AsyncClient::write: 1
AsyncClient::add: _tcp_write: 5744
_tcp_write 1
E (13010) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (13010) task_wdt:  - async_tcp (CPU 0/1)
E (13010) task_wdt: Tasks currently running:
E (13010) task_wdt: CPU 0: IDLE0
E (13010) task_wdt: CPU 1: IDLE1
E (13010) task_wdt: Aborting.
abort() was called at PC 0x40142ac0 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x400886dc:0x3ffbfa20 0x40088959:0x3ffbfa40 0x40142ac0:0x3ffbfa60 0x40086e6d:0x3ffbfa80 0x40176323:0x3ffbc220 0x401443bb:0x3ffbc240 0x4008b119:0x3ffbc260 0x4008996a:0x3ffbc280
  #0  0x400886dc:0x3ffbfa20 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40088959:0x3ffbfa40 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x40142ac0:0x3ffbfa60 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x40086e6d:0x3ffbfa80 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x40176323:0x3ffbc220 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
  #5  0x401443bb:0x3ffbc240 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #6  0x4008b119:0x3ffbc260 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #7  0x4008996a:0x3ffbc280 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
```

BbIKTOP, 2021-06-22:
Ok, together we've got a simpler test case:
```c++
#include <Arduino.h>
#include <LITTLEFS.h>
#include <WiFi.h>

#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

#include <wifi_creds.h>

extern void sendData(AsyncWebServerRequest *request);

AsyncWebServer webServer(80);

const char *ssid = WIFI_SSID;
const char *password = WIFI_PASSWD;

void notFound(AsyncWebServerRequest *request)
{
  request->send(404, "text/plain", "Not found");
}

void setup()
{
  Serial.begin(115200);
  Serial.printf("\n\nStarting...\n");

  LITTLEFS.begin();

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {

    Serial.printf(".");
    delay(100);
  }

  Serial.print("\nIP Address: ");
  Serial.println(WiFi.localIP());

  // webServer.on("/", sendData);
  // webServer.on("^\\/.*\\.html$", sendData);
  // webServer.on("^\\/.*\\.js$", sendData);
  // webServer.on("^\\/.*\\.ttf$", sendData);
  // webServer.on("^\\/.*\\.ico$", sendData);
  // webServer.on("^\\/.*\\.png$", sendData);
  // webServer.on("^\\/.*\\.svg$", sendData);
  // webServer.on("*.svg", sendData);
  // webServer.on("^\\/.*\\.css$", sendData);
  // webServer.on("/index.css", sendData);

  webServer.onNotFound(notFound);

  Serial.printf("Using serveStatic\n");
  webServer.serveStatic("/", LITTLEFS, "/").setDefaultFile("index.html");

  webServer.begin();

  Serial.printf("\n\nStarted.\n");
}

void loop()
{
  delay(1000);
}

```

BbIKTOP, 2021-06-22:
Using ``c=0;while [ $c -lt 5 ] ; do  (curl http://172.16.5.143/index.css>/dev/null 2>&1&); c=`expr $c + 1`;done;ls -l`` : 
```
Starting...
..........................
IP Address: 172.16.5.143
Using serveStatic

Started.
E (44395) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (44395) task_wdt:  - async_tcp (CPU 0/1)
E (44395) task_wdt: Tasks currently running:
E (44395) task_wdt: CPU 0: IDLE0
E (44395) task_wdt: CPU 1: IDLE1
E (44395) task_wdt: Aborting.
abort() was called at PC 0x40144cb8 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x400886dc:0x3ffbfa20 0x40088959:0x3ffbfa40 0x40144cb8:0x3ffbfa60 0x40086e6d:0x3ffbfa80 0x40178ab3:0x3ffbc220 0x40147eef:0x3ffbc240 0x4008b119:0x3ffbc260 0x4008996a:0x3ffbc280
  #0  0x400886dc:0x3ffbfa20 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40088959:0x3ffbfa40 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x40144cb8:0x3ffbfa60 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x40086e6d:0x3ffbfa80 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x40178ab3:0x3ffbc220 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
  #5  0x40147eef:0x3ffbc240 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #6  0x4008b119:0x3ffbc260 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #7  0x4008996a:0x3ffbc280 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

BbIKTOP, 2021-06-22:
P.S.: Sometimes it happens immediately, sometimes after a few tries

zekageri, 2021-06-22:
Have you tried with different kind of esp? I had an esp32 d1-mini which does the same. Reboot immidiately at first boot and things like that. It was a power problem because the onboard regulator was crap. The module looks like this: 

[image]

BbIKTOP, 2021-06-22:
@zekageri esp32 can’t draw even 1a. According to the datasheet, it can draw 0.5A maximum, but I never managed to make it consume such a power. Usually it takes 200-250mA. But I’ll try another one and report back

BbIKTOP, 2021-06-22:
Well, took a new esp wrover from the tape reel and tried it - the same result. So, the esp is not a reason this happens

zekageri, 2021-06-22:
Other problems that i encountered with:

- Because the callbacks are interrupts, you can not do much computation inside a callback. You must use deferred request handlers.
- The esp for some odd reason can not serve larger files than 130kb. You must divide the file to multiple parts. ( if you are usin SPIFFS it is even worse, because that filesystem is deprecated and slow )
- The esp's maximum allowed TCP socket connection is 10 at once in case of an esp32. ( On the 8266 its 5 or max 8 i think. ) So if you have surpassed this limit, the esp will crash instead of report an error in some form.

BbIKTOP, 2021-06-22:
@zekageri these problems are well known and not related to this issue. Like:
1) there’s no “callbacks” in the test case provided above 
2) the file is about 5kb or something 
3) amount of connections is defined in the core, and i have tried 16, but in any case, aws should handle this. And I observed aws returning 500 in such a cases. 

Anyway, there’s definitely an error in the code, and i cannot find where.

BbIKTOP, 2021-06-22:
Well, in order to summarise, as I think there's no reason to open a separate issue:
1) Let's use test code without callbacks with serveStatic (chunked gives the same result, but let's use serveStatic to avoid "foreign" code influence:
```c++
#include <Arduino.h>
#include <LITTLEFS.h>
#include <WiFi.h>

#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

#include <wifi_creds.h>

extern void sendData(AsyncWebServerRequest *request);

AsyncWebServer webServer(80);

const char *ssid = WIFI_SSID;
const char *password = WIFI_PASSWD;

void setup()
{
  Serial.begin(115200);
  Serial.printf("\n\nStarting...\n");

  LITTLEFS.begin();

  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED)
  {

    Serial.printf(".");
    delay(100);
  }

  Serial.print("\nIP Address: ");
  Serial.println(WiFi.localIP());

  Serial.printf("Using serveStatic\n");
  webServer.serveStatic("/", LITTLEFS, "/").setDefaultFile("index.html");

  webServer.begin();

  Serial.printf("\n\nStarted.\n");
}

void loop()
{
  delay(1000);
}
```
2) Put a single 16kb index.css file on the littlefs filesystem
3) Use following script to test aws (assuming it's ip is 172.16.5.151, as in my case)

```sh
while true; do c=0;while [ $c -lt 5 ] ; do  (curl http://172.16.5.151/index.css>/dev/null 2>&1&); c=`expr $c + 1`;done;sleep 1;done
```
4) Observe esp crash after a few seconds:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8

Starting...
.........................
IP Address: 172.16.5.151
Using serveStatic

Started.
E (78768) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (78768) task_wdt:  - async_tcp (CPU 0/1)
E (78768) task_wdt: Tasks currently running:
E (78768) task_wdt: CPU 0: IDLE0
E (78768) task_wdt: CPU 1: IDLE1
E (78768) task_wdt: Aborting.
abort() was called at PC 0x40144cb8 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x400886dc:0x3ffbfa20 0x40088959:0x3ffbfa40 0x40144cb8:0x3ffbfa60 0x40086e6d:0x3ffbfa80 0x40178ab3:0x3ffbc220 0x40147eef:0x3ffbc240 0x4008b119:0x3ffbc260 0x4008996a:0x3ffbc280
  #0  0x400886dc:0x3ffbfa20 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #1  0x40088959:0x3ffbfa40 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
  #2  0x40144cb8:0x3ffbfa60 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x40086e6d:0x3ffbfa80 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x40178ab3:0x3ffbc220 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
  #5  0x40147eef:0x3ffbc240 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #6  0x4008b119:0x3ffbc260 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #7  0x4008996a:0x3ffbc280 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
```

BbIKTOP, 2021-06-22:
And another remark: my first guess was, that littlefs is slow and this is the reason. But in the logs above, I fdiscovered, that then this wdt happens, file is read and content is in the memory. looks like it's a kinda deadlock when calling lwip

Pablo2048, 2021-06-22:
Did you try to use core 1 for asynctcp? ( 
https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.h#L34 
)

BbIKTOP, 2021-06-22:
I didn’t change anything, just got current aws version, so, it is exactly like in the sources here

Pablo2048, 2021-06-22:
Well, because of many discussions about this behavior in AsyncTCP library there can be problem inside AsyncTCP and not in this library. ( https://github.com/me-no-dev/AsyncTCP/issues/99 , https://github.com/me-no-dev/AsyncTCP/issues/74 , https://github.com/me-no-dev/AsyncTCP/issues/69 , ...)

BbIKTOP, 2021-06-22:
Maybe yes. Maybe no. I have no idea and still cannot find the reason i did as much as i could investigating it. Now i ran out of ideas

rpavluchenko, 2021-06-23:
> Ну, чтобы подвести итог, думаю, нет причин открывать отдельный выпуск:
> 
> 1. Давайте использовать тестовый код без обратных вызовов с serveStatic (chunked дает тот же результат, но давайте воспользуемся serveStatic, чтобы избежать влияния «чужого» кода:
> 
> ```c++
> # include  < Arduino.h > 
> # include  < LITTLEFS.h > 
> # include  < WiFi.h >
>  
> # include  < AsyncTCP.h > 
> # include  < ESPAsyncWebServer.h >
>  
> # include  < wifi_creds.h >
> 
>  extern  void  sendData (запрос AsyncWebServerRequest *) ; 
> 
> AsyncWebServer webServer ( 80 ); const char * ssid = WIFI_SSID;
> const char * пароль = WIFI_PASSWD;
> 
>   
> 
> void  setup () 
> { 
>   Последовательный. begin ( 115200 ); 
>   Серийный. printf ( " \ n \ n Запуск ... \ n " ); 
> 
>   МАЛЕНЬКИЕ. begin (); 
> 
>   Вай-фай. режим (WIFI_STA); 
>   Вай-фай. begin (ssid, пароль); while (WiFi. status ()! = WL_CONNECTED) 
>   { 
>     Последовательный. printf ( " . " );
>     задержка ( 100 ); 
>   } 
>   Последовательный. печать (
> 
>   
> 
> " \ n IP-адрес: " ); 
>   Серийный. println (WiFi. localIP ()); 
> 
>   Серийный. printf ( " Использование serveStatic \ n " ); 
>   веб сервер. serveStatic ( " / " , НЕМНОГО, " / " ). setDefaultFile ( " index.html " ); 
> 
>   веб сервер. begin (); 
> 
>   Серийный. printf ( " \ n \ n Запущено. \ n " ); 
> }
> 
> void  loop () 
> { задержка ( 1000 ); 
> }
>   
> ```
> 
> 1. Поместите один файл index.css размером 16 КБ в файловую систему littlefs
> 2. Используйте следующий скрипт для тестирования aws (при условии, что это ip 172.16.5.151, как в моем случае)
> 
> ```shell
> пока  правда ;  сделать c = 0 ; а [ $ c  -lt 5] ;  сделать   (curl http://172.16.5.151/index.css > / dev / null 2> & 1 & ) ; c = ` expr $ c + 1` ; сделано ; сон 1 ; Выполнено
> ```
> 
> 1. Наблюдайте за аварийным завершением работы esp через несколько секунд:
> 
> ```
> ets Jun  8 2016 00:22:57
> 
> rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
> flash read err, 1000
> ets_main.c 371 
> ets Jun  8 2016 00:22:57
> 
> rst:0x10 (RTCWDT_RTC_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
> configsip: 0, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1044
> load:0x40078000,len:10124
> load:0x40080400,len:5828
> entry 0x400806a8
> 
> 
> Starting...
> .........................
> IP Address: 172.16.5.151
> Using serveStatic
> 
> 
> Started.
> E (78768) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
> E (78768) task_wdt:  - async_tcp (CPU 0/1)
> E (78768) task_wdt: Tasks currently running:
> E (78768) task_wdt: CPU 0: IDLE0
> E (78768) task_wdt: CPU 1: IDLE1
> E (78768) task_wdt: Aborting.
> abort() was called at PC 0x40144cb8 on core 0
> 
> ELF file SHA256: 0000000000000000
> 
> Backtrace: 0x400886dc:0x3ffbfa20 0x40088959:0x3ffbfa40 0x40144cb8:0x3ffbfa60 0x40086e6d:0x3ffbfa80 0x40178ab3:0x3ffbc220 0x40147eef:0x3ffbc240 0x4008b119:0x3ffbc260 0x4008996a:0x3ffbc280
>   #0  0x400886dc:0x3ffbfa20 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
>   #1  0x40088959:0x3ffbfa40 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:715
>   #2  0x40144cb8:0x3ffbfa60 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
>   #3  0x40086e6d:0x3ffbfa80 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
>   #4  0x40178ab3:0x3ffbc220 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
>   #5  0x40147eef:0x3ffbc240 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
>   #6  0x4008b119:0x3ffbc260 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
>   #7  0x4008996a:0x3ffbc280 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)
> 
> Rebooting...
> ```

All right, but show the html code.

BbIKTOP, 2021-06-23:
> All right, but show the html code.
> Or put empty files in the file system and check how it behaves in this case.

Sorry, not sure I understand what html do you mean. Also, don’t see a reason to check it with empty file - how could it help? Suppose, it will work with the empty file. Or, it would not. How could it help?

rpavluchenko, 2021-06-26:
> > Хорошо, но покажите HTML-код. 
> > Или поместите в файловую систему пустые файлы и проверьте, как она себя ведет в этом случае.
> 
> Извините, я не совсем понимаю, что вы имеете в виду. Также не вижу смысла проверять это с пустым файлом - чем это может помочь? Предположим, он будет работать с пустым файлом. Или нет. Как это могло помочь?

I worked with this problem for a long time, and in my case it turned out that the js file contained a request for files from CDN and files from SPIFS and this script kept the connection for an unreasonably long time, which is why the timeout was triggered. I just propose to make sure that there are no random factors.

BbIKTOP, 2021-06-26:
I use littlefs, don't think it makes a big difference cause I observe data have been read and are in the buffer at this point. There's probably, some deadlock in this thread.
I'd like to notice, that esp32 arduino core is based on the rtos, so, I'd advice to use ordinary webserver in the separate dedicated task,as  it works just fine. There's no need for async, actually, as rtos supports multitasking. And since author is usually busy, I doubt we will get a fix soon, so it is a good alternative.

closed 2021-09-19 by stale[bot]