me-no-dev/ESPAsyncWebServer #1008: Template/placeholder issue

Brekel42, 2021-06-23:
There appears to be an issue when the progmem data contains % signs.

eg my data contains html  with the text "border-radius: 50%".  Later on in the html, I have the placeholder "%PLACEHOLDER%".

Looking at the data sent  back, when processed it is taking everything between the % in "50%" and the first % in "%PLACEHOLDER%" as being the placeholder, so all the intervening html code is replaced with "".

Brekel42, 2021-06-23:
Just found issue 300, same problem.,  plus resolution

closed 2021-06-23 by Brekel42

chupocro, 2021-07-14:
> 
> 
> Just found issue 300, same problem., plus resolution

I wouldn't say changing the library is the solution. What will happen after the library update?

Brekel42, 2021-07-14:
I agree, more of a work around than a solution. But I figured that since the issue was already raised on issue #300,  it didn't need to be on a separate issue as well.