me-no-dev/ESPAsyncWebServer #1009: Long taking downloads killed by watchdog

danielkucera, 2021-06-27:
I was able to isolate and create a test case for requests killed by watchdog. The error is following:
```
handle download: index:0x3F0FC maxLen:0
handle download: index:0x3F0FC maxLen:5744
handle download: index:0x4076C maxLen:0
handle download: index:0x4076C maxLen:5744
handle download: index:0x41DDC maxLen:0
handle download: index:0x41DDC maxLen:5744
handle download: index:0x4344C maxLen:0
handle download: index:0x4344C maxLen:4308
E (39700) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (39700) task_wdt:  - async_tcp (CPU 0/1)
E (39700) task_wdt: Tasks currently running:
E (39700) task_wdt: CPU 0: IDLE0
E (39700) task_wdt: CPU 1: loopTask
E (39700) task_wdt: Aborting.
abort() was called at PC 0x400e65bb on core 0

Backtrace: 0x4008c98c:0x3ffbe170 0x4008cbbd:0x3ffbe190 0x400e65bb:0x3ffbe1b0 0x40084b3d:0x3ffbe1d0 0x40153467:0x3ffbc0d0 0x400e799f:0x3ffbc0f0 0x4008a72d:0x3ffbc110 0x40088f49:0x3ffbc130
  #0  0x4008c98c:0x3ffbe170 in invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #1  0x4008cbbd:0x3ffbe190 in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c:707
  #2  0x400e65bb:0x3ffbe1b0 in task_wdt_isr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/task_wdt.c:252
  #3  0x40084b3d:0x3ffbe1d0 in _xt_lowint1 at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/xtensa_vectors.S:1154
  #4  0x40153467:0x3ffbc0d0 in esp_pm_impl_waiti at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/pm_esp32.c:492
  #5  0x400e799f:0x3ffbc0f0 in esp_vApplicationIdleHook at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/freertos_hooks.c:108
  #6  0x4008a72d:0x3ffbc110 in prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3507
  #7  0x40088f49:0x3ffbc130 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)

Rebooting...
ets Jun  8 2016 00:22:57
```
The point is that I want to send data to client over a longer period of time (because getting them takes some time - simulated by delay in test case).
You can see my test project here:
https://github.com/danielkucera/asynchttp-test/blob/master/src/main.cpp#L43

I am correctly returning from the handler but the code gets stuck somewhere in asyncweb/tcp.

Can you please help?

danielkucera, 2021-06-27:
To simulate, just run:
```
curl 'ESPIP/download_flash' > /dev/null
```

Pablo2048, 2021-06-27:
This is well known problem of the AsyncWebServer, which has been discussed in many threads without solution so far. Your example is bad anyway - did you read https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember , especially `You can not use yield or` **delay** `or any function that uses them inside the callbacks` ?

danielkucera, 2021-06-27:
okay, are you saying that if will compute pi instead of delay, it will work?

closed 2021-09-19 by stale[bot]