me-no-dev/ESPAsyncWebServer #1016: ESP Async WebServer/src/SPIFFSEditor.h:16:101: warning: 'SPIFFS' is deprecated:

j-robin-hunter, 2021-07-10:
SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems

embedded-creations, 2021-08-05:
See discussion in #879

taleroangel, 2022-01-07:
In file included from C:\Users\angel\Documents\Development\libraries\ESPAsyncWebServer\src\SPIFFSEditor.cpp:1:
C:\Users\angel\Documents\Development\libraries\ESPAsyncWebServer\src\SPIFFSEditor.h:16:101: warning: 'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
   16 |     SPIFFSEditor(const String& username=String(), const String& password=String(), const fs::FS& fs=SPIFFS);
      |                                                                                                     ^~~~~~
In file included from C:\Users\angel\Documents\Development\libraries\ESPAsyncWebServer\src/ESPAsyncWebServer.h:27,
                 from C:\Users\angel\Documents\Development\libraries\ESPAsyncWebServer\src\SPIFFSEditor.h:3,
                 from C:\Users\angel\Documents\Development\libraries\ESPAsyncWebServer\src\SPIFFSEditor.cpp:1:
C:\Users\angel\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\3.0.2\cores\esp8266/FS.h:275:15: note: declared here
  275 | extern fs::FS SPIFFS __attribute__((deprecated("SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems.")));
      |

closed 2022-04-16 by stale[bot]