me-no-dev/ESPAsyncWebServer #1017: Which fork to pick?

rikki78, 2021-07-10:
It seems the repository isn't updated anymore and issues remain open. There are many forks, but I'm not sure which one to pick. Any recommendations anyone?

proddy, 2021-11-05:
I'm on the same fork, and it seems solid. Although you may want to hunt around for better versions of the asynctcp library

zekageri, 2021-11-05:
I got websocket crashes with that fork. :(

proddy, 2021-11-05:
yes I saw your post on that.

rikki78, 2021-11-05:
So which one would you recommend?

zekageri, 2021-11-05:
Don't know man. I'm desperately trying to find one. This was really promising. Other than this stupid socket crash this is the most reliable. With yubo fixes lib

proddy, 2021-11-05:
> I got websocket crashes with that fork. :(

if you're getting crashes in asyntcp like
```
E (36601) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (36601) task_wdt:  - async_tcp (CPU 0/1)
E (36601) task_wdt: Tasks currently running:
E (36601) task_wdt: CPU 0: IDLE0
E (36601) task_wdt: CPU 1: IDLE1
E (36601) task_wdt: Aborting.
abort() was called at PC 0x40194548 on core 0
```

then try https://github.com/me-no-dev/AsyncTCP/issues/121 
it really helped on stability

zekageri, 2021-11-05:
Oh. I got no watchdog reset but I have problems with freeing the socket queue. I will try this one out. Thank you!

DRSDavidSoft, 2024-08-29:
It appears the updated version of the 0xFEEDC0DE64 fork is this one:
https://github.com/mathieucarbou/ESPAsyncWebServer

I have been using the WebSocket server on top of this fork for quite a while now (I only use `textAll()`) and it works smoothly.