me-no-dev/ESPAsyncWebServer #1018: Crash when using PZEM004TV3.0 Library. (SoftwareSerial)

Black2vs2, 2021-07-12:
Hi,

I'm trying to use [PZEM004TV3.0](https://github.com/mandulaj/PZEM-004T-v30) library. When i try to create or access a PZEM004TV30 object to build a JSON response to send back to the client it leads to a crash with this stack trace:

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:137 __yield

>>>stack>>>

ctx: sys
sp: 3fffeb50 end: 3fffffb0 offset: 0000
3fffeb50:  00000002 40282dec 00000000 40106e1d
3fffeb60:  000000fe 00000000 00000000 00000000
3fffeb70:  00000000 00000000 00000000 3fff1154  
3fffeb80:  00000000 00000020 4bc6a7f0 0000180a
3fffeb90:  3fff0e7c 3fffec20 00000000 40218832
3fffeba0:  0000180a 00000001 3fff517c 4021889f  
3fffebb0:  00000000 00000000 00000000 402182f9
3fffebc0:  3fff0e7c 3fffec20 00000000 4020828f
3fffebd0:  00000019 00000000 3fffec20 402082f9  
3fffebe0:  40215e0c 3ffe912a 000004f8 3fff0fc8
3fffebf0:  00000000 3fffec20 3fff0e7c 4020839e
3fffec00:  3fff3b24 00000002 3fffec20 402085a9
3fffec10:  3fff3b24 00000002 3fff0dec 40207e5b  
3fffec20:  3fff517c 3ffff801 3fffed34 40100dfc
3fffec30:  3fffecc0 3fffec60 3fffed40 3fff0fc8
3fffec40:  40215e0c 3fffec9f ffffff38 ffffffff  
3fffec50:  40215e0c 3fffec9f 3fff0fc8 4021614c
3fffec60:  00000002 3fffec9f 0000000a 40216170
3fffec70:  203a4449 00000010 3fffed40 40216480  
3fffec80:  00000000 4020a0cd 00000020 4021831a
3fffec90:  00000000 00000002 3ffe912b 40219594
3fffeca0:  3ffe912c 3ffe83c4 3fff4091 3fff4090
3fffecb0:  00000009 00000009 3fffed34 00000002  
3fffecc0:  40215e0c 3ffe912a 3fff3b24 3fff0dec
3fffecd0:  40215e0c 3ffe912a 3fff0fc8 4021614c
3fffece0:  00000004 3fffed1c 3fff0d28 402163c4  
3fffecf0:  3fff0fc8 00000002 3fff0fc8 00000002
3fffed00:  3fff0fc8 00000002 3fff4a3c 40208035
3fffed10:  00000000 00000003 00000000 40217364
3fffed20:  00000000 3fffed40 3fff4ea4 4020970d  
3fffed30:  00000000 3fff4dc8 3fff4a3c 40221024
3fffed40:  00000000 00637067 00000000 4020be2d
3fffed50:  3fff4a3c 3fff4a3c 3fff45ac 4020c94c  
3fffed60:  3fff1548 00000001 3fff4a70 00000002
3fffed70:  00000000 3fff4dc8 3fff4a3c 4020af58
3fffed80:  00000016 000000d4 00000020 40217fa8
3fffed90:  00000000 3fff4dc8 3fff4a70 40217650  
3fffeda0:  3fff4dc8 00000000 00000020 00000002
3fffedb0:  00000002 3fff4dc8 3fff4a3c 4020b01e
3fffedc0:  00000000 4bc6a7f0 7d70a3d7 001bbe41  
3fffedd0:  00000000 00000000 4bc6a7f0 00000000
3fffede0:  3fff4a70 00000000 40100445 00000000
3fffedf0:  000017fe 00000000 00000000 3fff4b84
3fffee00:  3fffee60 3fff4934 3fff4874 4020b200  
3fffee10:  3fffee60 3fff4934 3fff4874 4021b3fc
3fffee20:  00000200 3fff4bca 3fff4934 3fff4a3c
3fffee30:  00000000 3fff0e0c 3fff4874 005db584  
3fffee40:  3ffe0000 2c9f0300 4000050c 00000000
3fffee50:  3fff4b84 3fff4874 3fff4934 4021b45c
3fffee60:  3fff4a28 3fff4a1c 00000010 3fff1548  
3fffee70:  3fff17c8 3fff1548 3fff4b84 3fff1994
3fffee80:  3fff4874 3fff1998 3fff1999 4025cc95
3fffee90:  00000001 00001f90 00000000 3fffdab0
3fffeea0:  00000018 00000000 3fff427c 00000000  
3fffeeb0:  3fff0000 00000236 00000020 3fff1548
3fffeec0:  3fff17c8 3fff4ba2 3fff4b84 4025a421
3fffeed0:  00000014 3fff1548 3ffe88d0 40100bbb  
[... 249 lines cut ...]
3ffffe70:  00000376 00000376 3ffe88d0 40100bbb  
3ffffe80:  0000039f 0000039f 3ffe88d0 00000001
3ffffe90:  3fff4374 00000020 3fff472c 40100d9a
3ffffea0:  3fff4374 00000000 3fffff00 4024dd10  
3ffffeb0:  40222c88 00000000 3fffff00 40233928
3ffffec0:  00000000 4bc6a7f0 d645a1ca 001aa5af
3ffffed0:  00000000 00000000 4bc6a7f0 00000000
3ffffee0:  00000001 3fff2ab4 40100445 00000000  
3ffffef0:  0000170b 3fff2ab4 00000001 4021be68
3fffff00:  3fff0000 00000000 00000000 00000001
3fffff10:  3fff4374 3fff2ab4 3fff43a8 4021cc30  
3fffff20:  3fff4374 3fff2ab4 00000001 4021d123
3fffff30:  3fff0e00 3fff3b24 00000004 402164d0
3fffff40:  00000001 3fff43a8 3fff0f00 3fff0e0c  
3fffff50:  401060e5 00257147 3fff1174 00000000
3fffff60:  3fff0840 3fff1174 000007d0 3fff1174
3fffff70:  3fffdad0 3fff110c 4021823c 3fffefa0
3fffff80:  3fff1174 3fffdad0 3fff110c 4021895a  
3fffff90:  3fffdad0 00000000 3fff10f8 40205dbe
3fffffa0:  3fffdad0 00000000 3fff10f8 40218368
<<<stack<<<
--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

Is there any solution? I know that PZEM004TV30 uses SoftwareSerial inside, so probably there is some kind of delay() or yield() inside that leads to the crash. Any solution?

vortigont, 2021-07-23:
do not call PZEM inside async response. Poll the meter with scheduler and use web requests to reply with in-memory values.

Black2vs2, 2021-07-23:
Thank you, this fixed my issue

TL:DR:
Solution is to save the data in a variable in memory and send the response with that instead of calculating it in real time

closed 2021-07-23 by Black2vs2