me-no-dev/ESPAsyncWebServer #1020: async_tcp (CPU 0/1) issue when trying to use delay or any other function that takes a while.

krupis, 2021-07-14:
Hello. I have been reading about this issue for a while and there are many people having simmilar problems with the watchdog reset failure from the async tcp. I wonder if this issue has been looked at or if there is any workaround?

I am developing a program and it works as following:

1. After powering the device, The device makes an AP. I am connecting to this AP and setting WiFi credentails to which I want to connect. After submiting the values of ID and password, the device will switch to the STA mode an connect to the wifi. After connecting to the WiFi, it will attempt to perform OTA from the server. However, I am encountering the following error :
```
E (62965) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (62965) task_wdt:  - async_tcp (CPU 0/1)
E (62965) task_wdt: Tasks currently running:
E (62965) task_wdt: CPU 0: IDLE0
E (62965) task_wdt: CPU 1: loopTask
E (62965) task_wdt: Aborting.
abort() was called at PC 0x4013b78c on core 0
```

I have managed to simplified code a bit, the following function will be executed once I click "submit" button on my webserver:

```
  server.on("/action_page", HTTP_GET, [](AsyncWebServerRequest * request) {
    Serial.println("executing action page");
    String input_ID;
    String input_password;

    if (request->hasParam("Network_ID")) 
    {
        input_ID =  request->getParam("Network_ID")->value();
    }

    if (request->hasParam("Network_pass")) 
    {
        input_password =  request->getParam("Network_pass")->value();
    }

    //request->send(SPIFFS, "/action_page.html");
    request->send(200, "text/html", "<h1>The following values has been changed:</h1> <br>  Network_ID: " + input_ID +"<br> Network_Pass:"+input_password+"<br><a href=\"/\">Return to Home Page</a>");
    delay(1000);
    attempt_wifi_connection(input_ID,input_password);
  });
```
 
**Notice the attempt_wifi_connection(input_ID,input_password) function at the end of the code.**

```
void attempt_wifi_connection(String ID_string, String Pass_string){
  Serial.println("attempting wifi connection");
  char temp_buffer_ID[100]; // temporary storage for char array ID
  char temp_buffer_Pass[100]; // temporary storage for char array Password
   //Convert String type ID and password to char array
   ID_string.toCharArray(temp_buffer_ID, sizeof(temp_buffer_ID));
   Pass_string.toCharArray(temp_buffer_Pass, sizeof(temp_buffer_Pass));
   Serial.print("Attempting to connect to the wifi with ID and password=");
   Serial.println(temp_buffer_ID);
   Serial.println(temp_buffer_Pass);

   WiFi.mode(WIFI_STA);//start program by attempting to connect to wifi
   WiFi.begin(temp_buffer_ID,temp_buffer_Pass);
   //WiFi.begin(ID_string.c_str(),Pass_string.c_str());
   if(WiFi.waitForConnectResult() != WL_CONNECTED)
    {
    Serial.println("WiFi Failed!");
    handle_access_point();
    return;
    }
    
   Serial.println("Connection sucessful");
   Serial.print("IP Address: ");
   Serial.println(WiFi.localIP());
   Serial.println("performing OTA");
   //start_OTA();
   delay(4000);

}
```

I dont even need to try to perform OTA to crash the program, it is enough to call 4 seconds delay and the CPU will reset with the error that I have previously shown. I have tried 1 second, 2 seconds and 3 seconds delay as well, but the CPU does not reset with those delays. It is only happening with the 4 seconds or above!

Serial monitor:

```
attempting wifi connection
Attempting to connect to the wifi with ID and password=xxxxxxx
xxxx
Connection sucessful
IP Address: 192.168.4.172
performing OTA
E (21174) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (21174) task_wdt:  - async_tcp (CPU 0/1)
E (21174) task_wdt: Tasks currently running:
E (21174) task_wdt: CPU 0: IDLE0
E (21174) task_wdt: CPU 1: loopTask
E (21174) task_wdt: Aborting.
abort() was called at PC 0x4013b730 on core 0

ELF file SHA256: 0000000000000000

Backtrace: 0x400886dc:0x3ffbf810 0x40088959:0x3ffbf830 0x4013b730:0x3ffbf850 0x40086e6d:0x3ffbf870 0x40150bc3:0x3ffbc220 0x4013d02b:0x3ffbc240 0x4008b119:0x3ffbc260 0x4008996a:0x3ffbc280   

Rebooting...
```
See after printing out performing OTA, the device crashes. Any idea what would be the possible solutions?

Pablo2048, 2021-07-14:
I'm posting this here about third time:
... from the readme.md (https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember):
### Important things to remember

This is fully asynchronous server and as such does not run on the loop thread.
**You can not use yield or delay or any function that uses them inside the callbacks**

so the solution is probably **DO NOT USE THE DELAY!!!** and do not call `attempt_wifi_connection()` from the callback also...

JohnAndrewPurio, 2021-07-14:
I am also facing the same issue. My question is, is there a way to call a function after a specific request-response cycle from the server ends?  Basically, if the response ends, I want to connect my ESP32 to the wifi from the request body sent that I parsed in the callback.

krupis, 2021-07-15:
> I'm posting this here about third time:
> ... from the readme.md (https://github.com/me-no-dev/ESPAsyncWebServer#important-things-to-remember):
>
> ### Important things to remember
> This is fully [...]

Thanks for the reply. My mistake for not noticing. So do you suggest that I try to do everything in void loop? For example I would set a flag or something and then in my void loop constantly check that flag and if it is set, do the function. Also, I do not understand the following:

Calling the OTA from the callback gives me an error as I have shown, however, calling the OTA function from the void loop seems to works ( using the flag method as I have mentioned ) . However, I cant seem to understand what would be the difference? Since my OTA function is blocking, I do not think there is any difference whether it is called in void loop or any callback since during its execution nothing else is going to happen. Is that not true?

Also, what should I do if I have to execute a long function? For example lets say I want to perform OTA over the GSM and that can take some time. Is this not going to crash my program?

Pablo2048, 2021-07-15:
You have to remember that every async callback is made from asynctcp specific thread 
https://github.com/me-no-dev/AsyncTCP/blob/ca8ac5f919d02bea07b474531981ddbfd64de97c/src/AsyncTCP.cpp#L188 
and not from loopTask thread so you just can not do same things like in the loop() (no blocking function, no delay, no yield, ...). Use some kind of semaphore or mailbox to pass parameters or signals from background task to the foreground one.

Pablo2048, 2021-07-15:
> I am also facing the same issue. My question is, is there a way to call a function after a specific request-response cycle from the server ends? Basically, if the response ends, I want to connect my ESP32 to the wifi using the credentials from the request body sent that I parsed in the callback.

You can use onDisconnect callback:

https://github.com/me-no-dev/ESPAsyncWebServer/blob/2f784268f0a358741ee6384480d48656e159d726/src/WebRequest.cpp#L220

embedded-creations, 2021-09-14:
@Pablo2048 I made a new response type that has a number of improvements over the existing types, which I'll publish at some point hopefully soon.  One thing I added was a deconstructor that calls the callback with a `final` flag set to true so the callback can wrap things up when the request ends:

```
AsyncChunkedStreamResponse::~AsyncChunkedStreamResponse(){
  // there's no more chance to send data, so set the buffer sizes to zero
  if(_cbuf) delete _cbuf;
  _bufferRemaining = 0;

  // one last call to the callback to give a chance to wrap things up
  _content(this, _filledLength, true);
}
```

closed 2022-04-16 by stale[bot]