me-no-dev/ESPAsyncWebServer #1022: conflict between libraries   "HttpsOTAUpdate.h"  and   "HttpsOTAUpdate.h"

pedros89, 2021-07-19:
There is a declaration conflict between these two libraries.
Just put these two lines in the code to see it by yourself:

```
#include "ESPAsyncWebServer.h"
#include "HttpsOTAUpdate.h"
```

I'm using the official example and library for OTA from[ espressif ](https://github.com/espressif/arduino-esp32)
https://github.com/espressif/arduino-esp32/tree/master/libraries/Update/examples/HTTPS_OTA_Update

I need to update over the air  my ESP32 connecting to my https webserver that has the firmware bin in the FTP folder.
But I get this error for declaration.

error: redeclaration of 'HTTP_DELETE'

```

C:\Users\Pietro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/nghttp/http_parser.h:95:6: error: redeclaration of 'HTTP_DELETE'
   XX(0,  DELETE,      DELETE)       \
      ^
.......
```

pedros89, 2021-07-19:
I have tried to reduce the amount of errors with the   #define WEBSERVER_H

```
#include "WiFi.h"
#include "HttpsOTAUpdate.h"
#define WEBSERVER_H
#include "ESPAsyncWebServer.h"
```

But still the code has problems

```
C:\Users\Pietro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\ESPAsyncWebServer\src/WebHandlerImpl.h: In constructor 'AsyncCallbackWebHandler::AsyncCallbackWebHandler()':
C:\Users\Pietro\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\ESPAsyncWebServer\src/WebHandlerImpl.h:76:49: error: 'HTTP_ANY' was not declared in this scope
     AsyncCallbackWebHandler() : _uri(), _method(HTTP_ANY), _onRequest(NULL), _onUpload(NULL), _onBody(NULL), _isRegex(false) {}
```

In my code I need the ESP to connect to my website to download the firmware and install it

I am also including an https certificate

static const char *url = "https://mywebsite/esp32";
```

void updateESPFirmware(){
   HttpsOTA.onHttpEvent(HttpEvent);
   Serial.println("Starting OTA");
   HttpsOTA.begin(url, server_certificate); 

   Serial.println("Please Wait it takes some time ..."); 
}
```

perotom, 2022-01-19:
Did you managed to find a solution?

pedros89, 2022-01-19:
Yes I sorted it out but I have abandoned AsyncWebServer.
I have used normal webserver instead 
https://github.com/pedros89/ESP32-data-to-local-webpage

and I have added official idf OTA update and worked great

I believe that if you use the official OTA from idf it will work with AsyncWebServer as well. For sure it works with official Webserver as I have used it this way.
This repository I made always works with OTA updates:
https://github.com/pedros89/ESP32-update-both-OTA-Firmware-and-OTA-SPIFFS-with-Arduino

In these two repositories I made you will find help I hope

closed 2022-04-16 by stale[bot]