me-no-dev/ESPAsyncWebServer #1023: Value truncated if there is a space

mpbejo, 2021-07-23:
Hi everybody,
i am developing a project with a small html page where to write some credentials and then store the values on the Preferences.
When I enter in the page I show the previously saved data, but when there is a space in the string the value show is truncated.
My html page:
const char index_html[] PROGMEM = R"rawliteral(
.......
<form action="/get">
    User: <input type="text" name="User" value=%HTTP_USER%><br>
    User password: <input type="text" name="UserPass" value=%HTTP_PASSWORD%><br>
  <input type="submit" value="Submit"><br>
  </form>
.........
)rawliteral";

the server.on "call" the page:
server.on("/", HTTP_GET, [](AsyncWebServerRequest *request){
    if(!request->authenticate(http_username.c_str(), http_password.c_str()))
      return request->requestAuthentication();
    request->send_P(200, "text/html", index_html,processor);
  });

the processor  is used to write the values in the html page:

String processor(const String& var){

  if (var == "HTTP_USER")
  {
    Serial.println(http_username);
    return String(http_username);
  }
  if (var == "HTTP_PASSWORD")
  {
    Serial.println(http_username);
    return String(http_password);
  }
  return String();
}

if I  write "HelloWorld" for the HTTP_USER when I reload the page I see "HelloWorld", instead if I write "Hello World"
when I reload the page I see only "Hello". In the Serial Monitor I see always the data correctly this mean that they are stored correctly.
I can't find the problem, any advice?
Thank You 
Best Regards 
Marco

zekageri, 2021-07-23:
Where do you see this? On mobile browser?

On desktop you can right click and inspect the input element and see the actual value inside.

I think the text truncation happens because it can't fit into the viewport. Can you confirm that you can see the whole value on desktop ( big screen ) ?

mpbejo, 2021-07-23:
Hi zekageri,
thank You for your reply,
I am using desktop browser,  as You suggest I used the inspect  and I can see this for example:
<input type="text" name="WiFiSSID" value="Room" 200>
but my correct value "Room 200"
[image]

zekageri, 2021-07-25:
I see. Your processor function can not see the claw marks and the html puts it as it sees it is right.

Try the html page like this:

```HTML
User: <input type="text" name="User" value="%HTTP_USER%"><br>
User password: <input type="text" name="UserPass" value="%HTTP_PASSWORD%"><br>
```

Notice the " " clawmarks that surrounds the variable names.

mpbejo, 2021-07-25:
Hi zekageri, 
I think your tip is correct, I was looking on internet and I found similar suggestion.
I try and then I will tell You if solve the problem
Best Regards

mpbejo, 2021-07-27:
Hi zekageri, your tip was correct adding claw mark the  problem is solved, my fault.
Thank You very much
Best Regards

closed 2021-07-27 by mpbejo