me-no-dev/ESPAsyncWebServer #1027: Multiple libraries were found for "WiFi.h"

nicoff, 2021-07-31:
There seems to be a conflict with the Wifi library of arduino:
The following error message comes at compilation, is there a quick workaround?
I substitue personal paths with ***

Multiple libraries were found for "WiFi.h"
 Used: /home/***/snap/arduino/61/.arduino15/packages/esp32/hardware/esp32/1.0.6/libraries/WiFi
 Not used: /snap/arduino/61/libraries/WiFi
In file included from /***.ino:4:0:
/***/snap/arduino/61/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:33:22: fatal error: AsyncTCP.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board ESP32 Dev Module.

Pablo2048, 2021-07-31:
??? I don't see any multiple libraries, but there is clearly:

>/***/snap/arduino/61/Arduino/libraries/ESPAsyncWebServer-master/src/ESPAsyncWebServer.h:33:22: **fatal error: AsyncTCP.h: No such file or directory**
compilation terminated.

So the error seems to be **missing AsyncTCP library**...

nicoff, 2021-07-31:
I just added it, that was the problem! thanks for the check

closed 2022-04-16 by stale[bot]