me-no-dev/ESPAsyncWebServer #1034: Provide example on how to use _tempObject for Body data handling

ttkdroid, 2021-08-13:
I've tried many ways to use the `_tempObject ` to collect JSON data sent as part of the request body, however I've failed on any attempt. 

Could you provide an example on the repo documentation on how to do it?

raspberrypisig, 2021-08-19:
This worked for me.

```c
   server.on("/settings", HTTP_POST, [](AsyncWebServerRequest *request) { 
     
   }, NULL, [](AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total) {        
        if (index == 0) {
          request->_tempObject = new char[len];
          char* tempObject = (char *)request->_tempObject;
          for (int i=0; i<len; i++) {
            tempObject[i] = (char)data[i];
          }
          Serial.println(tempObject);
          request->send(200);
        }
  });

```

Not pretty, but does the job.

For some reason, I couldn't add multiple server.addHandler statements, so I couldn't use the body handling described [here](https://github.com/me-no-dev/ESPAsyncWebServer#json-body-handling-with-arduinojson)

ttkdroid, 2021-08-23:
@raspberrypisig thanks for your answer and I think I found a **sustainable solution**. 

I've defined the endpoints as below, note that the function to handle the body data is the same for every single endpoint.

```
server.on("/endpoin1", HTTP_ANY, function1, NULL, _bodyHandler);
server.on("/endpoint2", HTTP_ANY, function2, NULL, _bodyHandler);
server.on("/endpoint3", HTTP_ANY, function2, NULL, _bodyHandler);
```

Then based on your example I've created the body data handler function, considering that the body data comes fragmented, the tempObject gets enriched for the multiple calls 

```
void _bodyHandler(AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total)
{
    if (request->_tempObject == null)
    {
        request->_tempObject = new char[total + 1];
    }
    char *tempObject = (char *)request->_tempObject;
    for (int i = 0; i < len; i++)
    {
        tempObject[index + i] = (char)data[i];
    }
    tempObject[total] = '\0';
}

```
Then on the function that I want the body data, I just get the pointer from the _tempObject.

```
void function1(AsyncWebServerRequest *request)
{
        if (do_whatever((const char *)request->_tempObject))
        {
            request->send(200, "application/json", "{\"status\":\"whatever done\"}");
        }
        else
        {
            request->send(400, "application/json", "{\"status\":\"whatever error\"}");
        }
        return;
}
```

Works fine for JSON data coming as part of the request body.

closed 2021-08-23 by ttkdroid