me-no-dev/ESPAsyncWebServer #1035: Disable request body data chunking or how to

ttkdroid, 2021-08-13:
I have created this endpoint on my code as I want to handle both GET and POST on the same endpoint

server.on("/config", HTTP_ANY, _Config, NULL, _ConfigBody);

```
void _ConfigBody(AsyncWebServerRequest *request, uint8_t *data, size_t len, size_t index, size_t total)
{
    String X;
    for (size_t i = 0; i < len; i++)
    {
        X += (char)data[i];
    }
    logger(X);
    logger(String(len) + ":" + String(index) + ":" + String(total));
}
```

The problem happens when I send JSON data as part of the request body:

```
curl --location --request POST '10.0.0.2/config' \
--header 'Content-Type: application/json' \
--data-raw '{
    "NodeName": "sssssss",
    "Password": "ssssssss",
    "APIKEY": "",
    "Fingerprint" : "89 23 98 EF 44 93 21 5F 6A 25 E2 CE 29 BA 75 E9 5D BE 58 7F",
    "WebHook": {
        "Method": "",
        "URL": ""
    },
    "Wifi": [
        {
            "SSID": "sssss",
            "Password": "ssssss**"
        },
        {
            "SSID": "sssss",
            "Password": "sssssss"
        }
    ],
    "AP": {
        "SSID": "ssssss",
        "Password": "sssssssss"
    }
}'
```

For some reason the request get split/chunked in 2 parts and the `_ConfigBody` callback is called twice. 

**First call**
```
{
    "NodeName": "sssssss",
    "Password": "ssssssss",
    "APIKEY": "",
    "Fingerprint" : "89 23 98 EF 44 93 21 5F 6A 25 E2 CE 29 BA 75 E9 5D BE 58 7F",
    "WebHook": {
        "Method": "",
        "URL": ""
    },
    "Wifi": [
        {
            "SSID":
```
**len:index:total**
265:0:488

**Second Call**
```
 "sssss",
            "Password": "ssssss**"
        },
        {
            "SSID": "sssss",
            "Password": "sssssss"
        }
    ],
    "AP": {
        "SSID": "ssssss",
        "Password": "sssssssss"
    }
}"}
```
**len:index:total**
223:265:488

I would like to know if it is possible to increase the buffer, disable the chunking or how do I know the whole content data is received.

Could anyone give me an example?

SlavenIvanov, 2021-08-30:
I am also facing a similar issue with the AsyncWebSocket, please share if you have found a way around this!

ttkdroid, 2021-08-30:
@SlavenIvanov  - Have a look on this issue: https://github.com/me-no-dev/ESPAsyncWebServer/issues/1034

closed 2022-04-16 by stale[bot]