me-no-dev/ESPAsyncWebServer #1039: serveStatic() does not work from SPIFFS.

dibalavs, 2021-08-16:
Hello.

Framework: framework-arduinoespressif32 3.10006.210326 (1.0.6)
toolchain: toolchain-xtensa32 2.50200.97 (5.2.0)

Try to serve some static files:

```
    server.on("/index", HTTP_ANY, [](AsyncWebServerRequest *request){
        request->send(SPIFFS, "/static/index.html"); // <-- working OK.
    });

    log_e("SPIFFS:%p\n", &SPIFFS);
    server.serveStatic("/", SPIFFS, "/static/index.html");   // <-- does  NOT WORK.
    server.serveStatic("/static/", SPIFFS, "/static/");// .setCacheControl(CACHE_AGE); // <-- does NOT WORK
```

After some investigation. I found that internally AsyncStaticWebHandler create a copy of fs::FS object instead of using existing object by ref.

```
class AsyncStaticWebHandler: public AsyncWebHandler {
   using File = fs::File;
   using FS = fs::FS;
...
public:
    AsyncStaticWebHandler(const char* uri, FS& fs, const char* path, const char* cache_control);
...
  protected:
    FS _fs;  // <-- is it OK? assigning of _fs object will make a COPY.
```

After using locally _fs object by ref. Issue is gone.

```
  protected:
    FS &_fs;  // <-- works ok.
```

Looks strange, because this code does not modifies for 6 years.

closed 2022-04-16 by stale[bot]

Mark-81, 2022-06-16:
Even after this fix it still does not work for me on ESP32:

```
_server.on("/", HTTP_GET, [this](AsyncWebServerRequest *request)
{
  Serial.println("HTTP GET request");
  _server.serveStatic("/", SPIFFS, "/www").setDefaultFile("index.html"); // <-- this NOT work
  //request->send(SPIFFS, "/www/index.html", "text/html"); // <-- this DOES work
});
```

I've just applied the fix above:

```
protected:
    FS &_fs;
```

Is there anything else I should do?

Pablo2048, 2022-06-16:
Well, it's probably because you use it wrong - `serveStatic()` has to be invoked somewhere in `setup` (and just once before `_server.begin()`) , not in the callback...

Pablo2048, 2022-06-16:
an addition: see https://github.com/me-no-dev/ESPAsyncWebServer/blob/f71e3d427b5be9791a8a2c93cf8079792c3a9a26/examples/ESP_AsyncFSBrowser/ESP_AsyncFSBrowser.ino#L153