me-no-dev/ESPAsyncWebServer #1043: Conflict with library LinkedList

aster1cs, 2021-08-21:
I ran into the same problem as [#715](https://github.com/me-no-dev/ESPAsyncWebServer/issues/715).

Since the ESPAsyncWebServer implements its own LinkedList class in StringArray.h, the ESPAsyncWebServer library is incompatible to the LinkedList library. Locally renaming or wrapping either of them probably isn't a good workaround, as every user of the library would need to come up with a fix on their own. Probably a better solution would be to rename the LinkedList class into something more specific to avoid naming collisions.

multivrac, 2022-03-12:
Hi, i did had the same issue yesterday with LinkedList class.
I suggest that you simply replace all LinkedList by EspLinkedList. It worked for me fine and remove all conflict.
BR,

closed 2022-07-10 by stale[bot]

Counfhou, 2023-04-22:
It would be great if this could be tackled, running into this as well