me-no-dev/ESPAsyncWebServer #1046: Send to the HTML an array of data

AlvaroMacipe, 2021-08-24:
Hi!

I want to send with server.on(...) some Strings. 
Like String array[3] = {"A", "B", "C"}

I can send data for one object, but not more. I dont know if I can do a loop or I only can do this by JSON .

Thanks

dduehren, 2021-12-11:
JSON worked for me.  I use JSON to update multiple variables per page using xhhyp.  Here's my html script, for example:
`  function parseResponse(p1){
  jsonResponse = p1;
        document.getElementById("temperature").innerHTML = jsonResponse.TMP;
        document.getElementById("humidity").innerHTML = jsonResponse.HUM;
        document.getElementById("CO2Lvl").innerHTML = jsonResponse.CO2;
        document.getElementById("light").innerHTML = jsonResponse.LED;
        document.getElementById("time").innerHTML = jsonResponse.TIM;
		document.getElementById("dCount").innerHTML = jsonResponse.CNT;
		document.getElementById("error").innerHTML = jsonResponse.ERR;
		document.getElementById("wifi").innerHTML = jsonResponse.WIFI;
  }
  setInterval(function ( ) {
    var xhttp = new XMLHttpRequest();
    xhttp.onreadystatechange = function() {
        if (this.readyState == 4 && this.status == 200){
        var jsonResponse = JSON.parse(this.responseText);
        console.log(jsonResponse);
		parseResponse(jsonResponse);
        }
    };
    xhttp.open("GET", "/Main", true);
    xhttp.send();
  }, 2000 ) ;`

zekageri, 2021-12-11:
Show us your approach and I could help you with code

closed 2022-04-16 by stale[bot]