me-no-dev/ESPAsyncWebServer #1048: Concurrent access of SPIFFS/LITTLEFS causes watchdog timeout

keshav-raj-trekker, 2021-08-25:
# Concurrent access of SPIFFS/LITTLEFS

Hey all,

My code involves a logger task which periodically (500ms) writes to the filesystem.
I also have AsyncWebServer set up to serve a website from the filesystem.

Previously, using AsyncWebServer with SPIFFS, the watchdog timeout would occur without the logger operational and even if I disabled the watchdog on the async_tcp async service task, this task would still hang.

Switching over to LITTLEFS (as I understand it SPIFFS is depreciated) fixed this issue and the watchdog no longer timed out and the response when refreshing the page was snappy.

However after implementing the logger, the exact same issue came back. I am unable to figure out what is actually causing the service task to hang and what event it is waiting for (or has missed).
I put print statements in each of the event checks and the last thing that prints when it hangs is either:
* LWIP_TCP_POLL
* LWIP_TCP_RECV

I understand this is not as helpful as I would like but I am hoping other people have had this issue and are willing to elaborate on it.

zekageri, 2021-08-25:
Post your logger function and we try to help.

keshav-raj-trekker, 2021-08-25:
This function is being called every 500ms, it iterates through all my logging sources and writes a 13 byte entry into a file for every queued log entry in each log source.

```c++
void FS_LOGGER::write_all_logs(LOG_SOURCE *src){
    
    LOG_ENTRY entry;
    uint8_t topic = src->get_id();
    char buf[5+MAX_LOG_PAYLOAD];
    while(src->pop_entry(&entry)){
        memcpy(buf,&entry.timestamp,4);
        memcpy(buf+4,&topic,1);
        memcpy(buf+5,entry.data,MAX_LOG_PAYLOAD);
        File f = LITTLEFS.open(log_file,"r+");
        if(!f){
            continue;
        }
        f.seek(log_count*(4+1+MAX_LOG_PAYLOAD));
        for(int i = 0;i<(5+MAX_LOG_PAYLOAD);i++){
            f.write(buf[i]);
        }
        f.close();
        log_count++;
        if(log_count>=max_log_entries){
            Serial.println("Warning, maximum log entries reached. Overwriting oldest logs");
            log_count=0;
        }
    }
}
```

Pablo2048, 2021-08-25:
Can this be related with https://github.com/me-no-dev/ESPAsyncWebServer/pull/1040 ?

zekageri, 2021-08-25:
Do you call this within an async web callback?

keshav-raj-trekker, 2021-08-25:
I don't, this is running only in a separate task. The issue also occurs regardless of whether the service handler task runs on a different core or the same one.

closed 2022-04-16 by stale[bot]