me-no-dev/ESPAsyncWebServer #1055: Eventsources: problem with limiting sends

holgerlembke, 2021-09-02:
Moin!

I play around with a Lidar and Eventsources. Lidar sends with 153600 baud, my RX buffer is a 1024 bytes. Everything runs on Core 1, ESP: v4.4-dev-2313-gc69f0ec32, ChipModel: ESP32-D0WDQ6, ChipRevision: 0x1, Cores: 2, Features: WiFi802bgn BLE BT

My Eventsource send routine looks like this:

```
if ( (webeventclient) && (webeventclient->connected()) ) {
  if (webeventclient->packetsWaiting() < 2) {
    // kurz halten, datenvolumen senken
    webevents->send(msg.c_str(), "d", millis());
    send++;
  } else {
    dropped++;
  }
}
```

Statistics: 
Average msg length is around 60 bytes.
Send: 112625 Dropped: 83740 Underruns: 23 (underruns happen, when HardwareSerial(1) drops data...) 
heapsize: 361295 freeheap: 200907 minfreeheap: 190655 maxallocheap: 65524

I fetch the webeventclient with this:

```
AsyncEventSource * webevents = NULL;
AsyncEventSourceClient * webeventclient = NULL;

// Handle Web Server Events
void onWebEventConnect(AsyncEventSourceClient *client) {
  client->send("Lidar ready", NULL, millis(), 10000);
  webeventclient = client;
  // Löschbefehl
  webevents->send("", "c", millis());
};

//**********************************************************************************
void setupWebEvent(void) {
  webevents = new AsyncEventSource("/events");
  webevents->onConnect(onWebEventConnect);
  webserver->addHandler(webevents);
}
```

Unfortunately 
* I didn't find a way to get noted when webeventclient somehow is discarded (and I don't know if it ever will happen).
* I didn't find a build in way to limit sending. When I don't limit sending I get all sort of crashes (I assume out-of-memory situations).
* Limiting to 5 results in crashes.

Any advice?

tonydee2377, 2021-09-21:
i think it is also my problem, when I post these lines:

events.send("ping", NULL, millis());
events.send(String(hostname).c_str(), "host", millis());
events.send(String(Ver).c_str(), "revision", millis());
events.send(String(scanner).c_str(), "fnames", millis());
events.send(String(ssid).c_str(), "fname", millis());
events.send(String(password).c_str(), "fpass", millis());
events.send(String(apmode).c_str(), "wifimode", millis()); //Si fermi qui
events.send(String(sidebar).c_str(), "sidebar", millis()); 

from the client I get the first but not the last two, as if it did only the first 5 sends.
I have to divide them into two different mailings

tonydee2377, 2022-01-04:
**holgerlembke**, for this problem, use this solution: 
https://github.com/me-no-dev/ESPAsyncWebServer/issues/1057#issuecomment-940214680
upgrade 2 library. 
I solved it like this!

closed 2022-04-16 by stale[bot]