me-no-dev/ESPAsyncWebServer #1056: This library can be used to build a HTTPS server?

RobsonRojas, 2021-09-03:
I'm working in a project that uses this library to build a HTTP server. Now I have to enable the server to provide secure communication between the browser and my ESP-32. 

Anyone could give any help or tips about it?

zekageri, 2021-09-04:
Unfortunately this lib is not maintained anymore. I don't think that they want to support https. Really sad.

dduehren, 2021-12-11:
I've only started to look into https.  A couple of things are known.
1. The server has to provide a certificate
2. The stuff going over the wire has to be encrypted, so you need a library or something that does that.  I think you could do that before sending it
3. Different ports are used and I don't know if there's a way to say what ports are being listened to.
Here's a link with a bit more info.  It's about a flask server, but your requirements should be similar and there's an expansion of what you need to do.[https://blog.miguelgrinberg.com/post/running-your-flask-application-over-https](url)

zekageri, 2021-12-11:
Https nothing to do with ports. You can encrypt the data and send it in any port. It's just recommended to use port 443 but you can use any. The port can be changed easily anyway. The problem is the handshake part which the async lib should do as your http layer. It must upgrade the connection to https using TLS. And this lib can not do that. There is an approach for this but it works with esp8266 not with esp32.

closed 2022-04-16 by stale[bot]