me-no-dev/ESPAsyncWebServer #1058: Change port in routine?

andrecortica, 2021-09-16:
Its possible change port in my setup routine?

andrecortica, 2021-09-27:
please does anyone have any ideas?

Connect-and-Exchange, 2021-09-30:
`  server = new AsyncWebServer(port);`

andrecortica, 2021-10-02:
> ` server = new AsyncWebServer(port);`

Hi Connect-and-Exchange
Thanks for the help, but the following error appears:

exit status 1
invalid user-defined conversion from 'AsyncWebServer*' to 'const AsyncWebServer&' [-fpermissive]

andrecortica, 2021-10-02:
> ` server = new AsyncWebServer(port);`

Hi @Connect-and-Exchange 
Thanks for the help, but the following error appears:

exit status 1
invalid user-defined conversion from 'AsyncWebServer*' to 'const AsyncWebServer&' [-fpermissive]

vernonet, 2021-12-07:
AsyncWebServer * server;
......
// Start the server
server = new AsyncWebServer(8888);
server->begin();

zekageri, 2021-12-08:
This going to wreck your heap if you allocate two server space

vernonet, 2021-12-08:
AsyncWebServer * server;
......
server = new AsyncWebServer(7777);
server->begin();
.......
.......
delete server;
// Start the server
server = new AsyncWebServer(8888);
server->begin();

andrecortica, 2021-12-12:
Please, what would the following code look like

#include <Arduino.h>
#include <WiFi.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

const char* ssid = "";
const char* password = "";

AsyncWebServer server(80);

void setup(void) {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    request->send(200, "text/plain", "Hi! I am Andre.");
  });

  server.begin();
  Serial.println("HTTP server started");

}

void loop(void) {
  
}

zekageri, 2021-12-13:
Looks normal

vernonet, 2021-12-13:
to **andrecortica**

> ```
> #include <Arduino.h>
> #include <WiFi.h>
> #include <AsyncTCP.h>
> #include <ESPAsyncWebServer.h>
> 
> const char* ssid = "";
> const char* password = "";
> 
> AsyncWebServer *server;
> 
> void setup(void) {
> Serial.begin(115200);
> WiFi.mode(WIFI_STA);
> WiFi.begin(ssid, password);
> Serial.println("");
> 
> // Wait for connection
> while (WiFi.status() != WL_CONNECTED) {
> delay(500);
> Serial.print(".");
> }
> Serial.println("");
> Serial.print("Connected to ");
> Serial.println(ssid);
> Serial.print("IP address: ");
> Serial.println(WiFi.localIP());
> 
> server->on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
> request->send(200, "text/plain", "Hi! I am Andre.");
> });
> 
> server = new AsyncWebServer(80);
> server->begin();
> Serial.println("HTTP server started");
> 
> }
> 
> void loop(void) {
> 
> }
> ```

Pablo2048, 2021-12-13:
@vernonet very bad - you can not do `server->on()` before `server = new AsyncWebServer(80);` actually...

vernonet, 2021-12-13:
yes, I made a mistake :
```
.....
server = new AsyncWebServer(80);
server->on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
request->send(200, "text/plain", "Hi! I am Andre.");
});

server->begin();
Serial.println("HTTP server started");
}

void loop(void) {

}
```

andrecortica, 2022-01-05:
> yes, I made a mistake :
> ```
> .....
> server = new AsyncWebServer(80);
> server->on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
> request->send(200, "text/plain", "Hi! I am Andre.");
> });
> 
> server->begin();
> Serial.println("HTTP server started");
> }
> 
> void loop(void) {
> 
> }
> ```

Hi @vernonet , Thanks a lot for the help. Worked perfectly! 😀😀

closed 2022-01-05 by andrecortica