me-no-dev/ESPAsyncWebServer #1082: Async WebSocket performance

thanh01pmt, 2021-11-01:
Dear @me-no-dev,
I used Async Websocket to stream image file from ESP32-Cam but the ESP-Cam will inform "Too many messages queued" when I wrote  below code:

void loop()
{
  ws.cleanupClients();
  if (isWSConnected) {
    fb = esp_camera_fb_get();
    if (fb) {
      theClient->binary((const char *)fb->buf, fb->len);
      esp_camera_fb_return(fb);
      fb = NULL;
    }
    delay(0);
  } 
}

I tried to fix the problem "Too many messages queued" by changing the delay time in loop from 0 to 35ms. The web page showed the image from websocket but it is really lag and not stable, compare with - may be - the same code but using "ArduinoWebsockets.h" library as below:

void loop()
{
  auto client = server.accept();
  client.onMessage(handle_message);
  while (client.available()) {
    fb = esp_camera_fb_get();
    client.sendBinary((const char *)fb->buf, fb->len);
    esp_camera_fb_return(fb);
    fb = NULL;
 }

I wonder how to fix the problem by using "Async Websocket".
Hope you can help me on this.
Thank you so much

detmot5, 2021-11-06:
Yep I also found websockets completely unusable in real applications. That may be caused by too small maximum amount of messages queued. 
This should be possible to change at runtime or by defining some preprocessor directives.

zekageri, 2021-11-07:
static inline bool _init_async_event_queue(){
    if(!_async_queue){
        _async_queue = xQueueCreate(512, sizeof(lwip_event_packet_t *));
        if(!_async_queue){
            return false;
        }
    }
    return true;
}

sixtyfive, 2021-11-18:
i have this in my `loop()`:

```
  if (online && (t3 + 10000 < millis())) {
    ws.textAll("blah");
    t3 = millis();
  }
```

So there's 10 SECONDS between messages. yet every 30 or 40 seconds or so, it starts sending messages out as fast as they will go through, then of course running into "too many messages queued". How does it even get past my 10s limit?!

Viveka2001, 2021-11-18:
How to download ESPAsyncWebServer library  there is no download option in
GitHub?
Can u send the download link to my mail.

zekageri, 2021-11-18:
[image]
[image]

sixtyfive, 2021-11-18:
Bhaiya you just click on "Download ZIP". It is huge button on bottom of your screenshot!

Viveka2001, 2021-11-18:
I can't find the code option . It's just showing go to file what to do ?

sixtyfive, 2021-11-18:
[image]

closed 2022-04-16 by stale[bot]

oldeng, 2022-04-29:
i also wonder how to solve this problem。I also met for this problem
[image]

hyansuper, 2022-06-27:
`client->binary(data)` only copys the data to its message queue and returns immediately. that's why this lib is called "async".
In the loop() function, you keep feeding client's message queue with data without checking whether data is sent or not.
Also, you shouldn't use websocket client in the loop() function, you should use it in the server callback.
Maybe this lib is not suitable for your video stream application. because `client->binary(data)` copys data, which is not necessrary and unefficient.