me-no-dev/ESPAsyncWebServer #1085: compilation error with ESP 2.01 library

fbell58, 2021-11-11:
i have a sckech witch is compling ok in version 1.6 of  ESP32 library  arduino software (ESP32 Arduino Release 1.0.6 based on ESP-IDF v3.3.5)

I have upgrade to the new 2.0.1 ESP library soft (ESP32 Arduino 2.0.1 based on ESP-IDF 4.4)

and I get the following error ? anay idea of what I can do to solde that ?

_**Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/gcc8_4_0-esp-2021r2/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: libraries/ESPAsyncWebServer-master/WebAuthentication.cpp.o: in function `getMD5(unsigned char*, unsigned short, char*)':
/Volumes/homes-1/NAS_francois/Drive/Arduino/libraries/ESPAsyncWebServer-master/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'**_

fbell58, 2021-11-18:
**I found I think in Arduino new package old mbedtls have been removed (depreciated...)
I change with the new fonction in "WebAuthentication.h" and I can compile this out error ....
It is strange that I am the only one ????

extract from md(. h in new package ....*

_*
MBEDTLS_DEPRECATED void mbedtls_md5_starts( mbedtls_md5_context *ctx );

/**
 * \brief          MD5 process buffer
 *
 * \deprecated     Superseded by mbedtls_md5_update_ret() in 2.7.0
 *
 * \param ctx      MD5 context
 * \param input    buffer holding the data
 * \param ilen     length of the input data
 *
 * \warning        MD5 is considered a weak message digest and its use
 *                 constitutes a security risk. We recommend considering
 *                 stronger message digests instead.
 *
 */
MBEDTLS_DEPRECATED void mbedtls_md5_update( mbedtls_md5_context *ctx,
                                            const unsigned char *input,
                                            size_t ilen );

/**
 * \brief          MD5 final digest
 *
 * \deprecated     Superseded by mbedtls_md5_finish_ret() in 2.7.0
 *
 * \param ctx      MD5 context
 * \param output   MD5 checksum result
 *
 * \warning        MD5 is considered a weak message digest and its use
 *                 constitutes a security risk. We recommend considering
 *                 stronger message digests instead.
 *
 */
MBEDTLS_DEPRECATED void mbedtls_md5_finish( mbedtls_md5_context *ctx,
                                            unsigned char output[16] );

/**
 * \brief          MD5 process data block (internal use only)
 *
 * \deprecated     Superseded by mbedtls_internal_md5_process() in 2.7.0
 *
 * \param ctx      MD5 context
 * \param data     buffer holding one block of data
 *
 * \warning        MD5 is considered a weak message digest and its use
 *                 constitutes a security risk. We recommend considering
 *                 stronger message digests instead.
 *
 */
MBEDTLS_DEPRECATED void mbedtls_md5_process( mbedtls_md5_context *ctx,
                                             const unsigned char data[64] );

#undef MBEDTLS_DEPRECATED
#endif /* !MBEDTLS_DEPRECATED_REMOVED */_

beats0126, 2021-12-01:
meet the same issue

fbell58, 2021-12-01:
> meet the same issue

As I try to explain You can try to solve by replacing obsolet call by the new one ...I did not test all function but for me there is no compilation issue and the server is working fine .....

weoiss1998, 2022-01-06:
Hi, sorry to ask. @fbell58 what do I have to do exactly?

fbell58, 2022-01-10:
> Hi, sorry to ask. @fbell58 what do I have to do exactly?

you must modify WebAuthentication.cpp as this (it is working for me but of course i did not make all the possible tests for any specific case ....)

you modify in the library directory  your_directory/Arduino/libraries/ESPAsyncWebServer-master/src/WebAuthentication.cpp

line 72  with  this  (the comment are the old line ......)

#ifdef ESP32
  mbedtls_md5_init(&_ctx);
  mbedtls_md5_update_ret (&_ctx,data,len);
  mbedtls_md5_finish_ret(&_ctx,data);
  mbedtls_internal_md5_process( &_ctx ,data);
//  mbedtls_md5_starts(&_ctx);
//  mbedtls_md5_update(&_ctx, data, len);
//  mbedtls_md5_finish(&_ctx, _buf);

weoiss1998, 2022-01-13:
ok, thank you

drmpf, 2022-03-07:
Now could be a library version number issue.
There was no increase in this library's version when the mbed fix was made so no indication in Arduino library manager that you don't have the latest code.

MartinVerges, 2022-05-08:
Thank you @fbell58

MartinVerges, 2022-07-10:
needs to be fixed, not stale!

Renato-Sanchez, 2023-05-01:
ainda errado

per1234, 2024-12-17:
This has been fixed by https://github.com/me-no-dev/ESPAsyncWebServer/commit/17a8d442169f3ca8496b20ebd24bbc06f55d4a62

@fbell58 install the latest version of the library from this GitHub repository (make sure you don't install one of the forks which haven't pulled down the fix), then check to see if the bug is fixed. If so, you can close the issue by clicking the "**Close issue**" button here on the issue web page.