me-no-dev/ESPAsyncWebServer #1086: Large file upload behaviour

dakhnod, 2021-11-14:
I am building an application where an endpoint on the ESP will be used to u pload rather big files (~2kb).
The files will certainly be bigger than  the ESP ram.
Still, I know that the upload callback is only called with chunks of the uploaded file.
Now I would be extremely thankful for clarification with the following questions:
- what determines the size of the uploaded chunks?
- what will happen if the upload callback takes too much time?
- how much time would too much be?
- Is there a way to make the client wait, before sending a new chunk of data?

zekageri, 2021-11-14:
The client will be written in js? 

I think the available free ram determines the chunk sizes.

If the callback timeouts the esp will send a bad request back to client or it crashes. Depending how the timeout is handled. Most cases it will be an 500 server error

Callback timeout is defined in the header but it depends on the wdt and task switch periods too

Iam uploading 100, even 200-300 kb files to my esp. My client is js and working with ajax post requests. It can even upload several files, buffered on client side and sending one by one. 

Iam using websocket and buffering the chunks in the external ram memory before writing it to the file system.

But you can use regular endpoint callback too. Just write every chunk directly to the FS as the esp gets it and you don't have to worry about the ram.

dakhnod, 2021-11-14:
@zekageri thanks, great answer.
My client will probably just be curl uploading a file.
What I still don't quite understand is
- what is holding the client back from just sending the whole file at one, thus overflowing the ESP ram?
- let's say the upload endpoint takes it's time to write the chunk somewhere, will the client wait for the callback before sendin the next chunk?

zekageri, 2021-11-15:
The client sends it's data in its own chunks. The TCP layer handles the server own chunking. For example the curl client sends the data in lets say 1024 bytes. The esp gets the data one byte per one byte and puts it into a predefined buffer. If this buffer is full it sends an event or callback of some sort to the server implementation for the esp higher layer to do something with it. But it can happen that the esp's internal buffer is just 512 byte. So one chunk is not fit into it. So the TCP layer notifies the client how many bytes written after it sends an event to higher layer. The client's TCP layer gets the size written and sends an other chunk starting from that size.

That info is not complete and based on my own understanding. 

Since the esp must put the data into its own ram this must be most relevant solution. In the end after all data is written ( determines by the timeout between chunks and the server response code ) the TCP layer will do a final size check. 

In fact it will do a size check for every chunk.

closed 2022-04-16 by stale[bot]