me-no-dev/ESPAsyncWebServer #1087: Webserver only sends parts of html code

Linzm99, 2021-11-14:
Hi,
I am using an esp8266 with asyncwebserver as an iot device. Additionally I am using OTA flashing, a NTP client to get time information and an http client to get weather information.
After booting everything works fine but after some time when loading the page from the webserver(for example with chrome on my pc) the esp just returns parts of the original html code that should be returned. I also have issues pinging the esp but only for a short time. Then the ping works again, but not the webserver.
The main loop code of the esp keeps running all the time.

What could be the problem?

alixti, 2022-01-17:
Once this happened to me because the ESP didn't have enough Wi-Fi signal. When I moved it closer to the AP it worked fine.

closed 2022-04-16 by stale[bot]

JohnvanderPol2, 2022-05-03:
I had the same problem,
But the problem seems to be in OTA (AsyncElegantOTA)

I have code that is like
`void TConfigurationWebHandler::RequestHandler(AsyncWebServerRequest* request)
  {
  DBG_OUTPUT_PORT.println(F("TConfigurationWebHandler::RequestHandler"));
  if(request->method()==HTTP_POST)
    {
    }
  else 
    {
    TConfigurationWebHandler handler;
    String json=handler.getConfigurationData();//So the content is not released to early
    DBG_OUTPUT_PORT.println(F("before send"));
    DBG_OUTPUT_PORT.println(json);
    request->send(200, "text/json", json);
    DBG_OUTPUT_PORT.println(F("after send"));
    DBG_OUTPUT_PORT.println(json);
    }
  }
 `
 
  When OTA was enabled, after the send, the content of the json would be cleared.
  This while the string is passed as const String &
  
I am still investigating why OTA causes this behavior.