me-no-dev/ESPAsyncWebServer #1089: New Safari browser connect Websocket cause CORRUPT HEAP

moononournation, 2021-11-21:
SYMPTOM

- testing ESPAsyncWebServer ESP_AsyncFSBrowser example
- Tried iPhone SE gen1 and desktop Chrome and Firefox browser does not have this problems
- new iPhone and desktop Safari cause below reboot once connect

```
rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 14 - AP_START
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:391] _eventCallback(): Reason: 203 - ASSOC_FAIL
STA: Failed!
[I][ArduinoOTA.cpp:141] begin(): OTA server at: esp-async.local:3232
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:374] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:419] _eventCallback(): STA IP: 192.168.1.12, MASK: 255.255.255.0, GW: 192.168.1.1
ws[/ws][1] connect
ws[/ws][1] pong[0]: 
ws[/ws][1] binary-message start
ws[/ws][1] frame[0] start[0]
ws[/ws][1] frame[0] binary[0 - 0]: 
ws[/ws][1] frame[0] end[0]
ws[/ws][1] binary-message start
ws[/ws][1] frame[0] start[0]
ws[/ws][1] frame[0] binary[0 - 0]: 
ws[/ws][1] frame[0] end[0]
ws[/ws][1] binary-message start
ws[/ws][1] frame[0] start[0]
ws[/ws][1] frame[0] binary[0 - 0]: 
ws[/ws][1] frame[0] end[0]
ws[/ws][1] binary-message start
ws[/ws][1] frame[0] start[0]
ws[/ws][1] frame[0] binary[0 - 0]: 
ws[/ws][1] frame[0] end[0]
ws[/ws][1] binary-message start
ws[/ws][1] frame[0] start[49]
ws[/ws][1] frame[0] binary[0 - 49]: CORRUPT HEAP: Bad head at 0x3ffcf0f8. Expected 0xabba1234 got 0x3ffcf2a4
assertion "head != NULL" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/heap/multi_heap_poisoning.c", line 249, function: multi_heap_realloc
abort() was called at PC 0x400fe88b on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x400887c8:0x3ffd4360 0x40088a45:0x3ffd4380 0x400fe88b:0x3ffd43a0 0x4008d179:0x3ffd43d0 0x40081cb4:0x3ffd43f0 0x40081d1d:0x3ffd4410 0x400868f9:0x3ffd4430 0x4000bedd:0x3ffd4450 0x400e34ef:0x3ffd4470 0x400e3604:0x3ffd44a0 0x400e387b:0x3ffd44c0 0x400e3989:0x3ffd44e0 0x400d17fb:0x3ffd4500 0x40161e33:0x3ffd4560 0x401623b6:0x3ffd4580 0x400d9ec5:0x3ffd45d0 0x400d9ee1:0x3ffd4610 0x400d7049:0x3ffd4630 0x400d70c9:0x3ffd4670 0x400d77cd:0x3ffd4690 0x40089a56:0x3ffd46c0

Rebooting...
ets Jun  8 2016 00:22:57
```

moononournation, 2021-11-21:
Tried Arduino-esp32 v1.0.4, v1.0.5, v1.0.6 and v2.0.1 all have this problems.

arjenhiemstra, 2021-11-27:
I've combined a few patches from other forks which solved the safari issue for me:
https://github.com/arjenhiemstra/ESPAsyncWebServer

moononournation, 2021-11-27:
It works!

moononournation, 2021-11-28:
I found @arjenhiemstra code added some lock mechanism, seems original message queue is not thread safe.

closed 2022-04-16 by stale[bot]

teknoel, 2022-08-23:
is this related to https://github.com/me-no-dev/ESPAsyncWebServer/issues/1201