me-no-dev/ESPAsyncWebServer #1092: Seeking advice

dduehren, 2021-12-07:
Is there a way to stream/chunk upload several files on the 8266 to a host machine.  I have a log file system that can store several days of log information.  I can upload these files individually but I manage them in a round-robin way so knowing which file is first is a hassle.  I'd like to stitch them together on the fly if possible and in order.  So I'm wondering what might the best way to do this.  I have basic request->send(LittleFS, filepath, String(), true working.
Any insights would be very much appreciated.

zekageri, 2021-12-07:
If the host is always running, I would implement a node js or any other server on it. Then the esp could do http request or open a socket connection to the host and stream the file. If the host is on local there is no need for https, so there were no overhead.

dduehren, 2021-12-07:
Write now the 8266 is hosting a web site that I can attach to with a browser on my computer or phone.  Ideally I'd like to just be able to download the file to my computer, if that were at all possible.  I have a server that I use for OTA and I have thought about implementing a route to do the kind of thing you're talking about or that could automatically get the individual files and stitch them together on the server, and then I'd have to transfer it from the server.

dduehren, 2021-12-07:
Should have been Right now not Write Now

zekageri, 2021-12-08:
So would you download files on your own from the esp with a webserver or would you like to save the files automatically to a host server from the esp? I don't get it.

dduehren, 2021-12-09:
I'm going to do a server based implementation and combine the rolling logs on the server with some simple python code.  As for downloading to my computer, I'll also combine the rolling logs on there.  These seem like the easiest to do.  Thanks for the suggestions.

dduehren, 2021-12-10:
See my other issue, because I'm having trouble uploading the logs with a pointer that gets corrupted.

closed 2022-04-16 by stale[bot]