me-no-dev/ESPAsyncWebServer #1093: request_>send(LittleFS, filepath, String(), true); causes repeat of code

dduehren, 2021-12-09:
I am trying to understand and fix this problem and wondering if I should use something like handleUpload instead.
When I use the response above, it runs through the routine a 2nd time, as though the route was called again, but this doesn't` happen if I serve a normal HTML page. The code is below:
`< server.on("/LogB", HTTP_GET, [](AsyncWebServerRequest *request){ 
      int params = request->params();
      Debug.print(DBG_DEBUG,F("LogBtn route"));

      for (uint8_t i=0; i<params; i++){
        AsyncWebParameter *p = request->getParam(i);
        Debug.print(DBG_INFO, "Name: %s : %s", p->name().c_str(), p->value().c_str());
      }

      uint8_t level=-1;
      String filepath;
      //because these are lists, there really isn't a need to test hasParam.
      if(request->hasParam("loglevel")){
      inputMessage=request->getParam(F("loglevel"))->value();
      level = inputMessage.toInt();
      if (level != 10) {
       mlog.setLogLevel(level);
           saveFlag = true;
        } else {
       Debug.print(DBG_INFO,F("Log level is 10"));
      }
      }
      if(request->hasParam(F("logfile"))){
      inputMessage=request->getParam(F("logfile"))->value();
      level = inputMessage.toInt();
      saveFlag = true;
      filepath="";
      if (level != -1) {
        switch(level){
          case 0:
          if(logfilecnt == 0){
              intvar = mlog.getLogFilePtr();  // to prevent pointer change during upload. Means can't change web pages in middle of uploading
          }
          if (logfilecnt == 16){
            filepath = FPSTR(string_Mushlog);
          } else {
            uint8_t fileno = intvar + logfilecnt;
            if (fileno > 15){
              fileno = fileno - 16;
              }
          Debug.print(DBG_INFO, F("File No after: %d, intvar %d, logfilecnt %d"), fileno, intvar, logfilecnt);
          filepath = mlog.getLogFile(fileno); //file index 0 to 15.
          }
//          Debug.print(DBG_INFO, F("Logfilecnt before ++ %d"), logfilecnt);
          logfilecnt++;
          if (logfilecnt > 16){
            logfilecnt = 0;
            }
//          Debug.print(DBG_INFO, F("Logfilecnt after ++ %d"), logfilecnt);
          break;
          case 1:
          filepath = F("/config.json");
          break;
          case 2:
          filepath = F("/fileDir.txt");
          break;
          default:
           filepath ="";
           break;
           } // End of switch
         } //End of level -1

        if(LittleFS.exists(filepath)){
        Debug.print(DBG_INFO, F("File download %s"), filepath.c_str());
//        request->send(LittleFS, filepath, String(),true);
        } else {
        Debug.print(DBG_ERROR,F("filepath does not exist %s"), filepath.c_str());
        request->send(LittleFS,FPSTR(string_LogHTML), String(),false);
        }
      }//end of if logfile
  });>`

dduehren, 2021-12-11:
At this point in time, I've put a time filter on the code

> if(timeClient.getEpochTime()-epochTime <= 1){
        request->send(LittleFS,FPSTR(string_LogHTML), String(),false);
      } else {
      epochTime = timeClient.getEpochTime();
.....

zekageri, 2021-12-11:
If the request body is too large to fit in the ram, the handler will be called as much as the chunks of the data. For example if the client do a http request to your esp and sends a data of 1024 byte and you can fit 512 byte in your ram, the request will be handled twice because the server splits the data into two chunks.

zekageri, 2021-12-11:
I did not looked into your ( wrongly formatted ) code too much but check the client side script.

dduehren, 2021-12-11:
Below is a wire-shark capture that shows the host doing two requests, even though it got what looks like a proper 200 OK was sent.  Below is the wire-shark output and the corresponding HTML file served by the 8266.
[image]

And the HTML

`<!DOCTYPE html>
<html lang="en">
<head>
<meta charset="utf-8"/>
  <title>MushBox Config Server</title>
  <meta name="viewport" content="width=device-width, initial-scale=1">
  <link rel="stylesheet" type="text/css" href="style.css">
</head>
<body>
  <h1>LogData</h1>
  <div class="topnav">
  <a  href="index.html">Main</a>
  <a  href="Utl.html">Utilities</a>
</div>
<div class="rcenter">
<label for="logs">Select Log Level: %LOGLEVEL%</label>
<select id="logenable" name="loglevel" form="logging">
  <option value="10" selected="True">No change</option>
  <option value="-1">Disable Logging</option>
  <option value="0">Log Input</option>
  <option value="1">Log Inputs + Controls</option>
  <option value="2">Log Inputs + Controls + Ops</option>
  <option value="3">Log All</option>
</select>
<label for="logs">File download: </label>
Download log files by repeatedly selecting "Next Log File."
There are 17 log files to download.  Repeat selection until File Counter returns to initial value.
<p>Current file: <span id="filePtr">%filePtr%</span>
<br>File Counter: <span id="fileCnt">%fileCnt%</span><p>
<select id="logfileselect" name="logfile" form="logging">
  <option value="-1" selected="True">No file selected</option>
  <option value="0">Next Log File</option>
  <option value="1">Config File</option>
  <option value="2">File Directory</option>
  <option value="3">Log Array file</option>
</select>
<hr>
<form action="/LogB" method ="get" id="logging">
    <p><button type="submit" class="submitBtn">Submit</button></p>
</form>
</div>
</body>
<script>
  function parseResponse(p1){
  jsonResponse = p1;
        document.getElementById("filePtr").innerHTML = jsonResponse.FPTR;
        document.getElementById("fileCnt").innerHTML = jsonResponse.FCNT;
  }
  setInterval(function ( ) {
    var xhttp = new XMLHttpRequest();
    xhttp.onreadystatechange = function() {
        if (this.readyState == 4 && this.status == 200){
        var jsonResponse = JSON.parse(this.responseText);
        console.log(jsonResponse);
		parseResponse(jsonResponse);
        }
    };
    xhttp.open("GET", "/fPtr", true);
    xhttp.send();
  }, 2000 ) ;
</script>
</html>`
[image]

And the HTML

``

closed 2022-04-16 by stale[bot]