me-no-dev/ESPAsyncWebServer #1094: How to move url list outisde setup function

StefanoTesla, 2021-12-10:
Hi all, I need a little help for a easy thing that I can't do and I don't know why.

`AsyncWebServer server(80);

void setup()
{

  server.onNotFound(notFound);
  
a lot of url here and other stuff`

My ideas was:

`AsyncWebServer server(80);

void setup()
{

 ServerUrl();

`

and server.h

`
Void ServerUrl(){
list of url here
}
`

I would like to move all the url to another .h file, how can  I do?? If i move urls I get error "function "server" is not declared in this scope.

Thank for help

marcboon, 2021-12-10:
I you need to reference a global var from within another file than where it was declared/defined, you have to add a
```
extern <Type> var;
```
in the file where you need to reference it.
So for you it would probably be
```
extern ESPAsyncWebServer server;

void ServerUrl() {
  server.on(...);
}
```
in your file server.h

zekageri, 2021-12-10:
**You must declare the ESPAsyncWebServer object before you reference it in a file.**
_If you include a .h file in your main (.cpp or .ino) file, you must declare the server object before the include._ 

**Something like this:**

```C
AsyncWebServer server(80);
#include <Another_H_File.h>
void setup(){
 ServerUrl();
}

void loop(){
}
```

And in your external .h file you can do things like this:

```C
void ServerUrl(){
   //list of url here...
}
```

dduehren, 2021-12-11:
I don't know if this is what you're looking for, but what I did was have a function called startServer() that I called in Setup.  startServer() is defined in a separate .cpp file and contains all of the server.on code.

zekageri, 2021-12-11:
What do you mean not worked? What is the error you got?

closed 2021-12-13 by StefanoTesla