me-no-dev/ESPAsyncWebServer #1095: How change http code in a AsyncResponseStream??

StefanoTesla, 2021-12-13:
Hello everyone, 

Sincerly I have no idea why I'm using this, on stackoverflow raccomended me this to concatenate response:

I cannot use ArduinoJson library, and I'm trying to don't fall in ram fragmentation

Anyway, in this case I need to answer this but with http code 400

```
Alpserver.on("api/v1/dome/0/commandbool",                                               HTTP_PUT, [](AsyncWebServerRequest *request){
    GetClientTransactionID(request);
    AsyncResponseStream *response = request->beginResponseStream("application/json");
    response->print("{ ");
    response->print(clientTID);
    response->printf("%d, ", ClientTransactionID);
    response->print(serverTID);
    response->printf("%d, ", AlpServerID);
    response->print(methodNotImp);
    response->print("}");
    request->send(response);
});
```

I've tried this but I get error
AsyncResponseStream *response = request->beginResponseStream(400,"application/json");

And what mean AsyncResponseStream ??

StefanoTesla, 2021-12-14:
This json is valid cause spaces and " are stored in const char[], clientTID is a const char[] and is valued "\"data\": " and if you look olso after %d, there's a space...anyway json is validated from server and it's ok...

So, only in Async ResponseStrem I can do response->print, right?

marcboon, 2021-12-14:
If I understand it correctly, your problem is that you want to send a response code of 400 instead of 200?
You can just use:
```
request->send(400, "application/json", response);
```
If you don't  want to use ArduinoJSON, then your json response can be built as a String like so (using your code as example):
```
String response = "{";
response += clientTID;
response += ClientTransactionID;
response += serverTID;
response += AlpServerID;
response += methodNotImp;
response += "}";
```

StefanoTesla, 2021-12-15:
> If I understand it correctly, your problem is that you want to send a response code of 400 instead of 200? You can just use:
> 
> ```
> request->send(400, "application/json", response);
> ```
> 

If you use AsyncResponseStream way you cannot change http code cause is hardcoded in library, but is not a problem 👍🏻

closed 2021-12-15 by StefanoTesla