me-no-dev/ESPAsyncWebServer #1099: Port 80 i slower than others?

StefanoTesla, 2021-12-15:
I'm having a strange problem, but I don't know if is a library or browsers problem.

I have two server on esp32

one in port 80
one in port 4567

For to see strange "thing" look to the request time:
Port 80:
[image]

Port 4567:
[image]

The code is the same, but port 80 seem to be slower a lot, why? 
I was thinking about Chrome or Edge that have same enginge, but olso Firefox is slow

Any idea??

zekageri, 2021-12-15:
Hmm... Strange. I think it's a router problem other then esp problem. I will test it too on my side tomorrow. Iam curious.

StefanoTesla, 2021-12-16:
> Hmm... Strange. I think it's a router problem other then esp problem. I will test it too on my side tomorrow. Iam curious.

please let me know, at the moment i'm using smartphone as router, but I don't understand why I should be thi s problem sincerly and how understand where's the bottleneck

StefanoTesla, 2021-12-16:
I found the problem, olso if I don't understand the reasron this is(was) part of my code:
```
 server.serveStatic("/", SPIFFS, "/").setCacheControl("max-age=31536000");
  server.onNotFound(notFound);
  server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send(SPIFFS, "/index.html", "text/html");
  });

  server.on("/getswitchstate",            HTTP_GET, [](AsyncWebServerRequest *request) {
    int i;
    AsyncResponseStream *response = request->beginResponseStream("application/json");
    Serial.println(millis());
    response->print("{ \"State\": [");

    for (i=0;i<_MAX_SWTICH_ID_;i++){
        Switch[i].analog ? response->printf("%d",Switch[i].anaValue) : response->printf("%d",digitalRead(Switch[i].pin));
        if (i != _MAX_SWTICH_ID_ -1){
          response->printf(", ");
        }
    }
    response->print("]}");
    Serial.println(millis());
    request->send(response);
});
```

Moving server.on("/getswitchstate",  before the server.serveStatic give my request fast as in port 4567! 

```
  server.on("/getswitchstate",            HTTP_GET, [](AsyncWebServerRequest *request) {
    int i;
    AsyncResponseStream *response = request->beginResponseStream("application/json");
    Serial.println(millis());
    response->print("{ \"State\": [");

    for (i=0;i<_MAX_SWTICH_ID_;i++){
        Switch[i].analog ? response->printf("%d",Switch[i].anaValue) : response->printf("%d",digitalRead(Switch[i].pin));
        if (i != _MAX_SWTICH_ID_ -1){
          response->printf(", ");
        }
    }
    response->print("]}");
    Serial.println(millis());
    request->send(response);
});

 server.serveStatic("/", SPIFFS, "/").setCacheControl("max-age=31536000");
  server.onNotFound(notFound);
  server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
    request->send(SPIFFS, "/index.html", "text/html");
  });

```

I used server server.serveStatic("/", SPIFFS, "/").setCacheControl("max-age=31536000"); 
Thinking it avoiding redownloading pages every time, and **in part it work**, in my css (in /asset folder) file I have cache header but in index.html I don't have cache header..why?

closed 2022-04-16 by stale[bot]