me-no-dev/ESPAsyncWebServer #1100: ESP32 C3 - (Load access fault). Exception was unhandled.

Sys64736, 2021-12-17:
I know it doesn't specifically support this chip but hoping you could shed light on this....
On this line of the captive portal example:   
**Line40:     server.addHandler(new CaptiveRequestHandler()).setFilter(ON_AP_FILTER);//only when requested from AP**
It happens when I try to load the index page from the portal.
If I comment out, there's no exception.
______________________________

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42005322  RA      : 0x42005316  SP      : 0x3fcac2e0  GP      : 0x3fc8da00  
TP      : 0x3fc90d1c  T0      : 0x4005890e  T1      : 0x4038c474  T2      : 0x30312e34  
S0/FP   : 0x00000000  S1      : 0x29f8a984  A0      : 0x3fcadd84  A1      : 0x242f8d3d  
A2      : 0x3fc9ce04  A3      : 0x3fcac9a8  A4      : 0x3fcade94  A5      : 0x29f8a984  
A6      : 0x00000000  A7      : 0x00000000  S2      : 0x3fcadcdc  S3      : 0x3fcadcec  
S4      : 0x29f8a984  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x29f8a984
MHARTID : 0x00000000

Stack memory:
3fcac2e0: 0x0000005f 0x3fc968b4 0x29f8a984 0x00000000 0x00000000 0x00000000 0x00000000 0x030080c0
3fcac300: 0x00000000 0x00000000 0x00000000 0x06000000 0x00000090 0x00000000 0x3fcadcdc 0x3fcadd10
3fcac320: 0x3fcaf137 0x00000002 0x3fcadcdc 0x4200542e 0x42007c18 0x42007c18 0x420033a0 0x3fcadd10
3fcac340: 0x3fcaf137 0x3fcadd10 0x3fcaf137 0x3fcadd10 0x3fcaf137 0x00000002 0x00000001 0x42005514
3fcac360: 0x3fca8430 0x00000000 0x00000001 0x00000001 0x00000016 0x00000004 0x00000001 0x0000000c
3fcac380: 0x00000000 0x0000000c 0x3fc968a0 0x4038b85e 0x00000000 0x00000000 0x00000000 0x00000000
3fcac3a0: 0x00000000 0x3fcacbf8 0x3fcacb24 0x42005798 0x00000000 0x0000000c 0x0000000c 0x420bba12
3fcac3c0: 0x000001ef 0x3fcaef4a 0x3fcacb24 0x3fcadcdc 0x00000000 0x0000000c 0x00000001 0x420bba6e
3fcac3e0: 0x00000000 0x00000000 0x3fcaccb8 0x420bbab2 0x00000000 0x00000000 0x00000000 0x420bbc2e
3fcac400: 0x00000000 0x00000000 0x00000000 0x420bbcac 0x00000000 0x00000000 0x00000000 0x3fcaccb8
3fcac420: 0x00000000 0x00000000 0x00000000 0x40388d3c 0x00000000 0x00000000 0x00000000 0x00000000
3fcac440: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x27146414 0x01010107 0x00000001 0x000000ff
3fcac460: 0x00000000 0x00000000 0x3fe00000 0x00000000 0x3ff00000 0x00000000 0x40000000 0xffc00000
3fcac480: 0x41dfffff 0x94a03595 0x3fdfffff 0x35afe535 0x3fe00000 0x94a03595 0x3fcfffff 0x00000000
3fcac4a0: 0x3fc00000 0x00000000 0x40300000 0x00000000 0x3ff80000 0x636f4361 0x3fd287a7 0x8b60c8b3
3fcac4c0: 0x3fc68a28 0x509f79fb 0x3fd34413 0x00000000 0x40240000 0x00000000 0x401c0000 0x00000000
3fcac4e0: 0x40140000 0x00000000 0x43500000 0x00000000 0x7ff80000 0x7fc00000 0x00000000 0x00000010
3fcac500: 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fcac0e0 0x3fc968b4 0x3fc91d74 0x3fc91d74 0x3fcac50c
3fcac520: 0x3fc91d6c 0x00000016 0x3fca845c 0x3fca845c 0x3fcac50c 0x00000000 0x00000003 0x3fca8508
3fcac540: 0x6e797361 0x63745f63 0x50bb0070 0x00f2353c 0x00000000 0x3fcac500 0x00000003 0x00000000
3fcac560: 0x3fcaccf4 0x4201b82e 0x00000000 0x3fc970cc 0x3fc97134 0x3fc9719c 0x00000000 0x00000000
3fcac580: 0x00000001 0x00000000 0x00000000 0x00000000 0x420bcbc4 0x00000000 0x00000000 0x00000000
3fcac5a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcac5c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcac5e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcac600: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcac620: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcac640: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3fca0000
3fcac660: 0x00000054 0x3fcac664 0x3fcac664 0x3fcac664 0x3fcac664 0x00000000 0x3fcac67c 0xffffffff
3fcac680: 0x3fcac67c 0x3fcac67c 0x00000000 0x3fcac690 0xffffffff 0x3fcac690 0x3fcac690 0x00000000
3fcac6a0: 0x7fffffff 0x00000000 0x0000ffff 0x00000000 0x01980000 0x3fcac65c 0x0000000c 0x00000003
3fcac6c0: 0x3fcacef0 0x00000000 0x00000011 0x3fcade30 0x3fc968b4 0x312f5000 0x3fcac6c4 0x00000072

ELF file SHA256: 0000000000000000

jpswensen, 2022-03-08:
I am seeing the same issue, but for me it is happening anytime I try to serve a file from LittleFS or when I serve a page from a function. The interesting thing is that the crash is happening after the file or page has already been served. Attached below is a fairly simple example.

Some of the WiFi code is a little more complicated than it needs to be because I am testing on an WPA2 Enterprise server, so simplify that if you need to.

`#define LOG_LOCAL_LEVEL ESP_LOG_VERBOSE
#include "esp_log.h"

#include "esp_wpa2.h"
#include "esp_wifi.h"
#include "esp_wpa2.h"
#include <WiFi.h>

#include <Arduino.h>
#include "FS.h"
#include <LittleFS.h>

#ifndef CONFIG_LITTLEFS_FOR_IDF_3_2
 #include <time.h>
#endif

#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>

AsyncWebServer server(80);

#define FORMAT_LITTLEFS_IF_FAILED true

void listDir(fs::FS &fs, const char * dirname, uint8_t levels){
    Serial.printf("Listing directory: %s\r\n", dirname);

    File root = fs.open(dirname);
    if(!root){
        Serial.println("- failed to open directory");
        return;
    }
    if(!root.isDirectory()){
        Serial.println(" - not a directory");
        return;
    }

    File file = root.openNextFile();
    while(file){
        if(file.isDirectory()){
            Serial.print("  DIR : ");

#ifdef CONFIG_LITTLEFS_FOR_IDF_3_2
            Serial.println(file.name());
#else
            Serial.print(file.name());
            time_t t= file.getLastWrite();
            struct tm * tmstruct = localtime(&t);
            Serial.printf("  LAST WRITE: %d-%02d-%02d %02d:%02d:%02d\n",(tmstruct->tm_year)+1900,( tmstruct->tm_mon)+1, tmstruct->tm_mday,tmstruct->tm_hour , tmstruct->tm_min, tmstruct->tm_sec);
#endif

            if(levels){
                listDir(fs, file.name(), levels -1);
            }
        } else {
            Serial.print("  FILE: ");
            Serial.print(file.name());
            Serial.print("  SIZE: ");

#ifdef CONFIG_LITTLEFS_FOR_IDF_3_2
            Serial.println(file.size());
#else
            Serial.print(file.size());
            time_t t= file.getLastWrite();
            struct tm * tmstruct = localtime(&t);
            Serial.printf("  LAST WRITE: %d-%02d-%02d %02d:%02d:%02d\n",(tmstruct->tm_year)+1900,( tmstruct->tm_mon)+1, tmstruct->tm_mday,tmstruct->tm_hour , tmstruct->tm_min, tmstruct->tm_sec);
#endif
        }
        file = root.openNextFile();
    }
}

void notFound(AsyncWebServerRequest *request) {
    request->send(404, "text/plain", "Not found");
}

#define EAP_ANONYMOUS_IDENTITY "anonymous@example.com" //anonymous@example.com, or you can use also nickname@example.com
#define EAP_IDENTITY "nickname@example.com" //nickname@example.com, at some organizations should work nickname only without realm, but it is not recommended
#define EAP_PASSWORD "supersecretpassword" //password for eduroam account

//SSID NAME
const char* ssid = "SSID"; // eduroam SSID

void setup(){
    esp_log_level_set("*", ESP_LOG_ERROR); 

    Serial.begin(115200);
    if(!LittleFS.begin(FORMAT_LITTLEFS_IF_FAILED)){
        Serial.println("LITTLEFS Mount Failed");
        return;
    }
    
    printf("\n");
    delay(2000);

    listDir(LittleFS, "/", 0);

    Serial.println( "Test complete" );
  

    Serial.println();
    Serial.print(F("Connecting to network: "));
    Serial.println(ssid);
    WiFi.disconnect(true);  //disconnect form wifi to set new wifi connection
    WiFi.mode(WIFI_STA); //init wifi mode
    esp_wifi_sta_wpa2_ent_set_identity((uint8_t *)EAP_ANONYMOUS_IDENTITY, strlen(EAP_ANONYMOUS_IDENTITY));
    esp_wifi_sta_wpa2_ent_set_username((uint8_t *)EAP_IDENTITY, strlen(EAP_IDENTITY));
    esp_wifi_sta_wpa2_ent_set_password((uint8_t *)EAP_PASSWORD, strlen(EAP_PASSWORD));
    esp_wifi_sta_wpa2_ent_enable(); //set config settings to enable function
    WiFi.begin(ssid); //connect to wifi
    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(F("."));
    }
    Serial.println("");
    Serial.println(F("WiFi is connected!"));
    Serial.println(F("IP address set: "));
    Serial.println(WiFi.localIP()); //print LAN IP

    server.serveStatic("/", LittleFS, "/").setDefaultFile("index.html");;

    server.onNotFound(notFound);

    server.begin();
}

void loop(){
    vTaskDelay(100);
}`

jpswensen, 2022-03-08:
Additional note, it seems that there is a fix for this that hasn't been merged yet. I will try this fix and report back.

https://github.com/esphome/issues/issues/2257#issuecomment-885661458

https://github.com/me-no-dev/ESPAsyncWebServer/issues/951

closed 2022-07-10 by stale[bot]