me-no-dev/ESPAsyncWebServer #1101: ESP32-C3   ESP_AsyncFSBrowser example compile error  in Arduino 1.8.16 +  esp32-arduino 2.0.1

har-in-air, 2021-12-21:
/home/hari/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp: In member function 'IPAddress AsyncWebSocketClient::remoteIP()':
/home/hari/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp:832:28: error: call of overloaded 'IPAddress(unsigned int)' is ambiguous
         return IPAddress(0U);
                            ^
In file included from /home/hari/.arduino15/packages/esp32/hardware/esp32/2.0.1/cores/esp32/Arduino.h:163,
                 from /home/hari/Arduino/libraries/ESPAsyncWebServer/src/AsyncWebSocket.cpp:21:
/home/hari/.arduino15/packages/esp32/hardware/esp32/2.0.1/cores/esp32/IPAddress.h:51:5: note: candidate: 'IPAddress::IPAddress(const uint8_t*)'
     IPAddress(const uint8_t *address);
     ^~~~~~~~~
/home/hari/.arduino15/packages/esp32/hardware/esp32/2.0.1/cores/esp32/IPAddress.h:50:5: note: candidate: 'IPAddress::IPAddress(uint32_t)'
     IPAddress(uint32_t address);
     

NOTE : FIXED BY replacing IPAddress(0U) with IPAddress((uint32_t)0)

Hieromon, 2022-04-14:
@me-no-dev  A similar issue has been posted in the ESP32 core repo. There you mention a response on the library side. Do you have a plan to fix it ?
https://github.com/espressif/arduino-esp32/issues/5592

har-in-air, 2022-04-14:
return IPAddress((uint32_t)0);

has worked for me. I have been using this fix  for my VSC PlatformIO + Arduino Framework projects using ESP32-C3.

rkakrik, 2022-06-06:
Same problem

DTTerastar, 2022-07-31:
Ditto

QGB, 2022-09-24:
```
"C:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\riscv32-esp-elf-gcc\\gcc8_4_0-esp-2021r2-patch3/bin/riscv32-esp-elf-g++" -DHAVE_CONFIG_H "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -D_GNU_SOURCE "-DIDF_VER=\"v4.4.1-1-gb8050b365e\"" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/config" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/newlib/platform_include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/freertos/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/freertos/include/esp_additions/freertos" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/freertos/port/riscv/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/freertos/include/esp_additions" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_hw_support/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_hw_support/include/soc" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_hw_support/include/soc/esp32c3" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_hw_support/port/esp32c3" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_hw_support/port/esp32c3/private_include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/heap/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/log/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/lwip/include/apps" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/lwip/include/apps/sntp" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/lwip/lwip/src/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/lwip/port/esp32/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/lwip/port/esp32/include/arch" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/soc/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/soc/esp32c3" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/soc/esp32c3/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/hal/esp32c3/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/hal/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/hal/platform_port/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_rom/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_rom/include/esp32c3" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_rom/esp32c3" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_common/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_system/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_system/port/soc" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_system/port/include/riscv" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_system/port/public_compat" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/riscv/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/driver/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/driver/esp32c3/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_pm/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/esp_ringbuf/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\2.0.3/tools/sdk/esp32c3/include/efuse/include" "-IC:\\Users\\qgb\\AppData\\Local\\Arduin [... cut ...]  "C:\\Users\\qgb\\Documents\\Arduino\\libraries\\ESPAsyncWebServer\\src\\WebRequest.cpp" -o "C:\\Users\\qgb\\AppData\\Local\\Temp\\arduino_build_930089\\libraries\\ESPAsyncWebServer\\WebRequest.cpp.o"
使用 2.0.0  版本的库 WiFi 在文件夹： C:\Users\qgb\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.3\libraries\WiFi 
使用 1.1.1  版本的库 AsyncTCP 在文件夹： C:\Users\qgb\Documents\Arduino\libraries\AsyncTCP 
使用 1.2.3  版本的库 ESPAsyncWebServer 在文件夹： C:\Users\qgb\Documents\Arduino\libraries\ESPAsyncWebServer 

```

QGB, 2022-09-24:
安装库依赖
```

In [37]: cd C:\Users\qgb\Documents\Arduino\libraries
C:\Users\qgb\Documents\Arduino\libraries

In [38]: !git clone --config http.proxy=socks5h://127.0.0.1:21080 --depth=1 {U.cbg(e=1)}
https://github.com/me-no-dev/ESPAsyncWebServer
Cloning into 'ESPAsyncWebServer'...
remote: Enumerating objects: 61, done.
remote: Counting objects: 100% (61/61), done.
remote: Compressing objects: 100% (53/53), done.
remote: Total 61 (delta 3), reused 39 (delta 1), pack-reused 0
Unpacking objects: 100% (61/61), done.
Checking connectivity... done.

In [39]: !git clone --config http.proxy=socks5h://127.0.0.1:21080 --depth=1 {U.cbg(e=1)}
https://github.com/me-no-dev/AsyncTCP
Cloning into 'AsyncTCP'...
remote: Enumerating objects: 23, done.
remote: Counting objects: 100% (23/23), done.
remote: Compressing objects: 100% (21/21), done.
remote: Total 23 (delta 0), reused 14 (delta 0), pack-reused 0
Unpacking objects: 100% (23/23), done.
Checking connectivity... done.

In [40]: !git clone --config http.proxy=socks5h://127.0.0.1:21080 --depth=1 {U.cbg(e=1)}
https://github.com/Links2004/arduinoWebSockets
Cloning into 'arduinoWebSockets'...
remote: Enumerating objects: 94, done.
remote: Counting objects: 100% (94/94), done.
remote: Compressing objects: 100% (83/83), done.
remote: Total 94 (delta 12), reused 55 (delta 2), pack-reused 0
Unpacking objects: 100% (94/94), done.
Checking connectivity... done.

```

QGB, 2022-09-24:
```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4ab8657f6f5aa80b81c7dd649db3bd64a98999ed7c65acc5a2ab77ce80caf525
Expected: 6babbc7b9ea63e04e0fc89ec224f6268aa3f84392018f73ab33721f768ce2eb6
Attempting to boot anyway...
entry 0x403ce000
[   194][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=8088, ...)
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x8 (TG1WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40380082
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6a8
load:0x403d0000,len:0x2358
SHA-256 comparison failed:
Calculated: 4ab8657f6f5aa80b81c7dd649db3bd64a98999ed7c65acc5a2ab77ce80caf525
Expected: 6babbc7b9ea63e04e0fc89ec224f6268aa3f84392018f73ab33721f768ce2eb6
Attempting to boot anyway...
entry 0x403ce000
[   199][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=8088, ...)

```
编译烧写成功，运行出错，无限循环

dlarue, 2023-02-16:
using arduino-esp32 v2.0.6 and problem still exists for esp32-c3  and same fix applied and works.
~/Arduino/libraries/ESPAsyncWebSvr/src/AsyncWebSocket.cpp
line: 832
change from: return IPAddress(0U);
change to: return IPAddress((uint32_t)0);

iamjlm, 2023-04-07:
Same issue, with same solution: "return IPAddress((uint32_t)0);"

dlarue, 2023-06-18:
ping to keep this open

L1cardo, 2023-06-20:
I mean, hello? It has been almost 2 years

dlarue, 2023-06-20:
@L1cardo well, I just looked at the top level of this repo and it's not been touched in 2 years so the developer isn't updating it. I was going to do a pull request which just patches that one line but it'll likely not get merged. Not sure of what the solution is at this point other than everyone patching themselves.

moeburn, 2024-03-21:
Just had this issue on my new C3, does anyone know if the fix breaks support for the other boards?