me-no-dev/ESPAsyncWebServer #1106: webpage failed

vatsake, 2022-01-03:
When restarting ESP, the first time the page loads, it responds this:

`Òrˆ½ÿ?€HTTP/1.1 200 OK Content-Length: 1703 Content-Type: text/html Connection: close Accept-Ranges: none Òrˆ½ÿ?€HTTP/1.1 200 OK Content-Length: 1703 Content-Type: text/html Connection: close Accept-Ranges: none Òrˆ½ÿ?€HTTP/1.1 200 OK Content-Length: 1703 Content-Type: text/html Connection: closÝ—ƒ^Ò˜@äÛ@˜@˜@Œ@àÛ@DÜ@DÜ@˜@˜@DÜ@DÜ@@@@@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@˜@˜@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@DÜ@ }, series: [{ showInLegend: false, data: [[0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], [0,0.00], ] }], plotOptions: { line: { animation: true, dataLabels: { enabled: true } }, series: { color: '#059e8a' } }, xAxis: { type: 'datetime', }, yAxis: { title: { text: 'â‚¬' } }, credits: { enabled: false } });`

XBoter, 2022-01-06:
Im not quit sure where your Problem comes from without your source Code. But it looks familiar to a problem i had. Which method are you using for sending the data to the client? I had the same phenomenon but i used the send() method instead of the send_P() with my web page which was located in the flash and was relative large.

vatsake, 2022-01-07:
Thank you for your response!
My webpage is in PROGMEM, but I read it to a string variable, then I can replace some variables in html.
I use send() not send_P().

```
  server.on("/", HTTP_GET, [](AsyncWebServerRequest * request) {
    String body = FPSTR(HTTP_BODY);
    
    body.replace("{{time}}", TimeString()); //Returns time in 00:00 format
    ...
    ...
    request->send(200, "text/html", body);
  });
```

XBoter, 2022-01-07:
The send_P() method has a template processor that is doing the same as your String replacment. Maybe try this with the send_P() method. More infos are in the README under Template processing.
```
String processor(const String& var)
{
  if(var == "HELLO_FROM_TEMPLATE")
    return F("Hello world!");
  return String();
}

// ...

const char index_html[] PROGMEM = "..."; // large char array, tested with 14k
request->send_P(200, "text/html", index_html, processor);
```

vatsake, 2022-01-07:
I tried using it, but this creates 'chunked response', and I couldn't understand it. - Web page doesn't load completely.
What is **mysource** in chunked response example?
```

AsyncWebServerResponse *response = request->beginChunkedResponse("text/plain", [](uint8_t *buffer, size_t maxLen, size_t index) -> size_t {
  return mySource.read(buffer, maxLen);
}, processor);
```

closed 2022-01-07 by vatsake