me-no-dev/ESPAsyncWebServer #1107: ESP8266 crashes when calling AsyncWebServer reset() or removeHandler() Method inside a class instance

XBoter, 2022-01-06:
I first encountered the problem in my main Project, and made a little Arduino test Sketch to test if it has nothing to do with my other code.
When calling the reset() or removeHandler() Method of the AsyncWebServer instance inside the class A instance the ESP crashes with what looks like a malloc exception. 

Here is the Arduino Code im using to reproduce the error.
```
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>

class A {
  public:
    A() {}

    // ================ Data ================ //
  private:
    int state = 0;
    AsyncWebServer asyncWebServer = AsyncWebServer(80);
    AsyncWebSocket asyncWebSocket = AsyncWebSocket("/ws");

    // ================ Methods ================ //
  private:
    void WebpageEvent(AsyncWebServerRequest *request) {};
    void WebsocketEvent(AsyncWebSocket *server, AsyncWebSocketClient *client, AwsEventType type, void *arg, uint8_t *data, size_t len) {};

  public:
    void Init() {
      asyncWebServer.on("/", HTTP_ANY, std::bind(&A::WebpageEvent, this, std::placeholders::_1));
      asyncWebSocket.onEvent(std::bind(&A::WebsocketEvent, this, std::placeholders::_1, std::placeholders::_2, std::placeholders::_3, std::placeholders::_4, std::placeholders::_5, std::placeholders::_6));
      asyncWebServer.addHandler(&asyncWebSocket);
      asyncWebServer.begin();
    };
    void Run() {
      switch (state) {
        case 0:
          asyncWebServer.removeHandler(&asyncWebSocket);
          //asyncWebServer.reset();
          state++;
          break;
      };
    };
};

A a = A();

void setup() {
  Serial.begin(115200);
  Serial.println("//================ START ================//");
  a.Init();
};

void loop() {
  a.Run();
};

```
Here are the messages from the Terminal.
```
//================ START ================//

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic umm_malloc.cpp:229 umm_get_ptr_context

>>>stack>>>

ctx: cont
sp: 3ffffeb0 end: 3fffffc0 offset: 0000
3ffffeb0:  3fffff60 3ffefcac 3ffefd6c 4010041c  
3ffffec0:  000000fe 00000000 00000000 00000000  
3ffffed0:  00000000 00000000 00000000 3ffeeeb4  
3ffffee0:  3ffef610 00000000 3ffe8600 3ffeee6c  
3ffffef0:  3fffdad0 00000020 3ffeec50 402089d2  
3fffff00:  00000000 00000000 3ffeeca0 40208a3f  
3fffff10:  00000000 3ffefcac 3ffeebfc 40100920  
3fffff20:  00000000 3ffefcac 3ffeebfc 40100a21  
3fffff30:  00000000 00000000 3ffeec50 40212488  
3fffff40:  3ffefad4 3ffeebfc 3ffeec50 40212414  
3fffff50:  00000000 3ffeebfc 3ffeec50 40202078  
3fffff60:  00000000 00000000 3ffefbf4 4020b1e4  
3fffff70:  3ffefbf4 0000002b 4020101c 401001b9  
3fffff80:  3ffefc9c 3ffe8a87 00000000 40206943  
3fffff90:  3fffdad0 00000000 3ffeebf8 40201202  
3fffffa0:  feefeffe feefeffe 3ffeee58 40208514  
3fffffb0:  feefeffe feefeffe 3ffe85fc 40100cd9  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0004a950
~ld

```

vatsake, 2022-01-27:
Any updates on this issue?

XBoter, 2022-01-28:
nope

closed 2022-01-28 by XBoter

Connect-and-Exchange, 2022-08-26:
Seems like a dead project - the hole AsyncServer stuff... Been chasing same issue. Other issues also occur like extremely slow running code because the webserver 'hangs' / ' consumes' all cpu cycles. it seems also not updated since 2 years in the repo.

stevesch, 2022-08-26:
Yes, I'm not sure about "dead," but I was wondering how stable and maintained this is.  I like the project, but the versions I've used seem to have some issues under heavy load (especially using server events and possible websockets).
This fork https://github.com/ul-gh/ESPAsyncWebServer seems to have some ideas (I actually used it in the past with some luck-- was more stable), but the two have diverged a bit.
Anyone have any thoughts/updates on whether the ul-gh modifications would be legitimate to re-integrate into this master, or if the changes that have happened since should cover the cases (specifically, ul-gh adds some locks around critical sections).