me-no-dev/ESPAsyncWebServer #1112: Adding STA interface after AP

jmattioni, 2022-01-18:
This is more a question than an issue. I'm developing a way for the user to set up the device on Wifi by starting with the AP interface and serving a setup page there where the user enters their ssid and password. Once entered, the system attempts to test the entered credentials by adding the Wifi interface. With the server already bound to the AP interface the result of attempting Wifi.begin is a crash.

Is it possible to add the STA interface with AsyncWebServer already bound to the AP interface? Do I need to stop the server to add the STA interface and then start it again?

Thanks.

Siwykrzywy, 2022-02-02:
I have exactly the same problem. Did you find a solution?

jmattioni, 2022-02-03:
> I have exactly the same problem. Did you find a solution?

Sort of. I went back to using ESP8266WebServer. But, that also didn't fix the issue. What got it working for me was to place all the code for Wifi setup web interaction into setup() and not use loop(). When Wifi STA is subsequently enabled from setup() to test the credentials it works. This approach may also work for the Async web server, but I didn't test it.

Here's the code segment at the bottom of setup() when program is in Wifi setup mode:

```
void setupModeInit() {	
       //...
       //(code to enable AP mode and set up server pages)
       //....

        // All web interaction is done in setup() because http server
	// becomes unresponsive in loop() once Wifi STA connects
	// (actual issue may be from calling WiFi.begin in loop())

	while (true) { // This loop will exit only by reboot!
		dnsServer.processNextRequest();
		server.handleClient(); // Handles all web user interaction
	}
}
```
Not a good solution for long-term operation, but works fine for Wifi setup.

closed 2022-04-30 by stale[bot]

Belleson, 2022-04-30:
> Sort of. I went back to using ESP8266WebServer. But, that also didn't fix the issue. What got it working for me was to place all the code for Wifi setup web interaction into setup() and not use [...]
[snipped]
> 
> Not a good solution for long-term operation, but works fine for Wifi setup.

I'm creating a similar app on ESP32. Everything is working except mDNS -- so it connects to local wifi but I cannot tel a user the URL/IP to open the web page. @jmattioni do you have a solution for  that problem?

jmattioni, 2022-05-06:
Make sure the loop calls DNSServer processNextRequest() in addition to ESP8266WebServer handleClient().