me-no-dev/ESPAsyncWebServer #1117: Response slowdown due to handler registration order [not an issue, documentation emphasise]

Gobol, 2022-02-03:
Hi!

I just want to share my 5c about handler registration order.

As stated in README: 

**Handlers are evaluated in the order they are attached to the server. The canHandle is called only if the Filter that was set to the Handler return true.**

Very true. And keep in mind that ...

`server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.htm");`

will have disastrous slowdown effect on every handler below this statement.

eg. 

'
   server.serveStatic("/", SPIFFS, "/").setDefaultFile("index.htm");
   server.on("/heap", HTTP_GET, [](AsyncWebServerRequest *request){...}
   server.begin();
'

The subsequent requests to `/heap` may be delayed 500-1000ms due to SPIFFS processing...

Just to keep in mind.
Howgh.

zekageri, 2022-02-03:
Interesting... First of all use LITTLEFS instead of SPIFFS.

Second, the handler should not touch the file system if the url param is not matched. It's just simple String matching.

If you got /heap the handler looks for the first attached endpoint, it is "/" so it matching "/heap" == "/" if it's not the same goes to the next index.

Gobol, 2022-02-03:
Ok, I promise I will use LittleFS :)

But, nonetheless - `serveStatic` handler used before other routes makes my code respond with awful delay to request (using async fetch(), TBH ). I don't know why it works like that, but that is the fact; it looks like serveStatic is trying to find filename on FS that looks like a GET request...

zekageri, 2022-02-04:
Oh it actually makes sense. Server static in fact is scanning the file system for potential routes. So actually it has to make FS calls. But i think the readme mentions that you must place serve static to the end of the calls. Mine is right before my not found handler and static routes are before these.

closed 2022-04-30 by stale[bot]