me-no-dev/ESPAsyncWebServer #1118: WebSocket maximum data size for receiving

veneris1991, 2022-02-06:
In an ESP 32 project, I use `handleWebSocketMessage(void *arg, uint8_t *data, size_t len)` to receive a json from a web page. If the Web page sends a json that is too large (len > 1134 ), then the event does not fire. How to increase the buffer for receiving data? Memory in a heap of freedoms. 
Сomplete code:
```
void handleWebSocketMessage(void *arg, uint8_t *data, size_t len) {   //получаю от web страницы
  AwsFrameInfo *info = (AwsFrameInfo*)arg;
  if (info->final && info->index == 0 && info->len == len && info->opcode == WS_TEXT) {            
       Serial.println(len);       
       DynamicJsonDocument JsonFromWebSocket(10000);  
       deserializeJson(JsonFromWebSocket, data);
       serializeJsonPretty(JsonFromWebSocket, Serial);          
}
```

zekageri, 2022-02-06:
I run into similar problem before. First of all you must handle the desertification error because there will be in the future. 

The issue is that the socket messages will come in chunks if they not fit into the buffer. You are handling just the last chunk of that message and the deserialization will fail, but since you did not handle the error you don't know about it. 

What I did:

Create a buffer outside of this function. ( Make it global ) and put the messages into this buffer until final. If final triggers, deserialize that global variable and process. After deserialization make the buffer empty again, prepare it for the next message.

I will give you some example soon.

veneris1991, 2022-02-06:
> Я сталкивался с подобной проблемой раньше. Прежде всего, вы должны обработать ошибку десериализации, потому что она будет в будущем.
> 
> Проблема в том, что сообщения сокетов будут поступать кусками, если они не помещаются в буфер. Вы обрабатываете только последний фрагмент этого сообщения, и десериализация завершится ошибкой, но, поскольку вы не обработали ошибку, вы не знаете об этом.
> 
> Что я сделал:
> 
> Создайте буфер вне этой функции. (Сделайте его глобальным) и поместите сообщения в этот буфер до финала. Если срабатывает final, десериализуйте эту глобальную переменную и обработайте. После десериализации снова сделайте буфер пустым, подготовьте его к следующему сообщению.
> 
> Я скоро приведу вам пример.

Thanks for the answer. I did just that, but this solution doesn't seem elegant to me.

zekageri, 2022-02-06:
```C
char * socketData;
int currSocketBufferIndex = 0;
void setup(){
   // allocate memory for socketData in PSRAM if you have.
   socketData  = (char *) ps_malloc (SOCKET_DATA_SIZE * sizeof (char));
}

void handleWebSocketMessage(void *arg, uint8_t *data, size_t len) { 
  if (type == WS_EVT_DATA) {
        AwsFrameInfo * info = (AwsFrameInfo*)arg;
            if(info->opcode == WS_TEXT){
                for (size_t i = 0; i < len; i++){
                    socketData[currSocketBufferIndex] = data[i];
                    currSocketBufferIndex++;
            }
            if( currSocketBufferIndex >= info->len ){
                socketData[currSocketBufferIndex] = '\0';
                SpiRamJsonDocument doc(SOCKET_DATA_SIZE);
                DeserializationError error  =  deserializeJson(doc, socketData);
                currSocketBufferIndex = 0;
                if( !error ){
                     // Process socket data in doc
                }else{
                    // process deserialization error
                    Serial.printf("Socket data deserialization error: %s\n",error.c_str());
                }
            }
        }
    }     
}
```

```C
String socketData = "";

void handleWebSocketMessage(void *arg, uint8_t *data, size_t len) { 
  if (type == WS_EVT_DATA) {
        AwsFrameInfo * info = (AwsFrameInfo*)arg;
            if(info->opcode == WS_TEXT){
                socketData += String(data);
                if( info->final ){
                    SpiRamJsonDocument doc(SOCKET_DATA_SIZE);
                    DeserializationError error  =  deserializeJson(doc, socketData);
                    socketData = "";
                    if( !error ){
                         // Process socket data in doc
                    }else{
                        // process deserialization error
                        Serial.printf("Socket data deserialization error: %s\n",error.c_str());
                    }
                }
        }
    }     
}
```

You can do this with String ( not recommended ) char array , char array in PSram if you have.

zekageri, 2022-02-06:
Sorry for the formatting, this is not tested code i wrote it here.

zekageri, 2022-02-06:
Elegant or not, this is your only option. You must save the chucks and process the whole message.

veneris1991, 2022-02-06:
> ```c
> lizeJson(doc, socketData);
>                 c
> ```

Thank you very much for the replies. That's what I need.

closed 2022-02-06 by veneris1991