me-no-dev/ESPAsyncWebServer #1119: OTA problem with async

vatsake, 2022-02-07:
1. Choose a firmware for ESP
2. Start uploading the firmware
3. Let it upload for 1 second and then cancel the upload.
4. When you NOW upload a new firmware, the ESP wont boot at all.

vatsake, 2022-02-08:
Is there a way to get upload status?

Maybe if I could call Update.end() when file aborted, the esp will function as normal.

```
else if(_authenticated && upload.status == UPLOAD_FILE_ABORTED){
        Update.end();
        if (_serial_output) Serial.println("Update was aborted");
}
```

vatsake, 2022-04-06:
Can anyone else confirm this? I tried 5 different ESP8266s.

zekageri, 2022-06-15:
In my understanding the OTA should compare the hash of the two firmware. If the new firmware hash is broken it will load the older one and boot up. Can we see your update code?

zekageri, 2022-06-15:
I can see the problem. 
My sketch can not handle the refresh of the firmware update page too.
I'm interested in a solution too.
We should implement some kind of a timeout to the packets and close the request and end the update too...

zekageri, 2022-06-15:
My sketch looks like this:

```C
server.on("/newFirmware", HTTP_POST, [](AsyncWebServerRequest *request) {},
        [](AsyncWebServerRequest *request, const String &filename, size_t index, uint8_t *data, size_t len, bool final) {
            byte errCode = 1;
            if (!index) {
                hsh_Server.firmwareUpdating = true;
                int fileSize = UPDATE_SIZE_UNKNOWN;

                if (request->hasArg("fileSize")) {
                    fileSize = request->arg("fileSize").toInt();
                }

                if (!Update.begin(fileSize, U_FLASH)) {
                    #if SS_DEBUG_MODE
                        Serial.println("Update begin failed!");
                    #endif
                    errCode = 2;
                    final = true;
                }
            }

            if (Update.write(data, len) != len) {
                #if SS_DEBUG_MODE
                    Serial.println("Update write failed!");
                #endif
                errCode = 3;
                final = true;
            }

            if (final) {
                if (errCode == 2) {
                    request->send(500, "text/plain", "Update begin failed! Size mismatch or invalid file!");
                } else if (errCode == 3) {
                    request->send(500, "text/plain", "Update write failed!");
                    hsh_Server.sendFirmwareProgress(100, true, false);
                    hsh_Server.firmwareUpdating = false;
                } else if (!Update.end(true)) {
                    AsyncWebServerResponse *response = request->beginResponse(500, "text/plain", "Failed");
                    hsh_Server.sendFirmwareProgress(100, true, false);
                    request->send(response);
                    hsh_Server.firmwareUpdating = false;
                } else {
                    request->onDisconnect(firmwareRespHandler);
                    AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "Succeded");
                    hsh_Server.sendFirmwareProgress(100, true, true);
                    request->send(response);
                }
            };
        });
```

What i'm thinking on:

```C
#define FIRMWARE_UPDATE_TIMEOUT 1000
long lastFirmwareUpdt_Packet_MS = 0;

void checkFirmwareUpdate_Timeout(){
    if( hsh_Server.firmwareUpdating && millis() - lastFirmwareUpdt_Packet_MS >= FIRMWARE_UPDATE_TIMEOUT ){
        Update.end();
        hsh_Server.firmwareUpdating = false;
    }
}

server.on("/newFirmware", HTTP_POST, [](AsyncWebServerRequest *request) {},
        [](AsyncWebServerRequest *request, const String &filename, size_t index, uint8_t *data, size_t len, bool final) {
            lastFirmwareUpdt_Packet_MS = millis();
            byte errCode = 1;
            if (!index) {
                hsh_Server.firmwareUpdating = true;
                int fileSize = UPDATE_SIZE_UNKNOWN;

                if (request->hasArg("fileSize")) {
                    fileSize = request->arg("fileSize").toInt();
                }

                if (!Update.begin(fileSize, U_FLASH)) {
                    #if SS_DEBUG_MODE
                        Serial.println("Update begin failed!");
                    #endif
                    errCode = 2;
                    final = true;
                }
            }

            if (Update.write(data, len) != len) {
                #if SS_DEBUG_MODE
                    Serial.println("Update write failed!");
                #endif
                errCode = 3;
                final = true;
            }

            if (final) {
                if (errCode == 2) {
                    request->send(500, "text/plain", "Update begin failed! Size mismatch or invalid file!");
                } else if (errCode == 3) {
                    request->send(500, "text/plain", "Update write failed!");
                    hsh_Server.sendFirmwareProgress(100, true, false);
                    hsh_Server.firmwareUpdating = false;
                } else if (!Update.end(true)) {
                    AsyncWebServerResponse *response = request->beginResponse(500, "text/plain", "Failed");
                    hsh_Server.sendFirmwareProgress(100, true, false);
                    request->send(response);
                    hsh_Server.firmwareUpdating = false;
                } else {
                    request->onDisconnect(firmwareRespHandler);
                    AsyncWebServerResponse *response = request->beginResponse(200, "text/plain", "Succeded");
                    hsh_Server.sendFirmwareProgress(100, true, true);
                    request->send(response);
                }
            };
        });

void loop(){
    checkFirmwareUpdate_Timeout();
}
```

zekageri, 2022-06-15:
**Here is a simplified example. I'm testing it now.**

```C
#define FIRMWARE_UPDATE_TIMEOUT 1000
long lastFirmwareUpdt_Packet_MS = 0;
boolean firmwareUpdating = true;

void checkFirmwareUpdate_Timeout(){
    if( firmwareUpdating && millis() - lastFirmwareUpdt_Packet_MS >= FIRMWARE_UPDATE_TIMEOUT ){
        Update.end();
        firmwareUpdating = false;
    }
}

server.on("/newFirmware", HTTP_POST, [](AsyncWebServerRequest *request) {},
        [](AsyncWebServerRequest *request, const String &filename, size_t index, uint8_t *data, size_t len, bool final) {
            lastFirmwareUpdt_Packet_MS = millis();
            if (!index) {
                firmwareUpdating = true;
                int fileSize = UPDATE_SIZE_UNKNOWN;

                if (request->hasArg("fileSize")) {
                    fileSize = request->arg("fileSize").toInt();
                }

                if ( !Update.begin(fileSize, U_FLASH) ) {
                    Serial.println("Update begin failed!");
                    request->send(500, "text/plain", "Update failed!");
                    Update.end();
                    firmwareUpdating = false;
                }
            }

            if ( Update.write(data, len) != len ) {
                request->send(500, "text/plain", "Update failed!");
                Update.end();
                firmwareUpdating = false;
            }

            if (final) {
                if( !Update.end(true) ){
                    request->send(500, "text/plain", "Update write failed!");
                    Update.end();
                    firmwareUpdating = false;
                }else{
                    request->send(500, "text/plain", "Update write failed!");
                    Update.end();
                    firmwareUpdating = false;
                }
            }
        });

void loop(){
    checkFirmwareUpdate_Timeout();
}
```

zekageri, 2022-06-15:
It will end the update before it can begin.
This line is executed on the start of the update.

zekageri, 2022-06-15:
**Thinking about this:**

```C
void firmwareUpdateEnd( boolean isSuccess, AsyncWebServerRequest *request = NULL );

#define FIRMWARE_UPDATE_TIMEOUT 2000

boolean firmwareUpdating  = false;
long lastFirmwareUpdt_Packet_MS  = 0;

void checkFirmwareUpdate_Timeout(){
    if( firmwareUpdating && millis() - lastFirmwareUpdt_Packet_MS >= FIRMWARE_UPDATE_TIMEOUT ){
        firmwareUpdateEnd(false);
    }
}

void firmwareUpdateEnd( boolean isSuccess, AsyncWebServerRequest *request ){
    firmwareUpdating = false;
    sendFirmwareProgress(100, true, isSuccess); // <-- used to indicate the progress for the client via websockets.
    Update.end();
    if( isSuccess && request != NULL ){
        request->send(200, "text/plain", "Success");
    }else if( request != NULL ) {
        request->send(500, "text/plain", "Failed!");
    }
}

void firmwareHandler() {
    server.on("/newFirmware", HTTP_POST, [](AsyncWebServerRequest *request) {},
        [](AsyncWebServerRequest *request, const String &filename, size_t index, uint8_t *data, size_t len, bool final) {
            hsh_Server.lastFirmwareUpdt_Packet_MS = millis();

            if (!index) {
                hsh_Server.firmwareUpdating = true;
                int fileSize = UPDATE_SIZE_UNKNOWN;

                if (request->hasArg("fileSize")) {
                    fileSize = request->arg("fileSize").toInt();
                }

                if ( !Update.begin( fileSize, U_FLASH ) ) {
                    firmwareUpdateEnd(false);
                    return;
                }
            }

            if (Update.write(data, len) != len) {
                firmwareUpdateEnd(false);
                return;
            }

            if (final) {
                if (!Update.end(true)) {
                    firmwareUpdateEnd(false, request);
                }else{
                    firmwareUpdateEnd(true, request);
                }
            }
        });
}

void loop(){
    checkFirmwareUpdate_Timeout();
}

```

zekageri, 2022-06-15:
So you did not specify anything other then this:

```C
if (!index) {
    Update.end();
}
```

?

zekageri, 2022-06-15:
I see. But this approach can not cover other scenarios. What if you set a flag to indicate that the firmware is updating to the rest of your program. It will stay true if you not refresh the webpage and start a new update but anything else.

zekageri, 2022-06-15:
**I did it like this, just in case.** _( I'm on esp32 btw )_

```C
void firmwareRespHandler() {
    hsh_Performance.beginRestart();
}

void sSystem::checkFirmwareUpdate_Timeout(){
    if( firmwareUpdating && millis() - lastFirmwareUpdt_Packet_MS >= FIRMWARE_UPDATE_TIMEOUT ){
        firmwareUpdateEnd(false);
    }
}

void sSystem::firmwareUpdateEnd( boolean isSuccess, AsyncWebServerRequest *request ){
    firmwareUpdating = false;
    sendFirmwareProgress(100, true, isSuccess);
    Update.end();
    if( isSuccess && request != NULL ){
        request->onDisconnect(firmwareRespHandler);
        request->send(200, "text/plain", "Success");
    }else if( request != NULL ) {
        request->send(500, "text/plain", "Failed!");
        hshDisplay.addNoty("Failed to update firmware!");
    }
}

void sSystem::firmwareHandler() {
    server.on("/newFirmware", HTTP_POST, [](AsyncWebServerRequest *request) {},
        [](AsyncWebServerRequest *request, const String &filename, size_t index, uint8_t *data, size_t len, bool final) {
            hsh_Server.lastFirmwareUpdt_Packet_MS = millis();

            if (!index) {
                Update.end();
                hsh_Server.firmwareUpdating = true;
                int fileSize = UPDATE_SIZE_UNKNOWN;

                if (request->hasArg("fileSize")) {
                    fileSize = request->arg("fileSize").toInt();
                }

                if ( !Update.begin( fileSize, U_FLASH ) ) {
                    hsh_Server.firmwareUpdateEnd(false);
                    return;
                }
            }

            if (Update.write(data, len) != len) {
                hsh_Server.firmwareUpdateEnd(false);
                return;
            }

            if (final) {
                if (!Update.end(true)) {
                    hsh_Server.firmwareUpdateEnd(false, request);
                }else{
                    hsh_Server.firmwareUpdateEnd(true, request);
                }
            }
        });
}

void loop(){
    checkFirmwareUpdate_Timeout();
}

```

zekageri, 2022-06-15:
Nah, i think it is out of scope for the webserver. It is two different library. You can use the Update class with any other thing.
I have a NodeJS server where i store firmwares. The ESP can automatically check for new firmware by itself on the server and pulls it for itself via a regular HTTPS request. And it uses Update too. Not related to Async webserver.